sudar/Arduino-Makefile #314: I mistakenly overridden the master branch

sudar, 2015-01-23:
While adding credit for the recent modification (issue #313) I mistakenly did a force push with an outdated master. It seems to have overridden last couple of commits. I know I shouldn't have used force push...

Can someone who has the old code please send a pull request to get back these old commits?

(cc @sej7278 )

sej7278, 2015-01-23:
i've made #315 don't know if that's too old, i think its up-to-date other than the new teensy commit

sudar, 2015-01-23:
Thank you very much @sej7278. You saved me from my stupidity.

I will merge #313 on top of this now.

sej7278, 2015-01-23:
haha my fault for nagging about HISTORY.md ;)

sudar, 2015-01-23:
@sej7278,

> haha my fault for nagging about HISTORY.md ;)

Not at all :)

It is just that I learned an important lesson today :)

After I modified the `HISTORY.md` I did a `git push` and there was a connection issue. I did a git push and git complained. I thought it was because of the connection issue and then did a git push --force. As I pressed the enter key, I realised the mistake I did.

I think the repo should be back to its original state now. Thank you very much @sej7274

closed 2015-01-23 by sudar