sudar/Arduino-Makefile #316: MAX 32 build fails with undefined reference to `_image_header_info'

e76nystrom, 2015-02-01:
I am trying to build the Blink example for MAX32 image using mpide-0023-windows-20140821 using the latest version of chipKIT.mk from this repository.  The link fails with undefined reference to `_image_header_info'.

I did a lot of testing and compared this build to the mpide build and found several switches different in the link command and several files missing from the libcore.a

The mpide link has the following switches not found in this link -mdebugger -mno-peripheral-libs -nostartfiles

The files missing from libcore.a are cpp-startup.S.o  crti.S.o  crtn.S.o  pic32_software_reset.S.o  vector_table.S.o

I can get the link command from this script to work by adding the above switches and using the core.a produced by mpide.

Thanks

sej7278, 2015-02-01:
have you tried an earlier build, the 2014 builds are not yet supported: 

https://github.com/sudar/Arduino-Makefile/pull/256

under cygwin, we only support mpide-0023-windows-20130715 as per the travis-ci script

e76nystrom, 2015-02-02:
Thanks for the info.  That fixes it.

closed 2015-02-02 by e76nystrom