sudar/Arduino-Makefile #318: mmcu option is repeated several times

jforberg, 2015-02-19:
My makefile is outputting the MCU tag several times more than it should. I'm just compiling an empty project for the atmega328 using Arduino-Makefiled535bf5fde2d49b190913e858ce644c1a50c3c40

The generated compiler call looks like this:

```/usr/bin/avr-gcc -MMD -c -mmcu=atmega328p atmega328p atmega328p atmega328p atmega328p atmega328p atmega328p -DF_CPU=16000000L 8000000L -DARDUINO=158 -DARDUINO_ARCH_AVR -D__PROG_TYPES_COMPAT__ -I/usr/share/arduino/hardware/arduino/avr/cores/arduino -I/usr/share/arduino/hardware/arduino/avr/variants/    -Wall -ffunction-sections -fdata-sections -Os -std=gnu11 timelock.c -o bin/atmega328/timelock/timelock.o```

My Makefile: https://gist.github.com/jforberg/1c2dd895f97c9665e8c2

sej7278, 2015-02-20:
we don't really support arduino 1.5.8, try 1.0.6

although avr boards should work, but it depends how you've laid out your atmega328 entry in boards.txt, plus your makefile looks overly complicated.

also if you're using BareArduinoProject, you should report bugs there.

jforberg, 2015-02-20:
Thanks. That may explain some of the strange bugs i'm getting with recent versions of arduino.

I'm not aware of Barearduinoproject, not sure where i got the makefile from actually.

Anyway, I worked around this by manually specifying MCU, CPU, baudrate etc. So I'll close this issue.

closed 2015-02-20 by jforberg

sej7278, 2015-02-20:
you shouldn't need to manually specify things.

i use this (on 1.0.5) boards.txt when i'm using a bare atmega328p:

```
atmega328bb.name=ATmega328 on a breadboard (3.3v, 8 MHz internal clock)
atmega328bb.upload.maximum_size=30720
atmega328bb.upload.protocol=usbasp
atmega328bb.upload.using=usbasp
atmega328bb.bootloader.low_fuses=0xE2
atmega328bb.bootloader.high_fuses=0xD8
atmega328bb.bootloader.extended_fuses=0x07
atmega328bb.bootloader.path=atmega
atmega328bb.bootloader.file=ATmegaBOOT_168_atmega328_pro_8MHz.hex
atmega328bb.bootloader.unlock_bits=0x3F
atmega328bb.bootloader.lock_bits=0x0F
atmega328bb.build.mcu=atmega328p
atmega328bb.build.f_cpu=8000000L
atmega328bb.build.core=arduino
atmega328bb.build.variant=standard
```

and this is my local Makefile:

```
BOARD_TAG = atmega328bb
BOARDS_TXT = ~/arduino/hardware/breadboard/boards.txt
include /usr/share/arduino/Arduino.mk
```