sudar/Arduino-Makefile #319: Source file lines not included when dumping elf file

GreyGnome, 2015-02-20:
Hi,
I am studying the assembly language output of the compiler and I notice that Arduino-Makefile does not include the facility to allow me to see the lines in the source file that created the assembly output, when I avr-objdump the elf file.

I think it's because the -g option is missing from the compile. In general, it seems that Arduino-Makefile has changed some compile options (the aforementioned -g, also -Os when compiling the .ino file). I don't know if that's a safe thing to do; this makes output ever so slightly different (and then we have problems such as I have seen).

I am using Arduino 1.0.5 on Fedora 20, and I'm using 1.3.4 of Arduino.mk.

I tried to define DEBUG in my local Makefile:
DEBUG = 1
include /usr/share/arduino/Arduino.mk

but I got compile errors (these errors do not exist normally):
build-2560/Simple.o: In function `loop':
/home/schwager/sketchbook/libraries/EnableInterrupt/Examples/Simple/Simple.ino:97: undefined reference to `Z'
/home/schwager/sketchbook/libraries/EnableInterrupt/Examples/Simple/Simple.ino:128: undefined reference to `Z'
collect2: error: ld returned 1 exit status

Thanks!

GreyGnome, 2015-02-20:
I set this in my local Makefile:

CFLAGS_STD = -g
CXXFLAGS_STD = -g

...and that took care of it! Still, peculiar that DEBUG fails.

GreyGnome, 2015-02-20:
BTW, thanks for a very nice project! I think this will be terrifically useful for me going forward!

closed 2015-02-20 by GreyGnome

sej7278, 2015-02-20:
yes you're probably doing it the best way, we try to match the IDE's flags, and don't really want debugging (-g) on by default.