sudar/Arduino-Makefile #320: Allow tilde in all paths

mithat, 2015-02-22:
I have a copy of the repo in `/home/mithat/Build/Arduino-Makefile/Arduino.mk`.
I have a test project in `/home/mithat/Arduino/mk-sudar-test/`.

The following project makefile succeeds in compiling:

```makefile
# Project config
ARDUINO_LIBS = Arduino-IRremote-master
BOARD_TAG = uno
MONITOR_PORT = /dev/ttyACM0

# System config
ARDUINO_DIR = /home/mithat/.opt/arduino/
ARDMK_DIR = ~/Build/Arduino-Makefile/
AVR_TOOLS_DIR = /usr

# Include parent Makefile from <https://github.com/sudar/Arduino-Makefile>
include ~/Build/Arduino-Makefile/Arduino.mk
```

But this one does not.

```makefile
# Project config
ARDUINO_LIBS = Arduino-IRremote-master
BOARD_TAG = uno
MONITOR_PORT = /dev/ttyACM0

# System config
ARDUINO_DIR = ~/.opt/arduino/
ARDMK_DIR = ~/Build/Arduino-Makefile/
AVR_TOOLS_DIR = /usr

# Include parent Makefile from <https://github.com/sudar/Arduino-Makefile>
include ~/Build/Arduino-Makefile/Arduino.mk
```

It would be nice if I could tilde without fear.

sej7278, 2015-02-23:
are you sure you have the latest version from git, as i thought we sorted the tilde issues (caused by quotes). i can't see why only ARDUINO_DIR fails, can you post a build log?

mithat, 2015-02-23:
Sure. Here it is:
```
$ make
-------------------------
Arduino.mk Configuration:
- [AUTODETECTED]       CURRENT_OS = LINUX 
- [USER]               ARDUINO_DIR = ~/.opt/arduino-1.0.6/ 
- [USER]               ARDMK_DIR = ~/Build/Arduino-Makefile/ 
- [AUTODETECTED]       ARDUINO_VERSION = 106 
- [DEFAULT]            ARCHITECTURE =  
- [DEFAULT]            VENDOR = arduino 
- [AUTODETECTED]       ARDUINO_PREFERENCES_PATH = /home/mithat/.arduino/preferences.txt 
- [AUTODETECTED]       ARDUINO_SKETCHBOOK = /home/mithat/Arduino-106 (from arduino preferences file)
- [USER]               AVR_TOOLS_DIR = /usr 
- [COMPUTED]           ARDUINO_LIB_PATH = ~/.opt/arduino-1.0.6//libraries (from ARDUINO_DIR)
- [COMPUTED]           ARDUINO_VAR_PATH = ~/.opt/arduino-1.0.6//hardware/arduino//variants (from ARDUINO_DIR)
- [COMPUTED]           BOARDS_TXT = ~/.opt/arduino-1.0.6//hardware/arduino//boards.txt (from ARDUINO_DIR)
- [DEFAULT]            USER_LIB_PATH = /home/mithat/Arduino-106/libraries (in user sketchbook)
- [DEFAULT]            PRE_BUILD_HOOK = pre-build-hook.sh 
- [USER]               BOARD_TAG = uno 
- [COMPUTED]           CORE = arduino (from build.core)
- [COMPUTED]           OBJDIR = build-uno (from BOARD_TAG)
- [COMPUTED]           ARDUINO_CORE_PATH = ~/.opt/arduino-1.0.6//hardware/arduino//cores/arduino (from ARDUINO_DIR, BOARD_TAG and boards.txt)
- [DETECTED]           MONITOR_BAUDRATE = 9600  (in sketch)
- [DEFAULT]            OPTIMIZATION_LEVEL = s 
- [DEFAULT]            MCU_FLAG_NAME = mmcu 
- [DEFAULT]            CFLAGS_STD =  
- [DEFAULT]            CXXFLAGS_STD =  
- [COMPUTED]           DEVICE_PATH = /dev/ttyACM0 (from MONITOR_PORT)
- [DEFAULT]            FORCE_MONITOR_PORT =  
- [AUTODETECTED]       Size utility: AVR-aware for enhanced output
-
-                      ARDUINO_LIBS =
- [USER]                 Arduino-IRremote-master
- [COMPUTED]           BOOTLOADER_PARENT = ~/.opt/arduino-1.0.6//hardware/arduino//bootloaders (from ARDUINO_DIR)
- [COMPUTED]           ARDMK_VERSION = 1.3.4 
- [COMPUTED]           CC_VERSION = 4.8.1 (avr-gcc)
-------------------------
mkdir -p build-uno
/usr/bin/avr-g++ -x c++ -include Arduino.h -MMD -c -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=106  -D__PROG_TYPES_COMPAT__ -I~/.opt/arduino-1.0.6//hardware/arduino//cores/arduino -I~/.opt/arduino-1.0.6//hardware/arduino//variants/standard     -I/home/mithat/Arduino-106/libraries/Arduino-IRremote-master -Wall -ffunction-sections -fdata-sections -Os -fno-exceptions  mk-sudar-test.ino -o build-uno/mk-sudar-test.o
cc1plus: fatal error: Arduino.h: No such file or directory
compilation terminated.
/home/mithat/Build/Arduino-Makefile/Arduino.mk:1201: recipe for target 'build-uno/mk-sudar-test.o' failed
make: *** [build-uno/mk-sudar-test.o] Error 1
```

sej7278, 2015-02-23:
interesting, seems its normal for Makefiles not to do tilde expansion. i found the workaround is to use $(HOME) instead of ~

```make
BOARD_TAG 		= uno
MONITOR_PORT 	= /dev/ttyACM0
ARDUINO_DIR 	= $(HOME)/opt/arduino-1.0.6
AVR_TOOLS_DIR 	= $(HOME)/opt/arduino-1.0.6/hardware/tools/avr
```

alternatively we could patch Arduino.mk to add a space after -I in the include paths so your shell could do the expansion, but its still not reliable:

```diff
diff --git a/Arduino.mk b/Arduino.mk
index a5b9bba..95a9826 100644
--- a/Arduino.mk
+++ b/Arduino.mk
@@ -901,8 +901,8 @@ rwildcard=$(foreach d,$(wildcard $1*),$(call rwildcard,$d/,$2) $(filter $(subst
 
 # Gets include flags for library
 get_library_includes = $(if $(and $(wildcard $(1)/src), $(wildcard $(1)/library.properties)), \
-                           -I$(1)/src, \
-                           $(addprefix -I,$(1) $(wildcard $(1)/utility)))
+                           -I $(1)/src, \
+                           $(addprefix -I ,$(1) $(wildcard $(1)/utility)))
 
 # Gets all sources with given extension (param2) for library (path = param1)
 # for old (1.0.x) layout looks in . and "utility" directories
@@ -990,7 +990,7 @@ endif
 
 # Using += instead of =, so that CPPFLAGS can be set per sketch level
 CPPFLAGS      += -$(MCU_FLAG_NAME)=$(MCU) -DF_CPU=$(F_CPU) -DARDUINO=$(ARDUINO_VERSION) $(ARDUINO_ARCH_FLAG) -D__PROG_TYPES_COMPAT__ \
-        -I$(ARDUINO_CORE_PATH) -I$(ARDUINO_VAR_PATH)/$(VARIANT) \
+        -I $(ARDUINO_CORE_PATH) -I $(ARDUINO_VAR_PATH)/$(VARIANT) \
         $(SYS_INCLUDES) $(PLATFORM_INCLUDES) $(USER_INCLUDES) -Wall -ffunction-sections \
         -fdata-sections
```

mithat, 2015-02-23:
My thinking is that if tilde expansion isn't supported for Makefiles (learned something new!), then trying to support it will likely create more problems than it solves. I'm fine with `$(HOME)`.

sej7278, 2015-03-05:
can we close this?

mithat, 2015-03-05:
Closing it is fine with me. I think "won't fix" is the best option.

sudar, 2015-03-08:
I thought about it for a while and thought it would be best to leave this issue open, but mark it with `wontfix`, so that we don't get a duplicate issue regarding this again.

May be we can close it once we document in the README that tilde's are not supported.

doeg, 2016-04-02:
I just ran into this issue on OS X and this thread was a lifesaver. Thanks!