sudar/Arduino-Makefile #323: Compiling Eigen linear algebra library

maartenverbandt, 2015-03-05:
Example code refering to this issue is found in https://github.com/maartenverbandt/ballbot_arduino.

Issue: compiling the external library Eigen - http://eigen.tuxfamily.org/index.php?title=Main_Page

Compiling the program without the library does not result in issues. Adding the Eigen code does. Compiler error:
In file included from libraries/lib_shared/eigen313/lib/Eigen313.h:35:0,
                 from src/src_teensy/main.cpp:6:
libraries/lib_shared/eigen313/lib/Eigen/Core:28:19: fatal error: complex: No such file or directory
 #include <complex>
                   ^
compilation terminated.

I the past I already compiled the library for teensy using my own custom makefile, so the problem probabely lies in the arduino-makefile itself.

ladislas, 2015-03-05:
can you share your makefile in a gist please? :)

maartenverbandt, 2015-03-05:
I'm sorry but I don't fully understand what you mean ;) The makefile I 
used is available in the repo that I mentioned in the issue tracker.

sudar, 2015-03-05:
@maartenverbandt The makefile that is there in this repo is the master makefile.

You would have created a makefile from where you are including this master makefile and setting some variables right? @ladislas was asking for that makefile.

In addition to the makefile if you can send us the output of your make command then it would be very helpful for debugging this issue.

maartenverbandt, 2015-03-05:
The Makefile in the repo is the makefile where I declare where to find sources, libraries, etc.. in the arduino folder, you find the Arduino.mk and Common.mk which are distributed by this repo and included in my own makefile. So up to my knowledge, you call make from my code and you get the error. The compile error that pops up:

In file included from libraries/lib_shared/eigen313/lib/Eigen313.h:35:0,
from src/src_teensy/main.cpp:6:
libraries/lib_shared/eigen313/lib/Eigen/Core:28:19: fatal error: complex: No such file or directory
#include <complex>
^
compilation terminated.

It seems having trouble including the headers where there is no .h or .hpp extension because if I comment out the include in the Eigen/Core header file, it gets stuck on the next include without extension.
I hope this is the info you are looking for?

ladislas, 2015-03-05:
I guess the real `Makefile` is called `Makefile` and those with the `~` at the end are not working.

If that's the case, it does not look at all like the regular `Makefiles` and I can't even compile your code (I'm on OS X, what OS are you on?)

maartenverbandt, 2015-03-05:
The real Makefile is indeed called Makefile. The ones with ~ in the end 
slipped in by accident as they are automatically generated by linux and 
shouldn't be in the repo at all.
My appologies.

What do you mean with my code does not compile? Do you mean that there 
is an error occuring other than with the Eigen library?

I am working on ubuntu 14.4, so I am using linux' make. My main makefile 
is designed in such a way so that all of my code is automatically 
included in my project. But in the end, it is doing just the same as the 
examples I have seen from you guys, defining the LOCAL_C_SRCS, 
LOCAL_CPP_SRCS, CPPFLAGS and ARDUINO_LIBS.

ladislas, 2015-03-05:
I tried it on linux

```
/usr/share/arduino/hardware/tools/avr/bin/avr-g++ -MMD -c -I./arduino  -Ilibraries/lib_shared/eeprom/lib/  -Ilibraries/lib_shared/eigen313/lib/  -Ilibraries/lib_shared/mavlink/lib/  -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=105  -D__PROG_TYPES_COMPAT__ -I/usr/share/arduino/hardware/arduino//cores/arduino -I/usr/share/arduino/hardware/arduino//variants/eightanaloginputs    -I/usr/share/arduino/libraries/EEPROM   -Wall -ffunction-sections -fdata-sections -Os -fno-exceptions  src/src_teensy/main.cpp -o build-nano328/src/src_teensy/main.o
In file included from src/src_teensy/main.cpp:4:0:
libraries/lib_shared/mavlink/lib/mavlink.h:6:88: fatal error: /home/maarten/projects/mavlink/build/include/v1.0/ballbot_messages/mavlink.h: No such file or directory
 #include "/home/maarten/projects/mavlink/build/include/v1.0/ballbot_messages/mavlink.h"
                                                                                        ^
compilation terminated.
make: *** [build-nano328/src/src_teensy/main.o] Error 1
```

can you arrange your code so your links are not hard-coded for your computer?

maartenverbandt, 2015-03-05:
My appologies (once again..) I forgot about this dependency.

I pushed the update to my repo so now you should be able to compile the mavlink library and (hopefully) the Eigen error will pop up.

ladislas, 2015-03-05:
here is my error now:

```
/usr/share/arduino/hardware/tools/avr/bin/avr-g++ -MMD -c -I./arduino  -Ilibraries/lib_shared/eeprom/lib/  -Ilibraries/lib_shared/eigen313/lib/  -Ilibraries/lib_shared/mavlink/lib/  -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=105  -D__PROG_TYPES_COMPAT__ -I/usr/share/arduino/hardware/arduino//cores/arduino -I/usr/share/arduino/hardware/arduino//variants/eightanaloginputs    -I/usr/share/arduino/libraries/EEPROM   -Wall -ffunction-sections -fdata-sections -Os -fno-exceptions  src/src_teensy/main.cpp -o build-nano328/src/src_teensy/main.o
In file included from libraries/lib_shared/eigen313/lib/Eigen313.h:35:0,
                 from src/src_teensy/main.cpp:6:
libraries/lib_shared/eigen313/lib/Eigen/Core:28:19: fatal error: complex: No such file or directory
 #include <complex>
```

maartenverbandt, 2015-03-05:
That's the error I was talking about. I hope you have a better clue what is going wrong..

ladislas, 2015-03-05:
first it would be great if you could sanitize all your files.

it's quite hard to know what works and what doesn't, which one you use and so on.

maartenverbandt, 2015-03-05:
I hope you mean by sanitize: delete all files that slipped in by accident? If so, that is done. I pushed a clean set of files to the repo. 

If you try to compile the main.cpp without the Eigen-header, it should compile without a problem. If you do include it, the error pops up. To me, that means that Eigen is the only problem and you can thrust the other files to be sane. Also, I can compile the whole project for teensy (relying on my own makefile) so I think that the sources are all ok and there is going something wrong with the includes.

I hope you have what you need to find the root of the problem?

sej7278, 2015-03-05:
this isn't going to work for a start: https://github.com/maartenverbandt/ballbot_arduino/blob/master/libraries/lib_shared/eigen313/lib/Eigen/Core#L28

plus its including a bunch of files in src/Core/ which doesn't exist

and its including WProgram.h which doesn't even exist in arduino 1.x

maartenverbandt, 2015-03-06:
I know the includes are the problem, but Eigen should be able to include 
them somehow because I already compiled the source for my Teensy3.1 
using my very own makefile. As Far as I know, Eigen is self contained so 
no external sources should be loaded. Also if you search in the source 
directory, there are some files called complex.h, so I suppose they are 
meant to be included but somehow they are not.

If you go to the arduino folder, you see a file WProgram.h which is 
simply a wrapper for arduino.h, so that doesn't cause any trouble.

ladislas, 2015-03-06:
after compiling, did the program even work on your Teensy?

sej7278, 2015-03-06:
where there is ```#include <complex>``` try changing that to either of:

```cpp
#include <complex.h>
#include "complex.h"
```

also it seems the Core directory is one level up from where its expect - so the directory structure is wrong too.

maartenverbandt, 2015-03-06:
I made a new github repo with my teensy3.1 source and compilation output.

https://github.com/maartenverbandt/ballbot_teensy

If you have a teensy3.1, you can flash it by calling make upload and you 
will see it perfectly working. (I'm doing this project already for 8 
months but now I'm just trying to flash it to the arduino..)

The include structure is as far as I know completely the same, so 
comments on changing filenames and folder hierarchy are as far as I know 
irrelevant? Maybe you can check my teensy makefile for any differences 
with the arduino makefile?

maartenverbandt, 2015-03-06:
I pushed my bare teensy3.1 code and makefile to show that the include 
structure of Eigen shouldn't cause any problems, at least not with the 
makefile I uploaded.

Eigen is self-contained and only requires inclusion of the header file, 
as I did with my teensy..

stevenlovegrove, 2021-09-14:
I know this thread is old, but I just got here from google. I've found this problem is caused because Eigen requires a C++ standard library which is not included by default in Arduino. <complex> is part of the standard library (as are <climits> etc which are the other errors you will find if you comment out that one line in the Eigen header.

If you use Eigen packaged for Arduino (or other embedded platform) you also must add a stdlib implementation such as http://git.uclibc.org/uClibc++ (packaged for Arduino as ArduinoSTL for e.g.).