sudar/Arduino-Makefile #324: Thank you everyone

sudar, 2015-03-14:
I have been waiting for the number of stars to reach 500 to write this thank you note, but then it was struck at 499 for quite a few days, so I went ahead and published it anyway :)

This project won't had been a success without the help of some many people, especially contributors. At last count we have had contributions from [42 people](https://github.com/sudar/Arduino-Makefile/graphs/contributors). Yes, the magic number 42 ;)

I also want to take this opportunity to thank @sej7278 who addresses all the issues and pull requests before me. These days, I just wait for him to give a :+1: before merging a pull request :)

As a thank you note and also to make things much smoother, I have added him as a collaborator for this project. Thank you @sej7278 for all the hard work. It is a pleasure to work with you.

ladislas, 2015-03-14:
Congrats guys and thank you very much for you incredible work :) :+1:

sej7278, 2015-03-14:
Oh wow, now you're in trouble letting me commit! ;-) I feel quite privileged.

Thanks @sudar, @ladislas and everyone else who keep this project moving forward.

sudar, 2015-03-14:
> Oh wow, now you're in trouble letting me commit! ;-)

lol

But seriously, thank you for all the hard work.

dmcinnes, 2015-03-16:
Thank YOU! :+1:

sudar, 2015-03-17:
Now that we have crossed more than 500 stars, I am closing this issue.

closed 2015-03-17 by sudar