sudar/Arduino-Makefile #325: Arduino 1.6.1 + Teensy 31 support

mgcrea, 2015-03-16:
Just to share my experiences trying to make the build work with latest Arduino (1.6.1) on OSX for a teensy 3.1! Tried to exactly match the output of the Arduino compile but I have not been able to properly compile anything, but I guess it might be useful if someone tries to achieve the same thing:

https://gist.github.com/mgcrea/00880542f1ce8f561ad6

Not sure what your plans are for this version, tried the Arduino CLI interface without luck (relates to #92).

sej7278, 2015-03-16:
neither arduino-mk nor teensy supports 1.6.1, however 1.0.6 works fine.

i've not had a look at the arduino cli tools yet.

what's the diff show?

Mike-Menci, 2015-04-01:
1.0.6 works fine with Teensy3.1
BR

sudar, 2015-04-07:
@sej7278 Can we close this?

sej7278, 2015-04-07:
@sudar - probably, TeensyDuino itself is still a work in progress for 1.6.3, I suspect when its done the Makefile will "just work" (TM)

Mike-Menci, 2015-04-07:
Yes

BR, 
Mike

sudar, 2015-04-08:
Cool. Closing it for now. We can re-visit once TeenyDuino is updated.

closed 2015-04-08 by sudar

mgcrea, 2015-04-08:
http://www.pjrc.com/teensy/td_download.html
> Teensyduino 1.22 supports Arduino versions 1.0.6 and 1.6.1 and 1.6.2 and 1.6.3.

So it looks like it's finalized?

mgcrea, 2015-04-08:
By the way, here is the Teensy's author response regarding the "odd file structure":

>
Yes, Teensy makes much more use of the parameter substitution from boards.txt into platform.txt. If you're properly implementing Arduino's algorithms, platform.txt will resolve to the correct commands. But if you're hard-coding to the specific recipes Arduino publishes, then of course Teensy's very different recipes will seem rather foreign.
>
Three custom Tools menus are also used to define several important settings. You'll probably need to implement some way to select those 3 settings. The available settings are different for each board. You should probably also be aware I'm planning to add 3 more USB types later this year, with amazing USB functionality that's never been seen on any Arduino compatible board.
>
The Teensyduino installer also adds a couple minor patches to the IDE.
>
1: A recipe called "recipe.elfpatch.pattern" which is run if boards.txt defines "build.elfpatch". This is only needed on the 2.0 board when Disk(internal) is used. If you don't implement this, everything other than that 1 special USB option should work.
>
2: A special "{extra.time.local}" symbol is added to the build settings, which contains the "time_t" time the build was run. You'll need to substitute an integer for this name. If you use a fixed value, the only consequence is the optional RTC on Teensy 3.1 will not automatically set upon the first upload after a RTC battery removal.

sej7278, 2015-04-08:
the main problem with teensy3 support is that it seems to use avr as its architecture in paths when actually its arm.

its certainly works for 1.0.6 though, so that's your best bet for now until everything has settled down.