sudar/Arduino-Makefile #326: Arduino Makefile micro upload failure

kimogi, 2015-03-23:
Hi, i've ran into a problem with Arduino Makefile.

The code built fine with ARDUINO_TAG=micro but failed to exec 

```
make upload ARDUINO_PORT=/dev/ttyUSB0 MONITOR_PORT=/dev/ttyUSB0 
```
to Micriduino board with an output:

Connecting to programmer: .
Found programmer: Id = ""; type = �
    Software Version = .; Hardware Version = �.�
avrdude: error: buffered memory access not supported. Maybe it isn't
a butterfly/AVR109 but a AVR910 device?
/usr/share/arduino/Arduino.mk:1203: recipe for target 'do_upload' failed

Makefile
-------------------------------------------
```
BOARD_TAG = micro
ARDMK_DIR = /usr/share/arduino
ARDUINO_LIBS = SPI RF24 EEPROM
include /usr/share/arduino/Arduino.mk
```
test.ino
-----------------------------------------------
```
void setup (void) {
        Serial.begin(9600);
        Serial.println("Initializing ...");
}

void handleCmd () {
        char c;
        if (Serial.available()) {
                c = Serial.read();
                Serial.print("READ : "); Serial.println(c);
        }
}

void loop (void) {
        handleCmd();
}
```
Its uploaded fine compiled with BOARD_TAG=uno to Microduino though. But then Serial.read does not work.

sej7278, 2015-03-23:
it seems like the teensy3 it uses a modifed arduino ide, but there doesn't seem to be any source or documentation on their github, and no linux binaries.

it seems it uses a 32u4 so you may have more success building as an arduino leonardo rather than the uno.

sej7278, 2015-03-23:
yup, you're vastly over-simplifying things. first off you don't have an arduino micro you have a microduino (from some 3rd party) so using micro/uno as your BOARD_TAG won't work.

look at the boards.txt [here](https://github.com/wasdpkj/Microduino-IDE/blob/master/arduino-1.0.6-windows/hardware/Microduino/boards.txt) and install that with the bootloaders/cores/variants etc. and try to figure out which microduino you have - the one similar to an uno (core), the one similar to a leonardo (core usb) or the one similar to a mega (core+).

you may be able to get serial working by using the leonardo as your BOARD_TAG but i doubt it will work 100% as they won't have duplicated the pinout etc. you really need to install the files from microduino and set ALTERNATE_CORE.

for example, this is the makefile i use for the sparkfun pro micro (leonardo clone):

```
MONITOR_PORT = /dev/ttyACM0
BOARD_TAG = promicro16
ALTERNATE_CORE = promicro
BOOTLOADER_PARENT = $(HOME)/sketchbook/hardware/promicro/bootloaders
BOOTLOADER_PATH = caterina
BOOTLOADER_FILE = Caterina-promicro16.hex
BOARDS_TXT = $(HOME)/sketchbook/hardware/promicro/boards.txt
include /usr/share/arduino/Arduino.mk
```

and the promicro files in $(HOME)/sketchbook/hardware/promicro/ are:

```
├── boards.txt
├── bootloaders
│   └── caterina
│       ├── build.txt
│       ├── Caterina.c
│       ├── Caterina-fio.hex
│       ├── Caterina.h
│       ├── Caterina.hex
│       ├── Caterina-lilypadusb.hex
│       ├── Caterina-makey.hex
│       ├── Caterina-minibench.hex
│       ├── Caterina-promicro16.hex
│       ├── Caterina-promicro8.hex
│       ├── Caterina-wiflyin.hex
│       ├── Descriptors.c
│       ├── Descriptors.h
│       ├── Makefile
│       ├── program.txt
│       └── Readme.txt
└── variants
    └── promicro
        └── pins_arduino.h
```

sudar, 2015-04-07:
@kimogi

Did you try the above suggestion?

kimogi, 2015-04-07:
nope. I've switch to nano but it didn't work ether.
I'm using another console tool now

sej7278, 2015-04-07:
well if it builds and uploads to the nano ok, then its your code that's not working, not the makefile.

http://arduino.cc/en/Serial/read

reading and writing at the same time is difficult, you'd need two different screen sessions open (or the ide serial monitor and screen)

closed 2015-04-08 by sudar

kimogi, 2015-04-08:
It compiles and uploads only under Uno but not under Nano.

sej7278, 2015-04-08:
For nano you'd use this with a 1.0 environment:

```
BOARD_TAG = nano328
MONITOR_PORT = /dev/ttyUSB0
include /usr/share/arduino/Arduino.mk
```

Or a 1.6 setup:

```
BOARD_TAG = nano
BOARD_SUB = atmega328
ARDUINO_PORT = /dev/ttyACM0
include /usr/share/arduino/Arduino.mk
```

That's if you're actually using a Nano board, rather than trying to program the Microduino with a Nano core, which obviously won't work.