sudar/Arduino-Makefile #327: avr-g++: error: missing argument to '-mmcu='

briandw, 2015-03-24:
I'm trying to build the ATtinyBlink.ino example that is supplied but I get an error:

Here is the make file:
#CPPFLAGS = -mmcu=avrtiny #I tried adding this flag but it doesn’t replace the other -mmcu
# Arduino Make file. Refer to https://github.com/sudar/Arduino-Makefile

ARDUINO_DIR = /Applications/Arduino.app/Contents/Resources/Java
ARDMK_DIR = /Users/brian/opensource/Arduino-Makefile
AVR_TOOLS_DIR = /usr/local/CrossPack-AVR-20131216

# if you have placed the alternate core in your sketchbook directory, then you can just mention the core name alone.
ALTERNATE_CORE = attiny
# If not, you might have to include the full path.
#ALTERNATE_CORE_PATH = /Users/brian/Documents/Arduino/hardware/attiny

BOARD_TAG    = attiny85-8
ISP_PORT = /dev/ttyACM*

include $(ARDMK_DIR)/Arduino.mk

# !!! Important. You have to use make ispload to upload when using ISP programmer

Here is the output:

Blink % make
-------------------------
Arduino.mk Configuration:
- [AUTODETECTED]       CURRENT_OS = MAC 
- [USER]               ARDUINO_DIR = /Applications/Arduino.app/Contents/Resources/Java 
- [USER]               ARDMK_DIR = /Users/brian/opensource/Arduino-Makefile 
- [AUTODETECTED]       ARDUINO_VERSION = 105 
- [DEFAULT]            ARCHITECTURE =  
- [DEFAULT]            VENDOR = arduino 
- [DEFAULT]            ARDUINO_SKETCHBOOK = /Users/brian/sketchbook 
- [USER]               AVR_TOOLS_DIR = /usr/local/CrossPack-AVR-20131216 
- [COMPUTED]           ARDUINO_LIB_PATH = /Applications/Arduino.app/Contents/Resources/Java/libraries (from ARDUINO_DIR)
- [USER]               ALTERNATE_CORE = attiny 
- [COMPUTED]           ALTERNATE_CORE_PATH = /Users/brian/sketchbook/hardware/attiny  (from ARDUINO_SKETCHBOOK and ALTERNATE_CORE)
- [COMPUTED]           ARDUINO_VAR_PATH = /Users/brian/sketchbook/hardware/attiny/variants (from ALTERNATE_CORE_PATH)
- [COMPUTED]           BOARDS_TXT = /Users/brian/sketchbook/hardware/attiny/boards.txt (from ALTERNATE_CORE_PATH)
- [DEFAULT]            USER_LIB_PATH = /Users/brian/sketchbook/libraries (in user sketchbook)
- [DEFAULT]            PRE_BUILD_HOOK = pre-build-hook.sh 
- [USER]               BOARD_TAG = attiny85-8 
- [COMPUTED]           CORE =  (from build.core)
- [COMPUTED]           OBJDIR = build-attiny85-8 (from BOARD_TAG)
- [DEFAULT]            ARDUINO_CORE_PATH = /Applications/Arduino.app/Contents/Resources/Java/hardware/arduino//cores/arduino 
- [ASSUMED]            MONITOR_BAUDRATE = 9600 
- [DEFAULT]            OPTIMIZATION_LEVEL = s 
- [DEFAULT]            MCU_FLAG_NAME = mmcu 
- [DEFAULT]            CFLAGS_STD =  
- [DEFAULT]            CXXFLAGS_STD =  
- [AUTODETECTED]       DEVICE_PATH =  
- [DEFAULT]            FORCE_MONITOR_PORT =  
- [AUTODETECTED]       Size utility: AVR-aware for enhanced output
- [COMPUTED]           BOOTLOADER_PARENT = /Applications/Arduino.app/Contents/Resources/Java/hardware/arduino//bootloaders (from ARDUINO_DIR)
- [COMPUTED]           ARDMK_VERSION = 1.3.4 
- [COMPUTED]           CC_VERSION = 4.8.1 (avr-gcc)
-------------------------
mkdir -p build-attiny85-8
/usr/local/CrossPack-AVR-20131216/bin/avr-g++ -x c++ -include Arduino.h -MMD -c -mmcu=avrtiny -mmcu= -DF_CPU= -DARDUINO=105  -D__PROG_TYPES_COMPAT__ -I/Applications/Arduino.app/Contents/Resources/Java/hardware/arduino//cores/arduino -I/Users/brian/sketchbook/hardware/attiny/variants/    -Wall -ffunction-sections -fdata-sections -Os -fno-exceptions  ATtinyBlink.ino -o build-attiny85-8/ATtinyBlink.o
avr-g++: error: missing argument to '-mmcu='
make: *** [build-attiny85-8/ATtinyBlink.o] Error 1

Thanks

sej7278, 2015-03-24:
its because you haven't put the alternate core in your sketchbook as per the instructions.

```
- [DEFAULT]            ARDUINO_SKETCHBOOK = /Users/brian/sketchbook 
- [COMPUTED]           ALTERNATE_CORE_PATH = /Users/brian/sketchbook/hardware/attiny  (from ARDUINO_SKETCHBOOK and ALTERNATE_CORE)
#ALTERNATE_CORE_PATH = /Users/brian/Documents/Arduino/hardware/attiny
```

so either move the attiny directory from /Users/brian/Documents/Arduino/hardware/ to /Users/brian/sketchbook/hardware/ or define ARDUINO_SKETCHBOOK as /Users/brian/Documents/Arduino or uncomment your ALTERNATE_CORE_PATH line

briandw, 2015-03-25:
Thanks for the help but I did put the attiny folder in /Users/brian/sketchbook/hardware/
I have also tried specifying ARDUINO_SKETCHBOOK = /Users/brian/Documents/Arduino with the hardware directory there.

I've never run the Arduino.app since I don't want to install Java, hence the usefulness of this project.
It's still the same result.

I've made sure that there the correct files are at every dir listed, but still no good.

Arduino.mk Configuration:
- [AUTODETECTED]       CURRENT_OS = MAC 
- [USER]               ARDUINO_DIR = /Applications/Arduino.app/Contents/Resources/Java 
- [USER]               ARDMK_DIR = /Users/brian/opensource/Arduino-Makefile 
- [AUTODETECTED]       ARDUINO_VERSION = 105 
- [DEFAULT]            ARCHITECTURE =  
- [DEFAULT]            VENDOR = arduino 
- [USER]               ARDUINO_SKETCHBOOK = /Users/brian/Documents/Arduino 
- [USER]               AVR_TOOLS_DIR = /usr/local/CrossPack-AVR-20131216 
- [COMPUTED]           ARDUINO_LIB_PATH = /Applications/Arduino.app/Contents/Resources/Java/libraries (from ARDUINO_DIR)
- [COMPUTED]           ARDUINO_VAR_PATH = /Applications/Arduino.app/Contents/Resources/Java/hardware/arduino//variants (from ARDUINO_DIR)
- [COMPUTED]           BOARDS_TXT = /Applications/Arduino.app/Contents/Resources/Java/hardware/arduino//boards.txt (from ARDUINO_DIR)
- [DEFAULT]            USER_LIB_PATH = /Users/brian/Documents/Arduino/libraries (in user sketchbook)
- [DEFAULT]            PRE_BUILD_HOOK = pre-build-hook.sh 
- [USER]               BOARD_TAG = attiny85-8 
- [COMPUTED]           CORE =  (from build.core)
- [COMPUTED]           OBJDIR = build-attiny85-8 (from BOARD_TAG)
- [DEFAULT]            ARDUINO_CORE_PATH = /Applications/Arduino.app/Contents/Resources/Java/hardware/arduino//cores/arduino 
- [ASSUMED]            MONITOR_BAUDRATE = 9600 
- [DEFAULT]            OPTIMIZATION_LEVEL = s 
- [DEFAULT]            MCU_FLAG_NAME = mmcu 
- [DEFAULT]            CFLAGS_STD =  
- [DEFAULT]            CXXFLAGS_STD =  
- [AUTODETECTED]       DEVICE_PATH =  
- [DEFAULT]            FORCE_MONITOR_PORT =  
- [AUTODETECTED]       Size utility: AVR-aware for enhanced output
- [COMPUTED]           BOOTLOADER_PARENT = /Applications/Arduino.app/Contents/Resources/Java/hardware/arduino//bootloaders (from ARDUINO_DIR)
- [COMPUTED]           ARDMK_VERSION = 1.3.4 
- [COMPUTED]           CC_VERSION = 4.8.1 (avr-gcc)
-------------------------
mkdir -p build-attiny85-8
/usr/local/CrossPack-AVR-20131216/bin/avr-g++ -x c++ -include Arduino.h -MMD -c -mmcu= -DF_CPU= -DARDUINO=105  -D__PROG_TYPES_COMPAT__ -I/Applications/Arduino.app/Contents/Resources/Java/hardware/arduino//cores/arduino -I/Applications/Arduino.app/Contents/Resources/Java/hardware/arduino//variants/    -Wall -ffunction-sections -fdata-sections -Os -fno-exceptions  ATtinyBlink.ino -o build-attiny85-8/ATtinyBlink.o
avr-g++: error: missing argument to '-mmcu='
make: *** [build-attiny85-8/ATtinyBlink.o] Error 1

Thanks

sej7278, 2015-03-25:
you've not specified the ALTERNATE_CORE this time though

```
- [COMPUTED]           CORE =  (from build.core)
```

sej7278, 2015-03-28:
did you manage to retry this?

Scottmey, 2015-04-01:
I'm having a similiar problem using the Mega2560 & the blink example:

Makefile:
`
ARDMK_DIR  			= /usr/local/opt/arduino-mk

ARDUINO_DIR       	= /Applications/Arduino.app/Contents/Resources/Java
ARDUINO_CORE_PATH 	= /Applications/Arduino.app/Contents/Resources/Java/hardware/arduino/avr/cores/arduino

ARDUINO_VAR_PATH 	= /Applications/Arduino.app/Contents/Resources/Java/hardware/arduino/avr/variants

ARDUINO_SKETCHBOOK 	= /Users/example/Documents/Arduino

AVR_TOOLS_DIR 		= /Applications/Arduino.app/Contents/Resources/Java/hardware/tools/avr

BOARD_TAG    		= mega
BOARDS_TXT 			= /Applications/Arduino.app/Contents/Resources/Java/hardware/arduino/avr/boards.txt

include /usr/local/opt/arduino-mk/Arduino.mk
`

blink.ino
`
/*
  Blink
  Turns on an LED on for one second, then off for one second, repeatedly.
  This example code is in the public domain.
 */

void setup() {                
	// initialize the digital pin as an output.
	// Pin 13 has an LED connected on most Arduino boards:
	pinMode(13, OUTPUT);     
}

void loop() {
	digitalWrite(13, HIGH);   // set the LED on
	delay(1000);              // wait for a second
	digitalWrite(13, LOW);    // set the LED off
	delay(1000);              // wait for a second
}
`

Output:
`
-------------------------
Arduino.mk Configuration:
- [AUTODETECTED]       CURRENT_OS = MAC
- [USER]               ARDMK_DIR = /usr/local/opt/arduino-mk
- [USER]               ARDUINO_DIR = /Applications/Arduino.app/Contents/Resources/Java
- [AUTODETECTED]       ARDUINO_VERSION = 161
- [USER]               ARDUINO_SKETCHBOOK = /Users/example/Documents/Arduino
- [USER]               AVR_TOOLS_DIR = /Applications/Arduino.app/Contents/Resources/Java/hardware/tools/avr
- [COMPUTED]           ARDUINO_LIB_PATH = /Applications/Arduino.app/Contents/Resources/Java/libraries (from ARDUINO_DIR)
- [USER]               ARDUINO_CORE_PATH = /Applications/Arduino.app/Contents/Resources/Java/hardware/arduino/avr/cores/arduino
- [USER]               ARDUINO_VAR_PATH = /Applications/Arduino.app/Contents/Resources/Java/hardware/arduino/avr/variants
- [USER]               BOARDS_TXT = /Applications/Arduino.app/Contents/Resources/Java/hardware/arduino/avr/boards.txt
- [DEFAULT]            USER_LIB_PATH = /Users/example/Documents/Arduino/libraries (in user sketchbook)
- [DEFAULT]            PRE_BUILD_HOOK = pre-build-hook.sh
- [USER]               BOARD_TAG = mega
- [COMPUTED]           OBJDIR = build-mega (from BOARD_TAG)
- [ASSUMED]            MONITOR_BAUDRATE = 9600
- [DEFAULT]            OPTIMIZATION_LEVEL = s
- [DEFAULT]            MCU_FLAG_NAME = mmcu
- [DEFAULT]            CFLAGS_STD = -std=gnu99
- [AUTODETECTED]       DEVICE_PATH = /dev/tty.usbmodem1421
- [AUTODETECTED]       Size utility: AVR-aware for enhanced output
- [COMPUTED]           BOOTLOADER_PARENT = /Applications/Arduino.app/Contents/Resources/Java/hardware/arduino/bootloaders (from ARDUINO_DIR)
-------------------------
mkdir -p build-mega
/Applications/Arduino.app/Contents/Resources/Java/hardware/tools/avr/bin/avr-g++ -x c++ -include Arduino.h -MMD -c -mmcu= -DF_CPU=16000000L -DARDUINO=161 -D__PROG_TYPES_COMPAT__ -I. -I/Applications/Arduino.app/Contents/Resources/Java/hardware/arduino/avr/cores/arduino -I/Applications/Arduino.app/Contents/Resources/Java/hardware/arduino/avr/variants/mega   -Wall -ffunction-sections -fdata-sections -Os -fno-exceptions   blink.ino -o build-mega/blink.o
avr-g++: error: missing argument to '-mmcu='
make: *** [build-mega/blink.o] Error 1
`

sej7278, 2015-04-01:
with 1.6.1 you need to define BOARD_SUB as it has a submenu cpu in the IDE.

this works for me:

```
ARDUINO_DIR = $(HOME)/arduino-1.6.1
BOARD_TAG = mega
BOARD_SUB = atmega2560
MONITOR_PORT = /dev/ttyACM0

include $(HOME)/my_git/Arduino-Makefile/Arduino.mk
```

Scottmey, 2015-04-01:
Thanks, I've given that a shot and receive the same error.

My currently installed version is 1.6.1, I'm running on OSX. I'll spend sometime this afternoon digging into this a bit more.

sej7278, 2015-04-01:
bizarre, as i also got this working on osx. i'll have a look what i did later.

can you post your output when you added BOARD_SUB?

you are using the latest version from git aren't you?

you may be setting too many variables in your makefile, try just what i've put above (change for osx) and let the rest be auto-detected, you may be setting them the 1.0.x way

alternatively try arduino 1.0.6, there's really no reason to use 1.5/1.6 for a mega2560

Scottmey, 2015-04-01:
Sorry, I should have mentioned I used homebrew for installation.

``` ### 1.3.4 (2014-07-12) ```

I've tried a more barebones Makefile w/ the same results.

The current output:
```
-------------------------
Arduino.mk Configuration:
- [AUTODETECTED]       CURRENT_OS = MAC
- [USER]               ARDMK_DIR = /usr/local/opt/arduino-mk
- [USER]               ARDUINO_DIR = /Applications/Arduino.app
- [DEFAULT]            ARDUINO_VERSION = 100
- [USER]               ARDUINO_SKETCHBOOK = /Users/scottmeyers/Documents/Arduino
- [USER]               AVR_TOOLS_DIR = /Applications/Arduino.app/Contents/Resources/Java/hardware/tools/avr
- [COMPUTED]           ARDUINO_LIB_PATH = /Applications/Arduino.app/libraries (from ARDUINO_DIR)
- [USER]               ARDUINO_CORE_PATH = /Applications/Arduino.app/Contents/Resources/Java/hardware/arduino/avr/cores/arduino
- [USER]               ARDUINO_VAR_PATH = /Applications/Arduino.app/Contents/Resources/Java/hardware/arduino/avr/variants
- [USER]               BOARDS_TXT = /Applications/Arduino.app/Contents/Resources/Java/hardware/arduino/avr/boards.txt
- [DEFAULT]            USER_LIB_PATH = /Users/scottmeyers/Documents/Arduino/libraries (in user sketchbook)
- [DEFAULT]            PRE_BUILD_HOOK = pre-build-hook.sh
- [USER]               BOARD_TAG = mega
- [COMPUTED]           OBJDIR = build-mega (from BOARD_TAG)
- [ASSUMED]            MONITOR_BAUDRATE = 9600
- [DEFAULT]            OPTIMIZATION_LEVEL = s
- [DEFAULT]            MCU_FLAG_NAME = mmcu
- [DEFAULT]            CFLAGS_STD = -std=gnu99
- [AUTODETECTED]       DEVICE_PATH = /dev/tty.usbmodem1421
- [AUTODETECTED]       Size utility: AVR-aware for enhanced output
- [COMPUTED]           BOOTLOADER_PARENT = /Applications/Arduino.app/hardware/arduino/bootloaders (from ARDUINO_DIR)
-------------------------
mkdir -p build-mega
/Applications/Arduino.app/Contents/Resources/Java/hardware/tools/avr/bin/avr-g++ -x c++ -include Arduino.h -MMD -c -mmcu= -DF_CPU=16000000L -DARDUINO=100 -D__PROG_TYPES_COMPAT__ -I. -I/Applications/Arduino.app/Contents/Resources/Java/hardware/arduino/avr/cores/arduino -I/Applications/Arduino.app/Contents/Resources/Java/hardware/arduino/avr/variants/mega   -Wall -ffunction-sections -fdata-sections -Os -fno-exceptions   blink.ino -o build-mega/blink.o
avr-g++: error: missing argument to '-mmcu='
make: *** [build-mega/blink.o] Error 1
```

sej7278, 2015-04-01:
something's changed as the 2nd output shows it thinks its version 1.0.0 and you're not setting BOARD_SUB

ah wait are you using homebrew to install arduino-mk? you need to get the latest revision from git (as of about 2 days ago) or go back to arduino 1.0.6

sej7278, 2015-04-01:
with IDE 1.6.1 this (Blink) works for me on Yosemite:

```make
ARDUINO_DIR=/Applications/Arduino.app/Contents/Resources/Java/
BOARD_TAG = mega
BOARD_SUB = atmega2560

include ../../Arduino.mk
```

```
Arduino.mk Configuration:
- [AUTODETECTED]       CURRENT_OS = MAC 
- [USER]               ARDUINO_DIR = /Applications/Arduino.app/Contents/Resources/Java/ 
- [COMPUTED]           ARDMK_DIR = /Users/simon/arduino-mk/Arduino-Makefile (relative to Common.mk)
- [AUTODETECTED]       ARDUINO_VERSION = 161 
- [DEFAULT]            ARCHITECTURE = avr 
- [DEFAULT]            VENDOR = arduino 
- [AUTODETECTED]       ARDUINO_PREFERENCES_PATH = /Users/simon/Library/Arduino15/preferences.txt 
- [AUTODETECTED]       ARDUINO_SKETCHBOOK = /Users/simon/Documents/Arduino (from arduino preferences file)
- [BUNDLED]            AVR_TOOLS_DIR = /Applications/Arduino.app/Contents/Resources/Java//hardware/tools/avr (in Arduino distribution)
- [COMPUTED]           ARDUINO_LIB_PATH = /Applications/Arduino.app/Contents/Resources/Java//libraries (from ARDUINO_DIR)
- [COMPUTED]           ARDUINO_PLATFORM_LIB_PATH = /Applications/Arduino.app/Contents/Resources/Java//hardware/arduino/avr/libraries (from ARDUINO_DIR)
- [COMPUTED]           ARDUINO_VAR_PATH = /Applications/Arduino.app/Contents/Resources/Java//hardware/arduino/avr/variants (from ARDUINO_DIR)
- [COMPUTED]           BOARDS_TXT = /Applications/Arduino.app/Contents/Resources/Java//hardware/arduino/avr/boards.txt (from ARDUINO_DIR)
- [DEFAULT]            USER_LIB_PATH = /Users/simon/Documents/Arduino/libraries (in user sketchbook)
- [DEFAULT]            PRE_BUILD_HOOK = pre-build-hook.sh 
- [USER]               BOARD_SUB = atmega2560 
- [USER]               BOARD_TAG = mega 
- [COMPUTED]           CORE = arduino (from build.core)
- [COMPUTED]           VARIANT = mega (from build.variant)
- [COMPUTED]           OBJDIR = build-mega-atmega2560 (from BOARD_TAG)
- [COMPUTED]           ARDUINO_CORE_PATH = /Applications/Arduino.app/Contents/Resources/Java//hardware/arduino/avr/cores/arduino (from ARDUINO_DIR, BOARD_TAG and boards.txt)
- [ASSUMED]            MONITOR_BAUDRATE = 9600 
- [DEFAULT]            OPTIMIZATION_LEVEL = s 
- [DEFAULT]            MCU_FLAG_NAME = mmcu 
- [DEFAULT]            CFLAGS_STD =  
- [DEFAULT]            CXXFLAGS_STD =  
- [AUTODETECTED]       DEVICE_PATH =  
- [DEFAULT]            FORCE_MONITOR_PORT =  
- [AUTODETECTED]       Size utility: AVR-aware for enhanced output
- [COMPUTED]           BOOTLOADER_PARENT = /Applications/Arduino.app/Contents/Resources/Java//hardware/arduino/avr/bootloaders (from ARDUINO_DIR)
- [COMPUTED]           ARDMK_VERSION = 1.3.4 
- [COMPUTED]           CC_VERSION = 4.8.1 (avr-gcc)
```

Scottmey, 2015-04-01:
Thanks, you're correct the issue was w/ homebrew. I removed the homebrew package, and pulled the latest down directly from git and everything is now working as expected.

I appreciate your support. Cheers.

sej7278, 2015-04-01:
brilliant!

i don't know how the homebrew works, i guess it only works when we tag a release or something, not from git HEAD. we'll have to look into it.

sudar, 2015-04-07:
@briandw I am closing this for now because of inactivity.

Feel free to re-open it if needed.

closed 2015-04-07 by sudar