sudar/Arduino-Makefile #328: Provide default support for Arduino 1.5.x

sudar, 2015-03-25:
Now that Arduino IDE 1.5.x is out of beta (it is actually called 1.6.1) can we start to provide default support for Arduino 1.5.x?

@sej7278 I know you already worked on it and there were some pending issues. Let's start to list them down so that we can create individual tasks to track them.

Also I think we can just make a final release and tag it before we make it only 1.5.x or above.

sej7278, 2015-03-25:
i don't know, its pretty hard to figure out what doesn't work. generally avr works, but we've not done much for arm or stuff that uses non-arduino cores like attiny, partially because the people providing alternate cores have only just started on 1.6 support too.

i think its about time we made a 1.3.5 release or something yes.

sudar, 2015-03-25:
Okay, so in that case, let's do it in stages.

- Make a release out of the current master.
- Start to provide support to only AVR first.
- Once AVR works, we can make a release.
- Start working on alternate cores that support 1.5.x
- After ironing out all issues, we can start the support for non-avr.

I think supporting non-avr will be lot of work since the entire toolchain is going to change. Also we would need non-avr Arduino's to test it. I don't own one yet.

sej7278, 2015-03-25:
hold off on making a release for today would you, i've got a couple more easy fixes for 1.5 support and library detection almost done

sudar, 2015-03-25:
Sure.

Let me know when you are done and we can release it.

sej7278, 2015-03-25:
see PR #329 although it looks like Travis-CI is dead

you'll need to run your version update script as i've not updated any of the version numbers in fedora.spec etc.

sudar, 2015-03-25:
Let's do the release by this weekend.

I am giving a talk about this project in Arduino Day celebrations at a local hackerspace this Saturday. Since I have to prepare a couple of examples I will be testing the makefile a bit over the next few days and if we don't find any issues then we can make a release this weekend.

sej7278, 2015-03-25:
sounds cool. i must look around here for some hackerspaces, they're generally the other end of the country.

it won't go anywhere near debian for now because of the jessie freeze, but i can make a package up ready to submit anyway.

sej7278, 2015-03-25:
just to be annoying, i've made another little PR #330 which seems to fix ALTERNATE_CORES under 1.6.1, so now we can build for ATtiny85's too (and i remembered HISTORY.md)

that's it now, no more before the weekend.

sudar, 2015-03-26:
No problem at all and thanks for doing it.

sej7278, 2015-03-26:
How do you feel about one really really last PR #331 before the weekend?

Basically adds the BOARD_SUB to BOARD_TAG when defining OBJDIR in 1.5+, otherwise you end up with mega1280 and mega2560 using the same build-mega directory, or nano328p and nano168 using the same build-nano directory.

sudar, 2015-03-27:
Go for it.

Making a release this weekend is just an arbitrary deadline that I choose. We can delay it if needed or if you feel that we can get a couple of more fixes.

sej7278, 2015-03-27:
ok that's all now, make the release, i've got no more fixes in mind.

sej7278, 2015-04-04:
i wonder if we should say something about only supporting arduino.cc builds and not arduino.org ?

arduino.org has just released 1.7.0 of their IDE and they also seem to have older builds called 1.5.8.3 and 1.0.6.2, which seem to be designed to always be newer than arduino.cc's versions

arduino.org only seems to have the arduino zero, which is an arm sam board, so we don't support it anyway.

sudar, 2015-04-05:
Yes definitely we should add a note and I believe we have two reasons for that.

- As you mentioned we don't support Arduino Zero and that's the main reason.
- This one is kind of subjective. Even though I have not heard the story from both sides, I still kind of lean towards arduino.cc and not arduino.org. Personally I got into electronics only because of Arduino and I feel we should support them during their difficult times.

As I said, the second reason is completely subjective. What is your opinion on this?

sej7278, 2015-04-05:
I'd rather not get into the politics of it, but it definitely seems that .org is the underhanded side of the split, with .cc being the true Arduino.

I doubt anyone much would use .org, but it would at least save us some support requests if we plain say we don't support their IDE or hardware.

sudar, 2015-04-05:
> I'd rather not get into the politics of it

I completely agree with you.

I add a note about it in README under Limitations. Feel free to change the language if needed.

ladislas, 2015-04-05:
> I doubt anyone much would use .org, but it would at least save us some support requests if we plain say we don't support their IDE or hardware.

Agreed, saying we don't support it will protect us from unwanted request. If we change our mind in the future, we'll just have to change the README ;)

Did not know about .org before reading this thread, it seems it's not even very clear for them... [To the makers](http://arduino.org/blog/1-the-new-blog/to-the-makers).

I'll dig into this, but as @sudar said, I also lean towards .cc :)

sudar, 2015-04-07:
@sej7278 

Do you still have anything else in mind for this release? Can we go ahead and make the release?

sej7278, 2015-04-07:
nothing more from me, make the release

sudar, 2015-04-07:
Just released 1.5.

@sej7278 

Now that we support v1.5.x (1.6.x actually) can we change the following text in Readme?

```
Support for Arduino Software versions 1.0.x as well as 0.x. Support for Arduino 1.5.x is still work in progress
```

sej7278, 2015-04-07:
Ah yes, I thought we'd already done that.

commit 0cc6faf1 (2015-04-08): Update README about the version of Arduino that are supported

Fix #328
closed 2015-04-08 by sudar