sudar/Arduino-Makefile #332: New to the whole thing

RubenvWyk, 2015-03-27:
Hi All,

Currently I have issues with my Pi-Arduino setup. Hear me out, as it might look at first glance that this issue is meant for another forum :)

If I use the Arduino IDE to try and upload new code (for some reason this started today), I get these errors:

avrdude: Send: 0 [30] [20]
avrdude: Send: 0 [30] [20]
avrdude: Send: 0 [30] [20]
avrdude: Recv: { [7b]
avrdude: stk500_getsync(): not in sync: resp=0x7b

So anyway, I can't get it to communicate properly and was wondering if Arduino-Makefile might solve this? I have previous experience with Renesas chips and the IDE that causes problems... so I thought that maybe this might be the same issue...

The reason I am asking this without actually trying it out first, is because I have absolutely no idea how to get this Arduino-Makefile to work :D the tutorials seem to be written for peeps that know what is going on hehe

Of course, I believe the best way to learn is the hard way, but I need at least some bread crumbs :P 

Anyhoo, would Really appreciate any help :)

TIA

Regards,
Ruben

sej7278, 2015-03-27:
that's a pretty common error, certainly not pi-specific.

what kind of arduino are you using and how are you connecting - via usb or gpio or what? is it a standard arduino or one of these pi-duino or microduino 3rd party things?

sudar, 2015-04-07:
@RubenvWyk I am closing this for now because of inactivity.

Feel free to re-open it if needed with answers to the questions posted by @sej7278

closed 2015-04-07 by sudar