sudar/Arduino-Makefile #335: Compiling plain AVR C example is causing `sed` errors

sudar, 2015-03-28:
While trying to compile our [BlinkInAVRC](https://github.com/sudar/Arduino-Makefile/tree/master/examples/BlinkInAVRC) example, I am getting the following error in Mac.

@sej7278 Are you getting this error in Linux?

I think this is because of the small difference between sed in Linux and Mac.

```
$ make
-------------------------
Arduino.mk Configuration:
- [AUTODETECTED]       CURRENT_OS = MAC 
- [USER]               ARDUINO_DIR = /Users/sudar/Applications/Arduino.app/Contents/Resources/Java/ 
- [USER]               ARDMK_DIR = /Users/sudar/Dropbox/code/Arduino-Makefile 
- [AUTODETECTED]       ARDUINO_VERSION = 106 
- [DEFAULT]            ARCHITECTURE =  
- [DEFAULT]            VENDOR = arduino 
- [AUTODETECTED]       ARDUINO_PREFERENCES_PATH = /Users/sudar/Library/Arduino/preferences.txt 
- [AUTODETECTED]       ARDUINO_SKETCHBOOK = /Users/sudar/Dropbox/code/arduino-sketches (from arduino preferences file)
- [BUNDLED]            AVR_TOOLS_DIR = /Users/sudar/Applications/Arduino.app/Contents/Resources/Java//hardware/tools/avr (in Arduino distribution)
- [COMPUTED]           ARDUINO_LIB_PATH = /Users/sudar/Applications/Arduino.app/Contents/Resources/Java//libraries (from ARDUINO_DIR)
- [COMPUTED]           ARDUINO_VAR_PATH = /Users/sudar/Applications/Arduino.app/Contents/Resources/Java//hardware/arduino//variants (from ARDUINO_DIR)
- [COMPUTED]           BOARDS_TXT = /Users/sudar/Applications/Arduino.app/Contents/Resources/Java//hardware/arduino//boards.txt (from ARDUINO_DIR)
- [DEFAULT]            USER_LIB_PATH = /Users/sudar/Dropbox/code/arduino-sketches/libraries (in user sketchbook)
- [DEFAULT]            PRE_BUILD_HOOK = pre-build-hook.sh 
- [USER]               BOARD_TAG = atmega16 
- [COMPUTED]           OBJDIR = build-atmega16 (from BOARD_TAG)
- [DEFAULT]            ARDUINO_CORE_PATH = /Users/sudar/Applications/Arduino.app/Contents/Resources/Java//hardware/arduino//cores/arduino 
- [MANUAL]             NO_CORE set so core library will not be built
sed: RE error: illegal byte sequence
sed: RE error: illegal byte sequence
sed: RE error: illegal byte sequence
sed: RE error: illegal byte sequence
sed: RE error: illegal byte sequence
sed: RE error: illegal byte sequence
sed: RE error: illegal byte sequence
sed: RE error: illegal byte sequence
sed: RE error: illegal byte sequence
sed: RE error: illegal byte sequence
sed: RE error: illegal byte sequence
sed: RE error: illegal byte sequence
- [DEFAULT]            OPTIMIZATION_LEVEL = s 
- [DEFAULT]            MCU_FLAG_NAME = mmcu 
- [DEFAULT]            CFLAGS_STD =  
- [DEFAULT]            CXXFLAGS_STD =  
- [AUTODETECTED]       DEVICE_PATH =  
- [DEFAULT]            FORCE_MONITOR_PORT =  
- [AUTODETECTED]       Size utility: AVR-aware for enhanced output
sed: RE error: illegal byte sequence
sed: RE error: illegal byte sequence
sed: RE error: illegal byte sequence
sed: RE error: illegal byte sequence
- [COMPUTED]           BOOTLOADER_PARENT = /Users/sudar/Applications/Arduino.app/Contents/Resources/Java//hardware/arduino//bootloaders (from ARDUINO_DIR)
- [COMPUTED]           ARDMK_VERSION = 1.3.4 
- [COMPUTED]           CC_VERSION = 4.3.2 (avr-gcc)
-------------------------
mkdir -p build-atmega16
/Users/sudar/Applications/Arduino.app/Contents/Resources/Java//hardware/tools/avr/bin/avr-gcc -MMD -c -mmcu=atmega16 -DF_CPU=8000000L -DARDUINO=106  -D__PROG_TYPES_COMPAT__ -I/Users/sudar/Applications/Arduino.app/Contents/Resources/Java//hardware/arduino//cores/arduino -I/Users/sudar/Applications/Arduino.app/Contents/Resources/Java//hardware/arduino//variants/    -Wall -ffunction-sections -fdata-sections -Os  blink.c -o build-atmega16/blink.o
/Users/sudar/Applications/Arduino.app/Contents/Resources/Java//hardware/tools/avr/bin/avr-ar rcs build-atmega16/libcore.a        
/Users/sudar/Applications/Arduino.app/Contents/Resources/Java//hardware/tools/avr/bin/avr-gcc -mmcu=atmega16 -Wl,--gc-sections -Os -o build-atmega16/BlinkInAVRC.elf build-atmega16/blink.o build-atmega16/libcore.a  -lc -lm
/Users/sudar/Applications/Arduino.app/Contents/Resources/Java//hardware/tools/avr/bin/avr-objcopy -j .eeprom --set-section-flags=.eeprom='alloc,load' \
		--change-section-lma .eeprom=0 -O ihex build-atmega16/BlinkInAVRC.elf build-atmega16/BlinkInAVRC.eep
/Users/sudar/Applications/Arduino.app/Contents/Resources/Java//hardware/tools/avr/bin/avr-objcopy: --change-section-lma .eeprom=0x00000000 never used
/Users/sudar/Applications/Arduino.app/Contents/Resources/Java//hardware/tools/avr/bin/avr-objcopy -O ihex -R .eeprom build-atmega16/BlinkInAVRC.elf build-atmega16/BlinkInAVRC.hex

/Users/sudar/Applications/Arduino.app/Contents/Resources/Java//hardware/tools/avr/bin/avr-size --mcu=atmega16 -C --format=avr build-atmega16/BlinkInAVRC.elf
AVR Memory Usage
----------------
Device: atmega16

Program:     214 bytes (1.3% Full)
(.text + .data + .bootloader)

Data:          0 bytes (0.0% Full)
(.data + .bss + .noinit)

Maximum flash memory of atmega16 is not specified. Make sure the size of build-atmega16/BlinkInAVRC.hex is less than atmega16's flash memory
```

sudar, 2015-03-28:
Okay, I found the line where the error is happening.

```
ifndef ARDUINO_LIBS
    # automatically determine included libraries
    ARDUINO_LIBS += $(filter $(notdir $(wildcard $(ARDUINO_DIR)/libraries/*)), \
        $(shell sed -ne 's/^ *\# *include *[<\"]\(.*\)\.h[>\"]/\1/p' $(LOCAL_SRCS)))
    ARDUINO_LIBS += $(filter $(notdir $(wildcard $(ARDUINO_SKETCHBOOK)/libraries/*)), \
        $(shell sed -ne 's/^ *\# *include *[<\"]\(.*\)\.h[>\"]/\1/p' $(LOCAL_SRCS)))
    ARDUINO_LIBS += $(filter $(notdir $(wildcard $(USER_LIB_PATH)/*)), \
        $(shell sed -ne 's/^ *\# *include *[<\"]\(.*\)\.h[>\"]/\1/p' $(LOCAL_SRCS)))
    ARDUINO_LIBS += $(filter $(notdir $(wildcard $(ARDUINO_PLATFORM_LIB_PATH)/*)), \
        $(shell sed -ne 's/^ *\# *include *[<\"]\(.*\)\.h[>\"]/\1/p' $(LOCAL_SRCS)))
endif
```

sej7278, 2015-03-28:
no i don't get the error in linux

sej7278, 2015-03-28:
Fixed in 2nd commit of PR #336 

how do you split commits into different PR's without making different branches?

closed 2015-03-28 by sej7278

sudar, 2015-03-29:
Thank you very much for figuring it out. I was breaking my head with the regular expression..