sudar/Arduino-Makefile #338: 1.6.2+ support

sej7278, 2015-03-30:
Just when you thought it was safe to go back in the water.......  1.6.2 has moved the directory structure around again!

Now everything is stored in your home directory rather than the system directories, and also linked to the IDE version, for example:

```
BOARDS_TXT = $(HOME)/.arduino15/packages/arduino/hardware/avr/1.6.3/boards.txt
AVR_TOOLS_DIR = $(HOME)/.arduino15/packages/arduino/tools/avr-gcc/4.8.1-arduino3
ARDUINO_CORE_PATH =$(HOME)/.arduino15/packages/arduino/hardware/avr/1.6.3/cores/arduino
ARDUINO_VAR_PATH = $(HOME)/.arduino15/packages/arduino/hardware/avr/1.6.3/variants
```

i've managed to get it working with some jiggery pokery, but this hasn't been tested much - we also need to wait to see what people do about alternate cores like teensy/arduino-tiny/attiny etc. and official documentation would be nice.

i'd also like a cleaner way to find the dotted version "1.6.3" and the old version "163" rather than using two routines.

so now we basically have to support:

1.0.x (done)
1.5.x - 1.6.1 (mostly done for avr)
1.6.2+ (started)

```diff
diff --git a/Arduino.mk b/Arduino.mk
index 768aaf6..036bc1b 100644
--- a/Arduino.mk
+++ b/Arduino.mk
@@ -288,6 +288,21 @@ else
     $(call show_config_variable,ARDUINO_VERSION,[USER])
 endif
 
+ifndef ARDUINO_VERSION_RAW
+    # dotted version of above for 1.6.2+ paths
+    VERSION_FILE := $(ARDUINO_DIR)/lib/version.txt
+    AUTO_ARDUINO_VERSION_RAW := $(shell [ -e $(VERSION_FILE) ] && cat $(VERSION_FILE))
+    ifdef AUTO_ARDUINO_VERSION_RAW
+        ARDUINO_VERSION_RAW = $(AUTO_ARDUINO_VERSION_RAW)
+        $(call show_config_variable,ARDUINO_VERSION_RAW,[AUTODETECTED])
+    else
+        ARDUINO_VERSION_RAW = 1.0.0
+        $(call show_config_variable,ARDUINO_VERSION_RAW,[DEFAULT])
+    endif
+else
+    $(call show_config_variable,ARDUINO_VERSION_RAW,[USER])
+endif
+
 ########################################################################
 # 1.5.x architecture - avr or sam for arduino vendor
 ifndef ARCHITECTURE
@@ -506,14 +521,24 @@ ifdef ALTERNATE_CORE_PATH
 else
 
     ifndef ARDUINO_VAR_PATH
-        ARDUINO_VAR_PATH  = $(ARDUINO_DIR)/hardware/$(VENDOR)/$(ARCHITECTURE)/variants
+        ifeq ($(shell expr $(ARDUINO_VERSION) '>' 161), 1)
+            # 1.6.2 and later
+            ARDUINO_VAR_PATH = $(HOME)/.arduino15/packages/$(VENDOR)/hardware/$(ARCHITECTURE)/$(ARDUINO_VERSION_RAW)/variants
+        else
+            # 1.0 to 1.6.1
+            ARDUINO_VAR_PATH  = $(ARDUINO_DIR)/hardware/$(VENDOR)/$(ARCHITECTURE)/variants
+        endif
         $(call show_config_variable,ARDUINO_VAR_PATH,[COMPUTED],(from ARDUINO_DIR))
     else
         $(call show_config_variable,ARDUINO_VAR_PATH,[USER])
     endif
 
     ifndef BOARDS_TXT
-        BOARDS_TXT  = $(ARDUINO_DIR)/hardware/$(VENDOR)/$(ARCHITECTURE)/boards.txt
+        ifeq ($(shell expr $(ARDUINO_VERSION) '>' 161), 1)
+            BOARDS_TXT  = $(HOME)/.arduino15/packages/$(VENDOR)/hardware/$(ARCHITECTURE)/$(ARDUINO_VERSION_RAW)/boards.txt
+        else
+            BOARDS_TXT  = $(ARDUINO_DIR)/hardware/$(VENDOR)/$(ARCHITECTURE)/boards.txt
+        endif
         $(call show_config_variable,BOARDS_TXT,[COMPUTED],(from ARDUINO_DIR))
     else
         $(call show_config_variable,BOARDS_TXT,[USER])
@@ -718,7 +743,12 @@ ifndef ARDUINO_CORE_PATH
     else
         ARDUINO_CORE_PATH = $(ALTERNATE_CORE_PATH)/cores/$(CORE)
         ifeq ($(wildcard $(ARDUINO_CORE_PATH)),)
-            ARDUINO_CORE_PATH = $(ARDUINO_DIR)/hardware/$(VENDOR)/$(ARCHITECTURE)/cores/$(CORE)
+            ifeq ($(shell expr $(ARDUINO_VERSION) '>' 161), 1)
+                # 1.6.2 and later
+                ARDUINO_CORE_PATH = $(HOME)/.arduino15/packages/$(VENDOR)/hardware/$(ARCHITECTURE)/$(ARDUINO_VERSION_RAW)/cores/$(VENDOR)
+            else
+                ARDUINO_CORE_PATH = $(ARDUINO_DIR)/hardware/$(VENDOR)/$(ARCHITECTURE)/cores/$(CORE)
+            endif
             $(call show_config_variable,ARDUINO_CORE_PATH,[COMPUTED],(from ARDUINO_DIR, BOARD_TAG and boards.txt))
         else
             $(call show_config_variable,ARDUINO_CORE_PATH,[COMPUTED],(from ALTERNATE_CORE_PATH, BOARD_TAG and boards.txt))
```

sej7278, 2015-03-30:
might be easier to make $(ARDUINO_DIR) into $(HOME)/.arduino15/packages/arduino

but i'm not sure if the "arduino" bit is supposed to refer to $(VENDOR) or if its just plain "arduino"

sej7278, 2015-03-30:
probably need to make ```$(HOME)/.arduino15``` into something portable, as it could be ```$(HOME)/Library/Arduino15``` on mac, maybe ```$(ARDUINO_PREFERENCES_PATH)/../```

sej7278, 2015-03-30:
$(AVR_TOOLS_DIR) is going to be interesting, as its got versionned directories and avrdude is separate to avr-gcc etc; oh and its in your $(HOME) directory again. 

linux packagers are going to hate this.

sudar, 2015-04-01:
Damn it! Not sure why Arduino developers are doing it.

Anyways do you think it would be possible for us to support all 3 variations?

sej7278, 2015-04-01:
well 1.6.1 is working now i'd say for avr. i've been trying it out with quite a few sketches, cores and platforms and can't break it, but you'll always find a user who can lol.

so next we need to get arm and 1.6.2 working. i think 1.6.2 is going to be easier than arm, its just annoying as we could have started working on arm support, but now its back to messing with directories and stuff. and everyone is going to have to change their cores again (e.g. teensy, attiny etc.)

really i think 1.6.2 is the work of one arduino developer who obviously has tested only on the mac and not bothered to consult anyone, certainly not the community. 1.6.2 was 120+ commits overnight presumably just for arduino day, and some reverts since as it didn't even load on linux and didn't compile on windows.

i've no idea why they even released 1.6.0 if they were going to change stuff this much, i guess they just wanted to get 1.5.8 out of beta.

just goes to show that the arduino guys know nothing about programming and don't care about opensource software or their community.

the alternative is forking the cores and libraries (and toolchain?) from 1.6.1, ditching the ide/java parts and merging it with our makefile to provide a complete stack, but then that's not going to help people who want to use the ide as well as the makefile.

paul (teensy) is in the same boat - just got teensyduino 1.6.1 working and now is back to square one again.

sej7278, 2015-04-01:
after that rant, it seems in today's nightly that will become 1.6.3, they've put the avr hardware stuff back and only user-installed hardware/toolchains (maybe sam/arm?) will be in the ~/.arduino15 directory.

so 1.6.3 works pretty much the same as 1.6.1 (without my changes above) so essentially we just need to call 1.6.2 the unsupported evil screw-up spawn of satan build.

in fact i just uploaded Blink to my mega2560 using arduino-mk HEAD unmodified with 1.6.3 (nightly) using:

```
ARDUINO_DIR = $(HOME)/arduino-nightly

BOARD_TAG = mega
BOARD_SUB = atmega2560
MONITOR_PORT = /dev/ttyACM0

include $(HOME)/Arduino-Makefile/Arduino.mk
```

sudar, 2015-04-01:
@sej7278 Thanks for digging into it and I agree with you that we can just ignore 1.6.2 and add a note about it in the readme.

Arduino as a company is undergoing some tough time and I think this happened because of that. It is really unfortunate though..

ladislas, 2015-04-02:
@sej7278 ahah! :) you edited your comment, I was going to respond that I totally agree with regarding the way they handle the development process of the ide... I'm in their mailing list and I have to say that each day it's getting worst and worst... no roadmap, no task assignments (you never know if someone is working on the same thing as you), [rather bad documentation](https://groups.google.com/a/arduino.cc/forum/#!searchin/developers/arduino$20usb$20documentation/developers/u6SMuOTQVsM/o0o7UhMiMC8J), and they don't care about power users at all. For them, Arduino is a beginner platform and must only be used this way. For the rest of us it's hopeless...

> the alternative is forking the cores and libraries (and toolchain?) from 1.6.1, ditching the ide/java parts and merging it with our makefile to provide a complete stack, but then that's not going to help people who want to use the ide as well as the makefile.

That's something that I've requested on the mailing list but they did not like it. [Here is the short thread](https://groups.google.com/a/arduino.cc/forum/#!topic/developers/9a1qfyQO6pU).

The problem is that their repo is now huge and even if we clone and just keep the parts that we need, we'll still have all the history. On the other hand, removing the history and starting fresh will make it difficult to keep up with upstream and merge the changes.

@sej7278 I don't know git well enough to know if it is possible to remove only the history related to certain files, keep it for others and still be able to merge changes made on the full repo into our new lightweight repository. Do you have any ideas?

> Arduino as a company is undergoing some tough time and I think this happened because of that. It is really unfortunate though..

@sudar could not agree more... :(

sej7278, 2015-04-02:
as 1.6.3 has been released and fixes the issues for avr at least, i'm going to close this for now.

closed 2015-04-02 by sej7278