sudar/Arduino-Makefile #343: Add support for non-avr boards (SAM)

sudar, 2015-04-06:
Now that Arduino 1.6.3 is working, we should start supporting SAM and non-avr boards.

zallesov, 2015-04-22:
Ah, that is why I can not see my Due board in make show_boards.
You should really do it. Please!

sej7278, 2015-05-20:
we also need to be able to look for boards installed using 1.6.4 boards manager - that are saved in ~/.arduino15/packages/

bremme, 2015-10-05:
It would be really nice to have support for the [ESP8266](https://github.com/esp8266/Arduino) boards as well. I'm quite comfotable with Arduino, C/C++ and Linux, but rather new to Makefile's. But if you think there's anything I can do to help implement this feature let me know.

sej7278, 2015-10-05:
i did try to get the esp8266 working, but failed - the xtensa toolchain is yet another platform, not arm.

bremme, 2015-10-05:
Okay, what would be needed to make it work? As I said I'm really a newbie to Makefile's (but I'm sure I can learn it:), so forgive me if I ask stupid questions. Thanks to the [ESP8266 Arduino project](https://github.com/esp8266/Arduino) a lot of the hard work is already done, or am I mistaken? I also found some others who already worked on an [ESP8266 Arduino Makefile](https://github.com/thunderace/Esp8266-Arduino-Makefile) perhaps that helps as well?

sej7278, 2015-10-05:
yes @thunderace's makefile seems to be inspired by Arduino-Makefile, would be nice to have it merged back in like Teensy.mk

thunderace, 2015-10-05:
My ESP8266 Arduino Makefile is based on a very old version of Arduino Makefile (0.10 17.ix.12 if i remember well) with some enhancements from sudar's makefile. I think it will be a little hard to merge.

sej7278, 2015-10-09:
@thunderace i see you're using:

```
ARDUINO_CORE ?= $(ARDUINO_HOME)/hardware/$(ARDUINO_VENDOR)/$(ARDUINO_ARCH)
```

so you aren't parsing the boards manager installation, so does this rely on you copying the esp8266 arduino core into ~/sketchbook/hardware/ and the whole Arduino installation into the same directory as your Makefile ?

thunderace, 2015-10-12:
@sej7278 
I think you don't have the last version of the repo : The Esp8266-Arduino-Makefile is an all-in-one repo, the esp8266 arduino core is a submodule.
Try to install it with this command git clone --recursive https://github.com/thunderace/Esp8266-Arduino-Makefile.git and execute the  install.sh script.
After that, take a look at the example/AdvancedWebServer/Makefile it work (for me).

sej7278, 2015-10-12:
yeah i mean that you are pulling in the arduino source as a submodule, not just using an already installed/compiled arduino and boards manager, its not what end users are going to use.

thunderace, 2015-10-13:
It's that i need as i only use headless linux remote machines (no GUI).
And i think it is the goal of a Makefile : build without GUI.

sej7278, 2015-10-22:
i've given up with esp8266, my last one went crazy and fried itself and some regulators, just not enough quality control, i'm sticking with avr.

ladislas, 2015-10-22:
>  i'm sticking with avr.

ahaha! ;)

and what about the Due?

sej7278, 2015-10-22:
> and what about the Due?

i think i was just a bit sick in my mouth

ladislas, 2015-10-24:
:D come on!

I don't have much time on my hand at the moment but I'd like to get my hands dirty and try stuff with all of that.

I was thinking about something: to allow for more architecture (avr, teensy, arm, etc.) maybe it could be easier to have one big makefile that will call smaller makefiles depending on the architecture, like we do with the Teensy. they could all hide all the complexe stuff and be maintained separately. Making changes to one of them would not break the other as they would not be linked together (working on Teensy won't break avr or arm). Adding a new architecture would only be a matter of adding a new file and using another one as a template.

what do you guys think?

sej7278, 2015-10-24:
i think the main problem is parsing boards.txt - now made worse by boards manager and the stupid numbering scheme of the hardware directories:

```
~/.arduino15/packages/esp8266/hardware/esp8266$ ls
1.6.4-628-g545ffde  1.6.5-1160-gef26c5f  1.6.5-947-g39819f0
```

i think to go much further we'd need to switch almost entirely to python like your autolib and the reset script. perhaps even have python write a small Makefile, rather than running the whole thing every time you call ```make```. it would be pretty cool to call "make makefile" and have it create a totally standalone Makefile with all your board settings that you could commit to git.

i did try to write an Esp8266.mk (as a separate file, yes its no good making a huge Arduino.mk) but didn't get very far, but to be fair that's probably harder than getting ARM to work as you have the espressif SDK, the xtensa toolchain and then the arduino core (including modified standard libraries) on top.

Teensy.mk is broken now anyway with 1.6

sudar, 2015-10-28:
> it would be pretty cool to call "make makefile" and have it create a totally standalone Makefile with all your board settings that you could commit to git.

That would be pretty cool :)

rnestler, 2016-02-26:
What's the status of this? I'm trying to use Arduino-Makefile with the [SAMD21](https://learn.sparkfun.com/tutorials/samd21-minidev-breakout-hookup-guide/samd21-overview) from SparkFun but with no success. What I tried was setting the `BOARDS_TXT` variable:
```Makefile
ARDUINO_DIR   = arduino
ARDMK_DIR     = Arduino-Makefile

BOARD_TAG     = samd21_mini
BOARDS_TXT    = ~/.arduino15/packages/SparkFun/hardware/samd/1.1.0/boards.txt

LOCAL_INO_SRCS = arduino-minimal/arduino-minimal.ino

include Arduino-Makefile/Arduino.mk
```
Output:

```Shell
-------------------------
Arduino.mk Configuration:
- [AUTODETECTED]       CURRENT_OS = LINUX 
- [USER]               ARDUINO_DIR = arduino 
- [USER]               ARDMK_DIR = Arduino-Makefile 
- [AUTODETECTED]       ARDUINO_VERSION = 167 
- [DEFAULT]            ARCHITECTURE = avr 
- [DEFAULT]            ARDMK_VENDOR = arduino 
- [AUTODETECTED]       ARDUINO_PREFERENCES_PATH = /home/sensirion.lokal/rnestler/.arduino15/preferences.txt 
- [AUTODETECTED]       ARDUINO_SKETCHBOOK = /home/sensirion.lokal/rnestler/Arduino (from arduino preferences file)
- [BUNDLED]            AVR_TOOLS_DIR = arduino/hardware/tools/avr (in Arduino distribution)
- [COMPUTED]           ARDUINO_LIB_PATH = arduino/libraries (from ARDUINO_DIR)
- [COMPUTED]           ARDUINO_PLATFORM_LIB_PATH = arduino/hardware/arduino/avr/libraries (from ARDUINO_DIR)
- [COMPUTED]           ARDUINO_VAR_PATH = arduino/hardware/arduino/avr/variants (from ARDUINO_DIR)
- [USER]               BOARDS_TXT = ~/.arduino15/packages/SparkFun/hardware/samd/1.1.0/boards.txt 
- [DEFAULT]            USER_LIB_PATH = /home/sensirion.lokal/rnestler/Arduino/libraries (in user sketchbook)
- [DEFAULT]            PRE_BUILD_HOOK = pre-build-hook.sh 
- [USER]               BOARD_TAG = samd21_mini 
- [COMPUTED]           CORE = arduino (from build.core)
- [COMPUTED]           VARIANT = SparkFun_SAMD_Mini (from build.variant)
- [COMPUTED]           OBJDIR = build-samd21_mini (from BOARD_TAG)
- [COMPUTED]           ARDUINO_CORE_PATH = arduino/hardware/arduino/avr/cores/arduino (from ARDUINO_DIR, BOARD_TAG and boards.txt)
- [DETECTED]           MONITOR_BAUDRATE = 115200  (in sketch)
- [DEFAULT]            OPTIMIZATION_LEVEL = s 
- [DEFAULT]            MCU_FLAG_NAME = mmcu 
- [DEFAULT]            CFLAGS_STD =  
- [DEFAULT]            CXXFLAGS_STD =  
- [AUTODETECTED]       DEVICE_PATH = /dev/ttyACM0 
- [DEFAULT]            FORCE_MONITOR_PORT =  
- [AUTODETECTED]       Size utility: AVR-aware for enhanced output
-
-                      ARDUINO_LIBS =
- [USER]                 ArduinoPebbleSerial
- [PLATFORM]             SPI
- [PLATFORM]             Wire
- [COMPUTED]           BOOTLOADER_PARENT = arduino/hardware/arduino/avr/bootloaders (from ARDUINO_DIR)
- [COMPUTED]           ARDMK_VERSION = 1.5 
- [COMPUTED]           CC_VERSION = 4.8.1 (avr-gcc)
-------------------------
mkdir -p build-samd21_mini
arduino/hardware/tools/avr/bin/avr-g++ -x c++ -include Arduino.h -MMD -c -mmcu=cortex-m0plus -DF_CPU=48000000L -DARDUINO=167 -DARDUINO_ARCH_AVR -D__PROG_TYPES_COMPAT__ -Iarduino/hardware/arduino/avr/cores/arduino -Iarduino/hardware/arduino/avr/variants/SparkFun_SAMD_Mini    -Iarduino/hardware/arduino/avr/libraries/SPI   -Iarduino/hardware/arduino/avr/libraries/Wire -Iarduino/hardware/arduino/avr/libraries/Wire/utility   -I/home/sensirion.lokal/rnestler/Arduino/libraries/ArduinoPebbleSerial -I/home/sensirion.lokal/rnestler/Arduino/libraries/ArduinoPebbleSerial/utility -Wall -ffunction-sections -fdata-sections -Os -fno-exceptions  arduino-minimal/arduino-minimal.ino -o build-samd21_mini/arduino-minimal/arduino-minimal.ino.o
avr-g++: error: unrecognized argument in option '-mmcu=cortex-m0plus'
avr-g++: note: valid arguments to '-mmcu=' are: at43usb320 at43usb355 at76c711 at86rf401 at90c8534 at90can128 at90can32 at90can64 at90pwm1 at90pwm161 at90pwm2 at90pwm216 at90pwm2b at90pwm3 at90pwm316 at90pwm3b at90pwm81 at90s1200 at90s2313 at90s2323 at90s2333 at90s2343 at90s4414 at90s4433 at90s4434 at90s8515 at90s8535 at90scr100 at90usb1286 at90usb1287 at90usb162 at90usb646 at90usb647 at90usb82 at94k ata5272 ata5505 ata5702m322 ata5782 ata5790 ata5790n ata5795 ata5831 ata6285 ata6286 ata6289 ata6612c ata6613c ata6614q ata6616c ata6617c ata664251 atmega103 atmega128 atmega1280 atmega1281 atmega1284 atmega1284p atmega1284rfr2 atmega128a atmega128rfa1 atmega128rfr2 atmega16 atmega161 atmega162 atmega163 atmega164a atmega164p atmega164pa atmega165 atmega165a atmega165p atmega165pa atmega168 atmega168a atmega168p atmega168pa atmega168pb atmega169 atmega169a atmega169p atmega169pa atmega16a atmega16hva atmega16hva2 atmega16hvb atmega16hvbrevb atmega16m1 atmega16u2 atmega16u4 atmega2560 atmega2561 atmega2564rfr2 atmega256rfr2 atmega32 atmega323 atmega324a atmega324p atmega324pa atmega325 atmega3250 atmega3250a atmega3250p atmega3250pa atmega325a atmega325p atmega325pa atmega328 atmega328p atmega329 atmega3290 atmega3290a atmega3290p atmega3290pa atmega329a atmega329p atmega329pa atmega32a atmega32c1 atmega32hvb atmega32hvbrevb atmega32m1 atmega32u2 atmega32u4 atmega32u6 atmega406 atmega48 atmega48a atmega48p atmega48pa atmega48pb atmega64 atmega640 atmega644 atmega644a atmega644p atmega644pa atmega644rfr2 atmega645 atmega6450 atmega6450a atmega6450p atmega645a atmega645p atmega649 atmega6490 atmega6490a atmega6490p atmega649a atmega649p atmega64a atmega64c1 atmega64hve atmega64hve2 atmega64m1 atmega64rfr2 atmega8 atmega8515 atmega8535 atmega88 atmega88a atmega8 [... cut ...] 3b atxmega256a3bu atxmega256a3u atxmega256c3 atxmega256d3 atxmega32a4 atxmega32a4u atxmega32c3 atxmega32c4 atxmega32d3 atxmega32d4 atxmega32e5 atxmega384c3 atxmega384d3 atxmega64a1 atxmega64a1u atxmega64a3 atxmega64a3u atxmega64a4u atxmega64b1 atxmega64b3 atxmega64c3 atxmega64d3 atxmega64d4 atxmega8e5 avr1 avr2 avr25 avr3 avr31 avr35 avr4 avr5 avr51 avr6 avrtiny avrxmega2 avrxmega4 avrxmega5 avrxmega6 avrxmega7 m3000
make: *** [build-samd21_mini/arduino-minimal/arduino-minimal.ino.o] Error 1
```

PeterMitrano, 2016-03-15:
So has anyone compiled for Due without the Arduino IDE? I'm happy with some compromise of copying platforms.txt or boards.txt somewhere else to avoid the crazy new package manager scheme. I just need a way around the insane restrictions of the arduino IDE

rhaamo, 2016-08-01:
Still no progress ? Wants to use with a Due here.

meiower, 2016-10-20:
Seriously, just use http://platformio.org/

I was initially put off by the fact it looks 'complex looking' - but it is so straightforward, and provides the _easiest, most effortless experience_ I've **_ever_** had with build enviroments.  A couple of clicks, a run through the simple Quick Start (for CLI or IDE) and I had non-Arduino-IDE-borked code compiled and running on my AdaFruit Feather M0 / ATSAMD21.  And platform.io just has it all working, with NO messing around. And integrating the CLI into your existing build processes is just a case of cutting out all those rules and replacing them with a couple of simple commands.  It probably took me 10 times as long as it should to get a basic example uploaded and executing, _just because I expected it to be difficult or indeed impossible_, after banging my head against doing it via arduino-builder or Arduino-mk for days..!

Saying that, I had some success borking Arduino-mk to at least get the g++ steps working for my ATSAMD21.  I only know basic Makefile hacking, and I don't know the intents of either Arduino-mk or arduino-builder/Arduino IDE in terms of 'overall' purpose/approach or build steps to know whether they're mappable/analogous - I mean the same basic stuff has to occur, but I suspect the breakdown/ordering of steps is different between the two.  Beyond my ability to care after the requisite head-banging..!

Anyway, I believe the approach is valid, and probably just needs someone with good Arduino-mk / Makefile -fu to remunge things!

- Update your Arduino IDE install, platforms, boards, libraries etc (inc the Feather M0 from AdaFruit for me)

PeterMitrano, 2016-10-21:
+1 for just use platformio.

doronbehar, 2017-10-11:
I'm trying to work on this issue and create a PR with basic support for SAM boards. During the process, I noticed `Arduino.mk` doesn't read `platform.txt` for every `$ARCHITECTURE` it compiles to it; and in order to get the necessary compilation settings and compilation flags etc, `Arduino.mk` sets those by itself.
I was wondering, what if we would sort of re-write most of `Arduino.mk` and in it we will use shell functions that will parse `platform.txt` and `boards.txt` and get all the necessary variables from there? This way, we would get rid from most of the responsibility our dear `Arduino.mk` holds and we'll take all the settings and variables from those text files. After all, these text files are the one which the IDE parses so we couldn't get wrong this way. (Am I right?)

What do you think @sudar?

ladislas, 2017-10-12:
@doronbehar i like the idea of moving part of the logic of Arduino.mk to a scripting language, either shell but why not Python or Ruby. Parsing platform.txt and boards.txt will be a breeze using those and will allow us to use the makefile, fed with the right parameters, just to do the compilation part.

sudar, 2017-10-12:
I will be fine with either shell or Python (since we are already using Python binaries) to parse the `platform.txt` and `boards.txt` files.

sej7278, 2017-10-12:
a python script to parse the txt files and then a makefile wrapper around arduino-builder is another way to go

ladislas, 2017-10-13:
@meiower I've just taken a look at Platformio and installed the plugin for VSCode. I'm pretty impressed with what they are doing and I'm doing to dig deeper!

I love the fact they provide a basic folder organization just like I was trying to do with [bare arduino project](https://github.com/ladislas/Bare-Arduino-Project).

also adding a toolchain (to upgrade gcc for example), seems pretty trivial.

doronbehar, 2017-10-13:
@sej7278 IMO, adding `arduino-builder` as a dependency would be needless.

First of all, `arduino-builder` has some sort of makefile-like approach that was made in order to reduce the amount of command line arguments into one file. This is done with the `build.options.json` file as explained here: [https://wiki.archlinux.org/index.php/Arduino#Arduino-Builder](https://wiki.archlinux.org/index.php/Arduino#Arduino-Builder).

Further most, even when considering using a python script to parse the text files, this also might be considered as a redundant dependency because it is possible to do it all in a 'pure makefile' way, thus sparing the python dependency (at least for every target which is not the `reset` target).
In addition, not many people know Python, (I'm one of them as well :|) and the makefile scripting language is so similar to shell like scripting so IMHO it should encourage more and more contributors to help this project and fine-tune it and make it better and better.

sej7278, 2017-10-13:
@doronbehar - i'm all for keeping it a pure makefile, but i really don't think parsing boards.txt and platform.txt (especially the random stuff people are putting in submenu's these days) as well as handling the mess created by boards manager (randomly numbered directories with cores inside) is going to be achievable without a more robust scripting language, i doubt even shell could do it, but i'd love to be proven wrong.

yes arduino-builder is a bit of a duplication of effort, but for unsupported platforms like esp8266, i find writing a simple makefile around it less work than trying to extend arduino-mk, although its far less feature-rich than arduino-mk. i wish the arduino folks had contributed to arduino-mk rather than creating arduino-builder, but they don't seem to work that way and prefer reinventing the wheel with java.

tuna-f1sh, 2018-02-11:
I've created #543 for my attempt at SAM support. Could do with some more user testing and contribution.