sudar/Arduino-Makefile #344: Compilation StandardFirmata fails

mulderp, 2015-04-07:
Hi,

I was just trying to build the Standard Firmata sketch with the Makefile, but I get an error about missing delcaration of `enableI2CPins` 

```
mkdir -p build-leonardo
/Applications/Arduino.app/Contents/Java/hardware/tools/avr//bin/avr-g++ -x c++ -include Arduino.h -MMD -c -mmcu=atmega32u4 -DF_CPU=16000000L -DARDUINO=163 -DARDUINO_ARCH_AVR -D__PROG_TYPES_COMPAT__ -I/Applications/Arduino.app/Contents/Java//hardware/arduino/avr/cores/arduino -I/Applications/Arduino.app/Contents/Java//hardware/arduino/avr/variants/leonardo    -I/Applications/Arduino.app/Contents/Java//libraries/Firmata/src    -I/Applications/Arduino.app/Contents/Java//libraries/Servo/src   -I/Applications/Arduino.app/Contents/Java//hardware/arduino/avr/libraries/Wire -I/Applications/Arduino.app/Contents/Java//hardware/arduino/avr/libraries/Wire/utility  -Wall -ffunction-sections -fdata-sections -Os -DUSB_VID=0x2341 -DUSB_PID=0x8036 -fno-exceptions  standard_firmata.ino -o build-leonardo/standard_firmata.o
In file included from /Applications/Arduino.app/Contents/Java//libraries/Firmata/src/Firmata.h:16:0,
                 from standard_firmata.ino:34:
/Applications/Arduino.app/Contents/Java//libraries/Firmata/src/Boards.h:341:19: warning: extra tokens at end of #ifndef directive [enabled by default]
 #ifndef IS_PIN_SPI(p)
                   ^
/Applications/Arduino.app/Contents/Java//libraries/Firmata/src/Boards.h: In function 'unsigned char writePort(byte, byte, byte)':
/Applications/Arduino.app/Contents/Java//libraries/Firmata/src/Boards.h:414:1: warning: no return statement in function returning non-void [-Wreturn-type]
 }
 ^
standard_firmata.ino: In function 'void setPinModeCallback(byte, int)':
standard_firmata.ino:183:20: error: 'disableI2CPins' was not declared in this scope
     disableI2CPins();
                    ^
standard_firmata.ino:189:68: error: 'reportAnalogCallback' was not declared in this scope
     reportAnalogCallback(PIN_TO_ANALOG(pin), mode == ANALOG ? 1 : 0); // turn on/off reporting
                                                                    ^
standard_firmata.ino: In function 'void sysexCallback(byte, byte, byte*)':
standard_firmata.ino:396:47: warning: suggest parentheses around assignment used as truth value [-Wparentheses]
               if (query[i].addr = slaveAddress) {
                                               ^
standard_firmata.ino:424:23: error: 'enableI2CPins' was not declared in this scope
         enableI2CPins();
                       ^
standard_firmata.ino: In function 'void loop()':
standard_firmata.ino:622:40: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
   if (currentMillis - previousMillis > samplingInterval) {
                                        ^
make: *** [build-leonardo/standard_firmata.o] Error 1

```

mulderp, 2015-04-07:
Also with including the libraries I get:

    ARDUINO_LIBS = Servo SPI Wire Firmata

But probably this is a problem with the Sketch not with the Makefile

closed 2015-04-07 by mulderp

sudar, 2015-04-07:
@mulderp This is because the functions are not declared.

Standard C/C++ expects you to declare the functions but Arduino automatically does this in the background to help the beginners.

We decided not to do it in the makefile since the users of this makefile are mostly power users.

See #59 for further context on this.

mulderp, 2015-04-07:
Thanks, for your feedback. That is helpful for my current learnings.

By the way, background of my small project is to help others (software beginners probably) building Arduino firmware from the command line.

A Makefile approach seems quite interesting, although they might be other approaches like discussed in: https://github.com/Pinoccio/js-stk500/issues/7

sej7278, 2015-04-07:
just confirmed it does compile if you move some of the function declarations around in StandardFirmata.ino