sudar/Arduino-Makefile #345: Board tag "micro" fails to build

SamWhited, 2015-04-12:
Building for an Uno works fine, but if I change the board tag to `micro` I start getting build errors. It complains that the symbols `USB_PID` and `USB_VID` aren't defined in `USBCore.o`.

My sub-make file looks like this:

```make
PORT = /dev/ttyACM0
MONITOR_PORT = /dev/ttyACM0
TARGET = SomeProject
ARDUINO = /usr/share/arduino/lib/targets/arduino

ARDUINO_DIR = /usr/share/arduino
ARDMK_DIR = /usr/share/arduino
AVR_TOOLS_DIR = /usr
ARDUINO_CORE_PATH = /usr/share/arduino/hardware/arduino/avr/cores/arduino
BOARDS_TXT = /usr/share/arduino/hardware/arduino/avr/boards.txt
ARDUINO_VAR_PATH = /usr/share/arduino/hardware/arduino/avr/variants
BOOTLOADER_PARENT = /usr/share/arduino/hardware/arduino/avr/bootloaders

BOARD_TAG    = micro
ARDUINO_LIBS =

CFLAGS = -std=c11

USB_PID = 0x0037
USB_VID = 0x2341

include /usr/share/arduino/Arduino.mk
```

Manually adding the correct pre-processor directives in `hardware/arduino/avr/variants/micro/pins_arduino.h` seems to fix the build:

```c
#ifndef USB_PID
#define USB_PID 0x8037
#endif

#ifndef USB_VID
#define USB_VID 0x2341
#endif
```

I'm not sure if this is a problem with the makefile or a problem upstream, as most other variants don't seem to declare the USB ID's either, and I didn't see where else they would be defined or where the problem would be coming from (though admittedly, I just did a quick grep through the makefile and hardware dir and didn't spend much time on it).

Other potentially useful information:

```bash
$ uname -a
Linux macrocarpa 3.19.3-3-ARCH #1 SMP PREEMPT Wed Apr 8 14:10:00 CEST 2015 x86_64 GNU/Linux
$ make -v
GNU Make 4.1
Built for x86_64-unknown-linux-gnu [...]
$ lsusb | grep Arduino
Bus 002 Device 077: ID 2341:8037 Arduino SA
$ ls /dev/ttyA*
/dev/ttyACM0
```

Arduino 1.6.3 32-bit is installed.

sej7278, 2015-04-12:
i think you're defining too much in your makefile and maybe getting one of the settings wrong rather than letting it be computed/detected. i'm assuming you're using arduino-mk 1.5 from git and not 1.3.4 from arch?

this works for me:

```
ARDUINO_DIR = $(HOME)/arduino-1.6.3
MONITOR_PORT = /dev/ttyACM0
BOARD_TAG = micro
include /usr/share/arduino/Arduino.mk
```

and my output is:

```
Arduino.mk Configuration:
- [AUTODETECTED]       CURRENT_OS = LINUX 
- [USER]               ARDUINO_DIR = $(HOME)/arduino-1.6.3 
- [COMPUTED]           ARDMK_DIR = /usr/share/arduino (relative to Common.mk)
- [AUTODETECTED]       ARDUINO_VERSION = 163 
- [DEFAULT]            ARCHITECTURE = avr 
- [DEFAULT]            VENDOR = arduino 
- [AUTODETECTED]       ARDUINO_PREFERENCES_PATH = /home/simon/.arduino15/preferences.txt 
- [AUTODETECTED]       ARDUINO_SKETCHBOOK = $(HOME)/arduino16 (from arduino preferences file)
- [BUNDLED]            AVR_TOOLS_DIR = $(HOME)/arduino-1.6.3/hardware/tools/avr (in Arduino distribution)
- [COMPUTED]           ARDUINO_LIB_PATH = $(HOME)/arduino-1.6.3/libraries (from ARDUINO_DIR)
- [COMPUTED]           ARDUINO_PLATFORM_LIB_PATH = $(HOME)/arduino-1.6.3/hardware/arduino/avr/libraries (from ARDUINO_DIR)
- [COMPUTED]           ARDUINO_VAR_PATH = $(HOME)/arduino-1.6.3/hardware/arduino/avr/variants (from ARDUINO_DIR)
- [COMPUTED]           BOARDS_TXT = $(HOME)/arduino-1.6.3/hardware/arduino/avr/boards.txt (from ARDUINO_DIR)
- [DEFAULT]            USER_LIB_PATH = $(HOME)/arduino16/libraries (in user sketchbook)
- [DEFAULT]            PRE_BUILD_HOOK = pre-build-hook.sh 
- [USER]               BOARD_TAG = micro 
- [COMPUTED]           CORE = arduino (from build.core)
- [COMPUTED]           VARIANT = micro (from build.variant)
- [COMPUTED]           OBJDIR = build-micro (from BOARD_TAG)
- [COMPUTED]           ARDUINO_CORE_PATH = $(HOME)/arduino-1.6.3/hardware/arduino/avr/cores/arduino (from ARDUINO_DIR, BOARD_TAG and boards.txt)
- [ASSUMED]            MONITOR_BAUDRATE = 9600 
- [DEFAULT]            OPTIMIZATION_LEVEL = s 
- [DEFAULT]            MCU_FLAG_NAME = mmcu 
- [DEFAULT]            CFLAGS_STD =  
- [DEFAULT]            CXXFLAGS_STD =  
- [COMPUTED]           DEVICE_PATH = /dev/ttyACM0 (from MONITOR_PORT)
- [DEFAULT]            FORCE_MONITOR_PORT =  
- [AUTODETECTED]       Size utility: AVR-aware for enhanced output
-
-                      ARDUINO_LIBS =
- [USER]                 DallasTemperature
- [USER]                 OneWire
- [USER]                 RF24
- [USER]                 RF24Network
- [PLATFORM]             SPI
- [COMPUTED]           BOOTLOADER_PARENT = $(HOME)/arduino-1.6.3/hardware/arduino/avr/bootloaders (from ARDUINO_DIR)
- [COMPUTED]           ARDMK_VERSION = 1.5 
- [COMPUTED]           CC_VERSION = 4.8.1 (avr-gcc)
```

i've not tried uploading as i don't have a micro, but another 32u4 device (sf pro micro) works fine. they have recently added new usb flags to boards.txt, not really sure what that's about e.g.

```
micro.vid.0=0x2341
micro.pid.0=0x0037
micro.vid.1=0x2341
micro.pid.1=0x8037
micro.vid.2=0x2A03
micro.pid.2=0x0037
micro.vid.3=0x2A03
micro.pid.3=0x8037
micro.vid.0x2A03.warning=Uncertified
micro.build.vid=0x2341
micro.build.pid=0x8037
micro.build.usb_product="Arduino Micro"
micro.build.extra_flags={build.usb_flags}
```

SamWhited, 2015-04-12:
I'm using the version off the AUR.

Cleaning up everything in my make file and using the minimal version you provided doesn't work, it complains about not being able to find boards.txt. So I add `BOARDS_TXT` and then it complains about not being able to find Arduino.h, so I add `ARDUINO_CORE_PATH`, then it complains about not being able to find pins_arduino.h, so I add `ARDUINO_VAR_PATH`, at which point it complains about the build failing because it doesn't know the USB VID/PID.

Using the latest version from Git seems to work, however (I didn't pay enough attention to the PKGBUILD apparently). As it seems this problem was already fixed, I'll close this. Sorry for the noise.

closed 2015-04-12 by SamWhited

sej7278, 2015-04-12:
i don't know what the process is for getting the AUR package updated, but yes 1.3.4-2 won't work with arduino > 1.0.6