sudar/Arduino-Makefile #348: delay() seems to be hanging ATmega664

kautzz, 2015-04-17:
First things first:
Thanks a bunch for putting so much work into this project.
It's very straightforward to use and just what I'm looking for.

I have an ATmega644 with Ethernet connectivity (enc20j60 via spi) that will be master on a RS485 bus.
The controller uses the e6 ftftp bootloader to program itself with a binary from a raspberryPi 2.
Software is build on the Pi. It's running Ubuntu 14.04.02 LTS (GNU/Linux 3.18.0-14-rpi2 armv7l). Arduino version is 1.6.3 and I grabbed the latest Arduino-Makefile commit.

I followed http://hardwarefun.com/tutorials/compiling-arduino-sketches-using-makefile and http://hardwarefun.com/tutorials/use-arduino-code-in-non-arduino-avr-microcontroller to create a boards.txt, pinnings_arduino.h and the small project makefile.

So I'm trying to build a little blink example with an RGB LED. Make seems to be running just fine, output looks good as far as I can tell. I can flash the binary over ethernet (checksums ok) and the controller starts the new program. 

**But as soon as the code hits a delay, the controller seems to freeze.**

What could cause this? 
Did I make a mistake in the pinning_arduino.h?
Is there an error in the make output I don't see?

The weird things is, when I upload a more complex program (program 2500 and data 1399 bytes) the controller seem to be doing exacly as told and delay() seems to be fine.

Is this maybe some kind of compiler optimization bug? (heard those were common with ubuntu)

---

### Expected Output:
LED blue, delay 3s, LED Cyan, delay 3s, LED white

### Actual Output:
LED blue

### Controller Code master.cpp:

```cpp
#include <Arduino.h>

#define RS485TX 8
#define RS485RX 9
#define DriverEnable 20
#define ReceiverEnable 10
#define TerminationEnable 21 

#define LOG_RX_PIN 30
#define LOG_TX_PIN 31

#define REDPIN 13
#define GREENPIN 14
#define BLUEPIN 12

void setup()
{
  pinMode(REDPIN, OUTPUT);
  pinMode(GREENPIN, OUTPUT);
  pinMode(BLUEPIN, OUTPUT);
}

void loop()
{
  digitalWrite(BLUEPIN, HIGH);
  delay(3000);
    
  digitalWrite(GREENPIN, HIGH);
  delay(3000);
       
  digitalWrite(REDPIN, HIGH);
  delay(3000);

  digitalWrite(REDPIN, LOW);
  delay(3000);
}
```
[... 442 lines cut ...]
                            ^
/usr/share/arduino/hardware/arduino/avr/cores/arduino/Tone.cpp:452:28: warning: binary constants are a GCC extension [enabled by default]
         TCCR2B = (TCCR2B & 0b11111000) | (1 << CS22);
                            ^
/usr/bin/avr-g++ -MMD -c -mmcu=atmega644 -DF_CPU=16000000 -DARDUINO=100 -DARDUINO_ARCH_AVR -D__PROG_TYPES_COMPAT__ -I/usr/share/arduino/hardware/arduino/avr/cores/arduino -I/home/ubuntu/sketchbook/hardware/plantsandmachines/avr/variants/brain    -Wall -ffunction-sections -fdata-sections -Os -pedantic -Wall -Wextra -fno-exceptions -std=gnu++11 /usr/share/arduino/hardware/arduino/avr/cores/arduino/USBCore.cpp -o /home/ubuntu/controller-software/pinning/bin/plantsandmachines_brain_644/pinning/core/USBCore.o
/usr/bin/avr-g++ -MMD -c -mmcu=atmega644 -DF_CPU=16000000 -DARDUINO=100 -DARDUINO_ARCH_AVR -D__PROG_TYPES_COMPAT__ -I/usr/share/arduino/hardware/arduino/avr/cores/arduino -I/home/ubuntu/sketchbook/hardware/plantsandmachines/avr/variants/brain    -Wall -ffunction-sections -fdata-sections -Os -pedantic -Wall -Wextra -fno-exceptions -std=gnu++11 /usr/share/arduino/hardware/arduino/avr/cores/arduino/WMath.cpp -o /home/ubuntu/controller-software/pinning/bin/plantsandmachines_brain_644/pinning/core/WMath.o
/usr/bin/avr-g++ -MMD -c -mmcu=atmega644 -DF_CPU=16000000 -DARDUINO=100 -DARDUINO_ARCH_AVR -D__PROG_TYPES_COMPAT__ -I/usr/share/arduino/hardware/arduino/avr/cores/arduino -I/home/ubuntu/sketchbook/hardware/plantsandmachines/avr/variants/brain    -Wall -ffunction-sections -fdata-sections -Os -pedantic -Wall -Wextra -fno-exceptions -std=gnu++11 /usr/share/arduino/hardware/arduino/avr/cores/arduino/WString.cpp -o /home/ubuntu/controller-software/pinning/bin/plantsandmachines_brain_644/pinning/core/WString.o
/usr/bin/avr-ar rcs /home/ubuntu/controller-software/pinning/bin/plantsandmachines_brain_644/pinning/libcore.a  /home/ubuntu/controller-software/pinning/bin/plantsandmachines_brain_644/pinning/core/hooks.o  /home/ubuntu/controller-software/pinning/bin/plantsandmachines_brain_644/pinning/core/WInterrupts.o  /home/ubuntu/controller-software/pinning/bin/plantsandmachines_brain_644/pinning/core/wiring_analog.o  /home/ubuntu/controller-software/pinning/bin/plantsandmachines_brain_644/pinning/core/wiring.o  /home/ubuntu/controller-software/pinning/bin/plantsandmachines_brain_644/pinning/core/wiring_digital.o  /home/ubuntu/controller-software/pinning/bin/plantsandmachines_brain_644/pinning/core/wiring_pulse.o  /home/ubuntu/controller-software/pinning/bin/plantsandmachines_brain_644/pinning/core/wiring_shift.o  /home/ubuntu/controller-software/pinning/bin/plantsandmachines_brain_644/pinning/core/abi.o  /home/ubuntu/controller-software/pinning/bin/plantsandmachines_brain_644/pinning/core/CDC.o  /home/ubuntu/controller-software/pinning/bin/plantsandmachines_brain_644/pinning/core/HardwareSerial0.o  /home/ubuntu/controller-software/pinning/bin/plantsandmachines_brain_644/pinning/core/HardwareSerial1.o  /home/ubuntu/controller-software/pinning/bin/plantsandmachines_brain_644/pinning/core/HardwareSerial2.o  /home/ubuntu/controller-software/pinning/bin/plantsandmachines_brain_644/pinning/core/HardwareSerial3.o  /home/ubuntu/controller-software/pinning/bin/plantsandmachines_brain_644/pinning/core/HardwareSerial.o  /home/ubuntu/controller-software/pinning/bin/plantsandmachines_brain_644/pinning/core/HID.o  /home/ubuntu/controller-software/pinning/bin/plantsandmachines_brain_644/pinning/core/IPAddress.o  /home/ubuntu/controller-software/pinning/bin/plantsandmachines_brain_644/pinning/core/main.o  /home/ubuntu/controller-software/pinning/bin/plantsandmachines_brain_644/pinning/core/new.o  /home/ubuntu/controller-software/pinning/bin/plantsandmachines_brain_644/pinning/core/Print.o  /home/ubuntu/controller-software/pinning/bin/plantsandmachines_brain_644/pinning/core/Stream.o  /home/ubuntu/controller-software/pinning/bin/plantsandmachines_brain_644/pinning/core/Tone.o  /home/ubuntu/controller-software/pinning/bin/plantsandmachines_brain_644/pinning/core/USBCore.o  /home/ubuntu/controller-software/pinning/bin/plantsandmachines_brain_644/pinning/core/WMath.o  /home/ubuntu/controller-software/pinning/bin/plantsandmachines_brain_644/pinning/core/WString.o       
/usr/bin/avr-gcc -mmcu=atmega644 -Wl,--gc-sections -Os -o /home/ubuntu/controller-software/pinning/bin/plantsandmachines_brain_644/pinning/brain0.elf /home/ubuntu/controller-software/pinning/bin/plantsandmachines_brain_644/pinning/master.o /home/ubuntu/controller-software/pinning/bin/plantsandmachines_brain_644/pinning/libcore.a  -lc -lm
/usr/bin/avr-objcopy -j .eeprom --set-section-flags=.eeprom='alloc,load' \
		--change-section-lma .eeprom=0 -O ihex /home/ubuntu/controller-software/pinning/bin/plantsandmachines_brain_644/pinning/brain0.elf /home/ubuntu/controller-software/pinning/bin/plantsandmachines_brain_644/pinning/brain0.eep
/usr/bin/avr-objcopy: --change-section-lma .eeprom=0x00000000 never used
/usr/bin/avr-objcopy -O ihex -R .eeprom /home/ubuntu/controller-software/pinning/bin/plantsandmachines_brain_644/pinning/brain0.elf /home/ubuntu/controller-software/pinning/bin/plantsandmachines_brain_644/pinning/brain0.hex

/usr/bin/avr-size --mcu=atmega644 -C --format=avr /home/ubuntu/controller-software/pinning/bin/plantsandmachines_brain_644/pinning/brain0.elf
AVR Memory Usage
----------------
Device: atmega644

Program:    1136 bytes (1.7% Full)
(.text + .data + .bootloader)

Data:          9 bytes (0.2% Full)
(.data + .bss + .noinit)
```

sej7278, 2015-04-17:
no idea really, could it be something to do with the timer functions?

the makefile seems to think you are using arduino 1.0.0 (which is the raspbian wheezy default) so its not finding the 1.6.3 you say you are using, try defining:

```
ARDUINO_DIR = /path/to/arduino-1.6.3
```

i've just received at atmega1284p-pu so if i come across a similar issue (the 644/1284 are similar chips) i'll report back. 

have you looked at the sanguino stuff, that was all 644-based.

kautzz, 2015-04-17:
The path to arduino is already set:
```
ARDUINO_DIR = /usr/share/arduino
```

/usr/share/arduino is a symlink to the arduino git repo
I set the Arduino Version manually to 163, no change though.
```
ARDUINO_VERSION = 163
```

Yeah, I have been using sanguino before. But the core is sooooooo old  ;)
Thought I give it a try with the regular arduino core since I've seen some ```if (mcu == atmega 644)``` statements in there

do you mean the timer functions in arduino core?
Where would they be located?

sej7278, 2015-04-17:
you shouldn't have to set ```ARDUINO_VERSION```, its not being auto-detected for some reason, ah are you using a nightly (or git) instead of a release - as the lib/version.txt says "nightly" in that case instead of a numeric, in which case you do need to set it manually.

anyway, i've just had a play with the mighty-1284p and it doesn't have a problem with blink (with delay) but i did of course have to ```make set_fuses && make ispload``` as it was a raw chip and i didn't want to bootload it.

i assume you're using a 16mhz crystal and have set the fuses and/or installed a bootloader?

i can only think its a problem with the timers in pins_arduino.h - it shouldn't be called pin**ning**s_arduino.h btw

other than that you seem to be using @ladislas' BareArduinoProject makefile, and i don't think they've ported our 1.6.x ide support yet.

kautzz, 2015-04-17:
yeah, you are right. the nightly doesn't even seem to have ```$(ARDUINO_DIR)/lib/```
so I'll leave it manually set to 163.

I have a 16MHz crystal and my fuses are set to:
```
sudo avrdude -c avrispmkII -p m644 -P usb -e -B4 -v -U lfuse:w:0xff:m -U hfuse:w:0xd8:m -U efuse:w:0xff:m
```

After I flash the bootloader via ISP I set the lockbit so it can't overwrite itself
```
sudo avrdude -c avrispmkII -p m644 -P usb -B4 -v -U lock:w:0x0F:m
```

Would you please send me your 1284 ```pins_arduino.h``` for reference?

Yup, using BareArduinoProject Makefile... is there a example makefile that supports 1.6.x?
The ones in https://github.com/sudar/Arduino-Makefile/tree/master/examples seem a bit incomplete/small compared to @ladislas 's one

ladislas, 2015-04-17:
I'm working on it :)

sej7278, 2015-04-17:
If you're using the 1.6.4 nightly i'd say forget it, its totally messed up, go back to 1.6.3 release.

@ladislas makefiles are a bit specific, if you're going to use that then i'd suggest reporting it in his issue tracker. this works for me on 1.0.5, i've not tried it on 1.6.3

```
BOARD_TAG         = mighty_opt
BOARDS_TXT        = $(HOME)/arduino/hardware/mighty-1284p/boards.txt
BOOTLOADER_PARENT = $(HOME)/arduino/hardware/mighty-1284p/bootloaders
BOOTLOADER_PATH   = optiboot
BOOTLOADER_FILE   = optiboot_atmega1284p.hex
ISP_PROG     	  = usbasp
AVRDUDE_OPTS 	  = -v
include $(HOME)/Arduino-Makefile/Arduino.mk
```

https://github.com/wijnen/mighty-1284p/blob/master/variants/standard/pins_arduino.h

kautzz, 2015-04-17:
sorry I meant arduino 1.6.3 from github.

Compared the mighty-1284 pins to mine. Found nothing obvious.
Tried with a slightly modified version (for the 644) of the 1284 pin_arduino.h
No change.

I also tried your makefile:
```
BOARD_TAG         = plantsandmachines_brain_644                                      
BOARDS_TXT        = $(HOME)/sketchbook/hardware/plantsandmachines/avr/boards.txt     
BOOTLOADER_PARENT = $(HOME)/arduino/hardware/plantsandmachines/avr/bootloaders       
                                                                                     
AVRDUDE_OPTS      = -v                                                               

include /opt/arduino-mk/Arduino.mk
```

With the result of an error:
```
ubuntu@mrvbot:~/controller-software/pinning$ make 
-------------------------
Arduino.mk Configuration:
- [AUTODETECTED]       CURRENT_OS = LINUX 
- [AUTODETECTED]       ARDUINO_DIR = /usr/share/arduino 
- [COMPUTED]           ARDMK_DIR = /home/ubuntu/external_repos/arduino-mk (relative to Common.mk)
- [DEFAULT]            ARDUINO_VERSION = 100 
- [DEFAULT]            ARCHITECTURE =  
- [DEFAULT]            VENDOR = arduino 
- [DEFAULT]            ARDUINO_SKETCHBOOK = /home/ubuntu/sketchbook 
- [AUTODETECTED]       AVR_TOOLS_DIR = /usr (found in $PATH)
- [COMPUTED]           ARDUINO_LIB_PATH = /usr/share/arduino/libraries (from ARDUINO_DIR)
- [USER]               ARDUINO_VAR_PATH = /home/ubuntu/sketchbook/hardware/plantsandmachines/avr/variants 
- [USER]               BOARDS_TXT = /home/ubuntu/sketchbook/hardware/plantsandmachines/avr/boards.txt 
- [DEFAULT]            USER_LIB_PATH = /home/ubuntu/sketchbook/libraries (in user sketchbook)
- [DEFAULT]            PRE_BUILD_HOOK = pre-build-hook.sh 
- [USER]               BOARD_TAG = plantsandmachines_brain_644 
- [COMPUTED]           CORE = arduino (from build.core)
- [COMPUTED]           VARIANT = brain (from build.variant)
- [COMPUTED]           OBJDIR = build-plantsandmachines_brain_644 (from BOARD_TAG)
- [COMPUTED]           ARDUINO_CORE_PATH = /usr/share/arduino/hardware/arduino//cores/arduino (from ARDUINO_DIR, BOARD_TAG and boards.txt)
-                      No .pde or .ino files found. If you are compiling .c or .cpp files then you need to explicitly include Arduino header files
- [ASSUMED]            MONITOR_BAUDRATE = 9600 
- [DEFAULT]            OPTIMIZATION_LEVEL = s 
- [DEFAULT]            MCU_FLAG_NAME = mmcu 
- [DEFAULT]            CFLAGS_STD =  
- [DEFAULT]            CXXFLAGS_STD =  
- [AUTODETECTED]       DEVICE_PATH =  
- [DEFAULT]            FORCE_MONITOR_PORT =  
- [AUTODETECTED]       Size utility: AVR-aware for enhanced output
- [USER]               BOOTLOADER_PARENT = /home/ubuntu/arduino/hardware/plantsandmachines/avr/bootloaders 
- [COMPUTED]           ARDMK_VERSION = 1.5 
- [COMPUTED]           CC_VERSION = 4.8.2 (avr-gcc)
-------------------------
mkdir -p build-plantsandmachines_brain_644
/usr/bin/avr-g++ -MMD -c -mmcu=atmega644 -DF_CPU=16000000L -DARDUINO=100  -D__PROG_TYPES_COMPAT__ -I/usr/share/arduino/hardware/arduino//cores/arduino -I/home/ubuntu/sketchbook/hardware/plantsandmachines/avr/variants/brain    -Wall -ffunction-sections -fdata-sections -Os -fno-exceptions  master.cpp -o build-plantsandmachines_brain_644/master.o
master.cpp:1:21: fatal error: Arduino.h: No such file or directory
 #include <Arduino.h>
                     ^
compilation terminated.
make: *** [build-plantsandmachines_brain_644/master.o] Error 1
```

Somehow the ```ARDUINO_CORE_PATH``` does not compute?!

sej7278, 2015-04-17:
you'll have to set ```ARDUINO_VERSION = 163``` otherwise it might not be able to find all of the headers, but other than that, i've no idea. i certainly wouldn't use arduino from git or nightly, use a zip.

kautzz, 2015-04-18:
okay thanks!
Like this: https://github.com/arduino/Arduino/releases/tag/1.6.3 ?

will continue trying :D

ladislas, 2015-04-18:
@kautzz I just updated Bare-Arduino-Project to support 1.6.3. At least it's working on my machine.

Could you give it a try and report any issue you might have [here](https://github.com/ladislas/Bare-Arduino-Project/issues). We'll work on it together :)

kautzz, 2015-04-18:
Got it running now!

Was messing around with interrupts to see if timers are working...
So this runs fine:
```cpp
#include <avr/interrupt.h>
#include <Arduino.h>

#define RS485TX 8
#define RS485RX 9

#define DriverEnable 20
#define ReceiverEnable 10
#define TerminationEnable 21 

#define LOG_RX_PIN 30
#define LOG_TX_PIN 31

#define REDPIN 13
#define GREENPIN 14
#define BLUEPIN 12

volatile unsigned long seconds = 1;

void setup()
{
  pinMode(REDPIN, OUTPUT);
  pinMode(GREENPIN, OUTPUT);
  pinMode(BLUEPIN, OUTPUT);

  digitalWrite(REDPIN, LOW);
  digitalWrite(GREENPIN, LOW);
  digitalWrite(BLUEPIN, LOW);

  // INITIALIZE TIMER INTERRUPTS

  cli();                      // disable global interrupts
  TCCR1A = 0;                 // set entire TCCR1A register to 0
  TCCR1B = 0;                 // same for TCCR1B

  OCR1A = 15624;              // set compare match register to desired timer count. 16 MHz with 1024 prescaler = 15624 counts/s 

  TCCR1B |= (1 << WGM12);     // turn on CTC mode. clear timer on compare match
  TCCR1B |= (1 << CS10);      // Set CS10 and CS12 bits for 1024 prescaler
  TCCR1B |= (1 << CS12);

  TIMSK1 |= (1 << OCIE1A);    // enable timer compare interrupt

  sei();                      // enable global interrupts
}

void loop()
{
  digitalWrite(REDPIN, HIGH);
  delay(1000);

  digitalWrite(GREENPIN, HIGH);
  delay(1000);

  digitalWrite(BLUEPIN, HIGH);
  delay(1000);

  digitalWrite(BLUEPIN, LOW);
  delay(1000);

  digitalWrite(GREENPIN, LOW);
  delay(1000);

  digitalWrite(REDPIN, LOW);
  delay(1000);
}

ISR(TIMER1_COMPA_vect)
{
  seconds++;
}
```

I was doing some more tests and figured that if I stop timer/counter1 the code runs fine, too! (TCCR1B Bit 2:0 – CSn2:0: Clock Select to no clock source)
```cpp
TCCR1B &= ~(1<<0);
TCCR1B &= ~(1<<1);
TCCR1B &= ~(1<<2);
```

Any idea why this could be happening?
I was under the impression that Arduino uses timer0 for delay... so does Arduino set up TCCR1B at all?

@ladislas yeah, great job! Sure, I'll test it later, thanks!
This is the makefile I've been using so far:
```

BOARD_TAG         = plantsandmachines_brain_644
BOARDS_TXT        = $(HOME)/sketchbook/hardware/plantsandmachines/avr/boards.txt
BOOTLOADER_PARENT = $(HOME)/arduino/hardware/plantsandmachines/avr/bootloaders

ARDUINO_VERSION = 163

ARDUINO_CORE_PATH = /usr/share/arduino/hardware/arduino/avr/cores/arduino
ARDUINO_VAR_PATH = $(HOME)/sketchbook/hardware/plantsandmachines/avr/variants

AVRDUDE_OPTS      = -v

TARGET = brain0

include /opt/arduino-mk/Arduino.mk

t:
        make
        avr-objcopy -I ihex -O binary build-plantsandmachines_brain_644/brain0.hex /tftpboot/brain0.bin
```

kautzz, 2015-04-19:
Okay, so @sej7278  was right. It was a timer problem.
It was not due to a faulty ```pins_arduino.h``` but rather due to hick ups when applying the timer settings
Somehow arduino-core messed up setting the right registers for timer0, 1 and 2.

I now set them manually, like so:
```
cli();
TCCR0A = B00000011;
TCCR0B = B00000011;
TIMSK0 = B00000001;
TCCR1A = B10100001;
TCCR1B = B00000011;
TCCR2A = B00000001;
TCCR2B = B00000100;
TIMSK1 = B00000000;
sei();
```

```delay()``` and ```analogWrite()``` are running like a charm now.
I'll go ahead and close this issue because it's probably not an error on the Arduino-Makefile side.

If anyone knows whats going wrong with setting the timer registers through arduino-core, please let me know. :)

closed 2015-04-19 by kautzz