sudar/Arduino-Makefile #349: Additional source directories

moonshadow565, 2015-04-24:
How about variable for additional source directory?

ladislas, 2015-04-24:
You mean like the [`USER_LIB_PATH`](https://github.com/sudar/Arduino-Makefile/blob/master/arduino-mk-vars.md#user_lib_path)?

moonshadow565, 2015-04-25:
No i mean like i put my main code in subfolder of Makefile 
something like SOURCE

ladislas, 2015-04-25:
I don't understand. Could you show an exemple of what you mean?

sej7278, 2015-04-25:
i guess this is a duplicate of #267 ?

moonshadow565, 2015-04-25:
Yeah it is but i trivial matter to implement this stuff  :P

luto, 2015-06-05:
I've got the same problem. My structure looks like this:

```
- main.c
- state.h
- state.c
- decode
  - decode_0.h
  - decode_0.c
  - decode_1.h
  - decode_1.c
```
(mostly made-up names)

Arduino-Makefile just does `LOCAL_C_SRCS ?= $(wildcard *.c)`. So the files in the subdirectory `decode` are not compiled.

I'd suggest a new variable `LOCAL_SRC_DIRS` which contains a list of directory to load source files from. Initially it would only contain `.`, which mimics the current behavior. Is this something you'd like to see? If yes, just tell me and I'll prepare a PR.

Is this what you wanted @moonshadow565?

luto, 2015-06-05:
Oh, https://github.com/sudar/Arduino-Makefile/pull/252. I'll have a look at this.

moonshadow565, 2015-06-08:
@luto yeah thats what i meant

sej7278, 2015-06-08:
rather than yet another variable which the user has to set, i'd rather a recursive search for files in all subdirectories of the sketch's directory (but not a separate python script) like the [spark core](https://github.com/spark/firmware/blob/master/build/makefile) firmware works.

dhlii, 2017-03-18:
I would like to see this too. 
Though I am more inclined to want to specify the source directories in the Makefile. 
As I often have project trees that build multiple ways - including building a windows or X11 simulator and I do NOT want that code getting sucked into Arduino Builds. 
As an example I might have:
Lock:
-Makefile
-lock.cpp
-GFX
--GFX.cpp
--LCD
---QDT
----lcd.cpp
---X11
----lcd.cpp

Where I have an x86 Makefile in the lock/GFX/LCD/X11 directory to build an X11 simulator for my project
But Lock/Makefile builds the arduino code