sudar/Arduino-Makefile #352: OS X Missbehaviour

itay-grudev, 2015-05-15:
The correct path of the ```ARDUINO_DIR``` on OSX should be:
```
ARDUINO_DIR   = /Applications/Arduino.app/Contents/Java
```
Currently the ```Java``` directory resides directly under the ```Contents``` directory instead of ```Contents/Resources```. This issue relates to the OSX examples in your ```README```.

Additionaly, when using ```BOARD_TAG = mega2560``` I get a:
```
fatal error: Arduino.h: No such file or directory
```
It took me half an hour to trace the issue and at the end I just had to replace ```mega2560``` with just ```mega```. Apparently the "stripping" of the number in your Makefile doesn't work or was not used in one of the steps. You can just fix the ```README``` or hunt down the bug.

And this last one __IS A BUG__. I couldn't really trace it's source but I sort of got around that. When running the make utility, all of the compiled files were stored in a ```build-uno``` directory instead of ```build-mega```. Because of that I had to copy the contents of ```build-uno``` to ```build-mega``` multiple times before a successful ```make```.

P.S. Since on OSX you need to install the AVR GCC tools and one of the best sources of that is [CrossPack](https://www.obdev.at/products/crosspack/index.html), can you just document it as a requirement on OSX so other users won't have to search for solutions for that either. If you do that, update the ```AVR_TOOLS_DIR``` to ```/usr/local/CrossPack-AVR``` in the OSX example Makefile.

ladislas, 2015-05-15:
I assume you are using Arduino IDE 1.6.3 or later, so regarding the `ARDUINO_DIR` you are right.

Regarding the `BOARD_TAG`, you must use the following:

```
BOARD_TAG         = mega
BOARD_SUB         = atmega2560
```

> Since on OSX you need to install the AVR GCC Tools and one of the best source for that is CrossPack

I don't totally agree, even though I have to admit I've never used CrossPack. The simplest way for OS X users is to install [homebrew](brew.sh) and then install https://github.com/osx-cross/homebrew-avr.  
By doing so, you won't have to update 'AVR_TOOLS_DIR' and you will get more recent versions of the packages than CrossPack.

itay-grudev, 2015-05-15:
I eventually figured it out. Will you please update the ```README``` so future users won't waste their time as I did?

sej7278, 2015-05-15:
you don't need the avr tools, they are bundled with the ide. if you don't install the ide then where do you get board definitions and core libraries from? on linux you can just install arduino-core, but don't think the same goes for homebrew/crosspack.

the path to ARDUINO_DIR is not wrong, its just in the readme its referring to the 1.0.x setup.

there's code in Common.mk that should automatically find it, so you don't need to set it:

```
AUTO_ARDUINO_DIR := $(firstword \
        $(call dir_if_exists,/usr/share/arduino) \
        $(call dir_if_exists,/Applications/Arduino.app/Contents/Resources/Java) \
        $(call dir_if_exists,/Applications/Arduino.app/Contents/Java) )
```

the info about BOARD_SUB is already in readme:

```
On Arduino 1.5.x installs, you should set the architecture to either avr or sam and if using a submenu CPU, then also set that:

    ARCHITECTURE  = avr
    BOARD_SUB     = atmega168
```

we'll have to have a look into updating the docs a bit more for 1.6 i guess.

itay-grudev, 2015-05-15:
Thank you a lot. I managed to get it running eventually, but it took me 2 hours. Can you just correct the ```mega``` examples in the ```README```, so other users won't waste their time.
Thank you.

closed 2015-05-15 by itay-grudev

sudar, 2015-05-16:
@itay-grudev We have updated the README file to be more explicitly about the changes needed for Arduino 1.5+ See #353

Let's us know if there was anything else that was confusing to you.

PR #353 (2015-05-16): Updated some documentation to clarify 1.5+ settings.
Fixes #352 

Not sure what else we can do to clarify things - most variables are auto-detected anyway, really its just BOARD_TAG / BOARD_SUB that is 1.5+ specific
commit eaa8f1fe (2015-05-16): Updated some documentation to clarify 1.5+ settings.

Fixes #352