sudar/Arduino-Makefile #356: Windows/Cygwin: Arduino.h: No such file or directory

mp02951, 2015-05-26:
I tried following all the instructions to get this to work in Windows but I'm at my wit's end so please forgive me. I am running Cygwin on 64-bit Windows 8.1. In Cygwin, I have pyserial, perl, and Python 2.7.9, and make all installed and running correctly. I have Arduino 1.6.4 installed. I made a symbolic link that goes to Arduino at C:\Arduino. 

I was a little confused on Cygwin installation so I just downloaded from github and put everything in my cygwin folder and didn't do anything else, edit any files etc. 

I tried making blink. I put in the command:
```
     mp0295@MikePCl ~/Arduino-Makefile/examples/Blink
     $ make ARDUINO_DIR=../../../../../cygdrive/C/Arduino
```

This is what I then get:
````
-------------------------
Arduino.mk Configuration:
- [AUTODETECTED]       CURRENT_OS = WINDOWS
- [USER]               ARDUINO_DIR = ../../../../../cygdrive/C/Arduino
- [COMPUTED]           ARDMK_DIR = /home/mp029_000/Arduino-Makefile (relative to Common.mk)
- [AUTODETECTED]       ARDUINO_VERSION = 164
- [DEFAULT]            ARCHITECTURE = avr
- [DEFAULT]            VENDOR = arduino
- [DEFAULT]            ARDUINO_SKETCHBOOK =
- [BUNDLED]            AVR_TOOLS_DIR = ../../../../../cygdrive/C/Arduino/hardware/tools/avr (in Arduino distribution)
- [COMPUTED]           ARDUINO_LIB_PATH = ../../../../../cygdrive/C/Arduino/libraries (from ARDUINO_DIR)
- [COMPUTED]           ARDUINO_PLATFORM_LIB_PATH = ../../../../../cygdrive/C/Arduino/hardware/arduino/avr/libraries (from ARDUINO_DIR)
- [COMPUTED]           ARDUINO_VAR_PATH = ../../../../../cygdrive/C/Arduino/hardware/arduino/avr/variants (from ARDUINO_DIR)
- [COMPUTED]           BOARDS_TXT = ../../../../../cygdrive/C/Arduino/hardware/arduino/avr/boards.txt (from ARDUINO_DIR)
- [DEFAULT]            USER_LIB_PATH = /libraries (in user sketchbook)
- [DEFAULT]            PRE_BUILD_HOOK = pre-build-hook.sh
- [USER]               BOARD_TAG = uno
- [COMPUTED]           CORE = arduino (from build.core)
- [COMPUTED]           VARIANT = standard (from build.variant)
- [COMPUTED]           OBJDIR = build-uno (from BOARD_TAG)
- [COMPUTED]           ARDUINO_CORE_PATH = ../../../../../cygdrive/C/Arduino/hardware/arduino/avr/cores/arduino (from ARDUINO_DIR, BOARD_TAG and boards.txt)
- [ASSUMED]            MONITOR_BAUDRATE = 9600
- [DEFAULT]            OPTIMIZATION_LEVEL = s
- [DEFAULT]            MCU_FLAG_NAME = mmcu
- [DEFAULT]            CFLAGS_STD =
- [DEFAULT]            CXXFLAGS_STD =
- [AUTODETECTED]       DEVICE_PATH =
- [DEFAULT]            FORCE_MONITOR_PORT =
- [AUTODETECTED]       Size utility: AVR-aware for enhanced output
- [COMPUTED]           BOOTLOADER_PARENT = ../../../../../cygdrive/C/Arduino/hardware/arduino/avr/bootloaders (from ARDUINO_DIR)
- [COMPUTED]           ARDMK_VERSION = 1.5
- [COMPUTED]           CC_VERSION = 4.8.1 (avr-gcc)
-------------------------
mkdir -p build-uno
../../../../../cygdrive/C/Arduino/hardware/tools/avr/bin/avr-g++ -x c++ -include Arduino.h -MMD -c -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=164 -DARDUINO_ARCH_AVR -D__PROG_TYPES_COMPAT__ -I../../../../../cygdrive/C/Arduino/hardware/arduino/avr/cores/arduino -I../../../../../cygdrive/C/Arduino/hardware/arduino/avr/variants/standard    -Wall -ffunction-sections -fdata-sections -Os -fno-exceptions  Blink.ino -o build-uno/Blink.o
cc1plus.exe: fatal error: Arduino.h: No such file or directory
compilation terminated.
../../Arduino.mk:1215: recipe for target 'build-uno/Blink.o' failed
make: *** [build-uno/Blink.o] Error 1
````

What am I doing wrong?

mp02951, 2015-05-26:
Didn't solve it, but for for the hell of it I copied all the Arduino Core files into the same folder the blink makefile. I also put the arduino directory into the makefile so I can stop writing that each time.This is what then happened:

````
mp0295@MikePCl ~/Arduino-Makefile/examples/Blink
$ make
-------------------------
Arduino.mk Configuration:
- [AUTODETECTED]       CURRENT_OS = WINDOWS
- [USER]               ARDUINO_DIR = ../../../../../cygdrive/C/Arduino
- [COMPUTED]           ARDMK_DIR = /home/mp0295/Arduino-Makefile (relative to Common.mk)
- [AUTODETECTED]       ARDUINO_VERSION = 164
- [DEFAULT]            ARCHITECTURE = avr
- [DEFAULT]            VENDOR = arduino
- [DEFAULT]            ARDUINO_SKETCHBOOK =
- [BUNDLED]            AVR_TOOLS_DIR = ../../../../../cygdrive/C/Arduino/hardware/tools/avr (in Arduino distribution)
- [COMPUTED]           ARDUINO_LIB_PATH = ../../../../../cygdrive/C/Arduino/libraries (from ARDUINO_DIR)
- [COMPUTED]           ARDUINO_PLATFORM_LIB_PATH = ../../../../../cygdrive/C/Arduino/hardware/arduino/avr/libraries (from ARDUINO_DIR)
- [COMPUTED]           ARDUINO_VAR_PATH = ../../../../../cygdrive/C/Arduino/hardware/arduino/avr/variants (from ARDUINO_DIR)
- [COMPUTED]           BOARDS_TXT = ../../../../../cygdrive/C/Arduino/hardware/arduino/avr/boards.txt (from ARDUINO_DIR)
- [DEFAULT]            USER_LIB_PATH = /libraries (in user sketchbook)
- [DEFAULT]            PRE_BUILD_HOOK = pre-build-hook.sh
- [USER]               BOARD_TAG = uno
- [COMPUTED]           CORE = arduino (from build.core)
- [COMPUTED]           VARIANT = standard (from build.variant)
- [COMPUTED]           OBJDIR = build-uno (from BOARD_TAG)
- [USER]               ARDUINO_CORE_PATH = ../../../../../cygdrive/C/Arduino/hardware/arduino/avr/cores/arduino
- [ASSUMED]            MONITOR_BAUDRATE = 9600
- [DEFAULT]            OPTIMIZATION_LEVEL = s
- [DEFAULT]            MCU_FLAG_NAME = mmcu
- [DEFAULT]            CFLAGS_STD =
- [DEFAULT]            CXXFLAGS_STD =
- [AUTODETECTED]       DEVICE_PATH =
- [DEFAULT]            FORCE_MONITOR_PORT =
- [AUTODETECTED]       Size utility: AVR-aware for enhanced output
- [COMPUTED]           BOOTLOADER_PARENT = ../../../../../cygdrive/C/Arduino/hardware/arduino/avr/bootloaders (from ARDUINO_DIR)
- [COMPUTED]           ARDMK_VERSION = 1.5
- [COMPUTED]           CC_VERSION = 4.8.1 (avr-gcc)
-------------------------
mkdir -p build-uno
../../../../../cygdrive/C/Arduino/hardware/tools/avr/bin/avr-g++ -MMD -c -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=164 -DARDUINO_ARCH_AVR -D__PROG_TYPES_COMPAT__ -I../../../../../cygdrive/C/Arduino/hardware/arduino/avr/cores/arduino -I../../../../../cygdrive/C/Arduino/hardware/arduino/avr/variants/standard    -Wall -ffunction-sections -fdata-sections -Os -fno-exceptions  main.cpp -o build-uno/main.o
../../../../../cygdrive/C/Arduino/hardware/tools/avr/bin/avr-g++ -MMD -c -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=164 -DARDUINO_ARCH_AVR -D__PROG_TYPES_COMPAT__ -I../../../../../cygdrive/C/Arduino/hardware/arduino/avr/cores/arduino -I../../../../../cygdrive/C/Arduino/hardware/arduino/avr/variants/standard    -Wall -ffunction-sections -fdata-sections -Os -fno-exceptions  WMath.cpp -o build-uno/WMath.o
../../../../../cygdrive/C/Arduino/hardware/tools/avr/bin/avr-g++ -MMD -c -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=164 -DARDUINO_ARCH_AVR -D__PROG_TYPES_COMPAT__ -I../../../../../cygdrive/C/Arduino/hardware/arduino/avr/cores/arduino -I../../../../../cygdrive/C/Arduino/hardware/arduino/avr/variants/standard    -Wall -ffunction-sections -fdata-sections -Os -fno-exceptions  new.cpp -o build-uno/new.o
../../../../../cygdrive/C/Arduino/hardware/tools/avr/bin/avr-g++ -MMD -c -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=164 -DARDUINO_ARCH_AVR -D__PROG_TYPES_COMPAT__ -I../../../../../cygdrive/C/Arduino/hardware/arduino/avr/cores/arduino -I../../../../../cygdrive/C/Arduino/hardware/arduino/avr/variants/standard    -Wall -ffunction-sections -fdata-sections -Os -fno-exceptions  USBCore.cpp -o build-uno/USBCore.o
../../../../../cygdrive/C/Arduino/hardware/tools/avr/bin/avr-g++ -MMD -c -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=164 -DARDUINO_ARCH_AVR -D__PROG_TYPES_COMPAT__ -I../../../../../cygdrive/C/Arduino/hardware/arduino/avr/cores/arduino -I../../../../../cygdrive/C/Arduino/hardware/arduino/avr/variants/standard    -Wall -ffunction-sections -fdata-sections -Os -fno-exceptions  abi.cpp -o build-uno/abi.o
../../../../../cygdrive/C/Arduino/hardware/tools/avr/bin/avr-g++ -MMD -c -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=164 -DARDUINO_ARCH_AVR -D__PROG_TYPES_COMPAT__ -I../../../../../cygdrive/C/Arduino/hardware/arduino/avr/cores/arduino -I../../../../../cygdrive/C/Arduino/hardware/arduino/avr/variants/standard    -Wall -ffunction-sections -fdata-sections -Os -fno-exceptions  Stream.cpp -o build-uno/Stream.o
../../../../../cygdrive/C/Arduino/hardware/tools/avr/bin/avr-g++ -MMD -c -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=164 -DARDUINO_ARCH_AVR -D__PROG_TYPES_COMPAT__ -I../../../../../cygdrive/C/Arduino/hardware/arduino/avr/cores/arduino -I../../../../../cygdrive/C/Arduino/hardware/arduino/avr/variants/standard    -Wall -ffunction-sections -fdata-sections -Os -fno-exceptions  HID.cpp -o build-uno/HID.o
../../../../../cygdrive/C/Arduino/hardware/tools/avr/bin/avr-g++ -MMD -c -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=164 -DARDUINO_ARCH_AVR -D__PROG_TYPES_COMPAT__ -I../../../../../cygdrive/C/Arduino/hardware/arduino/avr/cores/arduino  [... cut ...] pp -o build-uno/IPAddress.o
IPAddress.cpp:20:21: fatal error: Arduino.h: No such file or directory
 #include <Arduino.h>
                     ^
compilation terminated.
../../Arduino.mk:1197: recipe for target 'build-uno/IPAddress.o' failed
make: *** [build-uno/IPAddress.o] Error 1
````
This is what the make file looks like. 
````
# Arduino Make file. Refer to https://github.com/sudar/Arduino-Makefile

BOARD_TAG    = uno
ARDUINO_DIR=../../../../../cygdrive/C/Arduino

include ../../Arduino.mk
````

pieterverberne, 2016-12-04:
I have the same problem. Compiling with -v gives you a hint:

<snip>
ignoring nonexistent directory "d:\program files (x86)\arduino\hardware\tools\avr\bin\../lib/gcc/avr/4.9.2/../../../../avr/include/c++/4.9.2"
<snip>
#include <...> search starts here:
 d:\program files (x86)\arduino\hardware\tools\avr\bin\../lib/gcc/avr/4.9.2/include
 d:\program files (x86)\arduino\hardware\tools\avr\bin\../lib/gcc/avr/4.9.2/include-fixed
 d:\program files (x86)\arduino\hardware\tools\avr\bin\../lib/gcc/avr/4.9.2/../../../../avr/include

The path names are correct (and ugly) but they just don't work.

gsimard, 2017-02-15:
Just had the same problem. I think this is related to how Windows/Cygwin handles drive paths (c:, d:, etc.). Try not having to use a relative path that must go down to /cygdrive.

For me making sure the arduino folder and the projects were in the same drive avoided this issue.

I notice one of you has spaces in their paths and this may be another root cause. Try making a Windows symlink like d:\arduino that points to d:\program files (x86)\arduino and use that in your relative paths instead.

Good luck.