sudar/Arduino-Makefile #363: For what files is the arduino package required?

Simpfally, 2015-07-05:
It's quite difficult to get the arduino package installed on a raspberry, I finally managed to do it after many errors.
It'd be great to be able to download all the necessary things without having to install arduino completely.
Looks like it only use some txt files and the libraries to link. Avr-g++ and others are available in other places.

sej7278, 2015-07-05:
that's the problem with raspbian, if they'd just use regular debian it would be simple.

this should work on the pi, it won't install the dreadful ide (most of which doesn't work on the pi anyway) but will install the boards, avr-gcc, avrdude etc.

```sudo apt-get install arduino-core python-serial arduino-mk```

Simpfally, 2015-07-06:
Oh sorry forgot to mention I'm on archlinux. I guess I'll use arduino-core next time. I don't really need the java runtime.. But yeah, now it works so it's all good!

sej7278, 2015-07-06:
cool, i assume you found the AUR package? could you close the issue please?

https://aur.archlinux.org/packages/arduino-mk/

Simpfally, 2015-07-06:
The AUR package doesn't support arm through, you have to change the PKGBUILD and then download all the avr tools somewhere else, but it works.

closed 2015-07-06 by Simpfally

sej7278, 2015-07-06:
arduino-mk is a source package so architecture is irrelevant, but i guess you need to get your dependencies (avrdude, arv-gcc etc.) for the pi. its straightforward on raspbian on the pi and debian on the beaglebone, so i guess its an arch issue.

Simpfally, 2015-07-06:
Yup, only the arduino package was a bit problematic from the AUR. Had to edit the pkgbuild to get some files since executables wouldn't work anyway, that's why I created this issue/question.

sej7278, 2015-07-06:
yes the ide itself is a problem on arm - requires java and some binaries for serial comms that are x86 only.

on debian platforms at least you have the arduino-core package which is avr-g++, avr-libc, avrdude, the boards.txt etc. but not the ide or java dependency. makes a good mix with arduino-mk, as you don't need a gui etc.

reporting stuff to the arduino guys is a bit hit and miss - they seem to be totally disinterested in linux, rarely test anything on macosx and both platforms suffer in favour of windows (like this crazy business of putting arm compilers in home directories).

Simpfally, 2015-07-06:
Yeah, I think most of the users are on windows, they try to make it works for their target audience at least.