sudar/Arduino-Makefile #364: missing wiring_pulse.S from compilation

thinkberg, 2015-07-08:
MacOS X 10.10.4
avr stuff via homebrew
Arduino IDE 1.6.5

I have tried to compile a sketch which uses pulseIn(). This in turn uses a function from wiring_pulse.S that is not linked/compiled for some reason. The Arduino IDE 1.6.5 compiles it 

Arduino IDE build process:
```
/Applications/Arduino.app/Contents/Java/hardware/tools/avr/bin/avr-gcc -c -g -x assembler-with-cpp -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10605 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR -I/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/cores/arduino -I/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/variants/standard /Applications/Arduino.app/Contents/Java/hardware/arduino/avr/cores/arduino/wiring_pulse.S -o /var/folders/c5/9sqmvxp54m3cmjb6hmtw2xlm0000gn/T/build5848034434382423610.tmp/wiring_pulse.S.o 
````

what my make produces (countPulseASM is in wiring_pulse.S which is not mentioned during the build process):
```
/Applications/Arduino.app/Contents/Java/hardware/tools/avr/bin/avr-gcc -mmcu=atmega328p -Wl,--gc-sections -Os -o build-uno/hc-sr04-led.elf build-uno/hc-sr04-led.o build-uno/libcore.a  -lc -lm
build-uno/libcore.a(wiring_pulse.o): In function `pulseIn':
wiring_pulse.c:(.text.pulseIn+0x32): undefined reference to `countPulseASM'
collect2: error: ld returned 1 exit status
make: *** [build-uno/hc-sr04-led.elf] Error 1
```

My makefile:

```
# http://hardwarefun.com/tutorials/compiling-arduino-sketches-using-makefile

#BOARD_TAG=uno
ARDUINO_PORT=/dev/cu.SLAB_USBtoUART

include /usr/local/opt/arduino-mk/Arduino.mk
```

From a quick look, the arduino-makefile looks at all the source types, maybe it misses on wiring_pulse because it exists as wiring_pulse.c and wiring_pulse.S?

sej7278, 2015-07-08:
we need the full output from "make" so we can see your settings, for example is arduino-mk even finding your arduino 1.6.5 installation?

you need to specify a BOARD_TAG (uncomment it) in your Makefile or it won't do anything at all.

thinkberg, 2015-07-08:
```
tahar:hc-sr04-led leo$ make
-------------------------
Arduino.mk Configuration:
- [AUTODETECTED]       CURRENT_OS = MAC 
- [AUTODETECTED]       ARDUINO_DIR = /Applications/Arduino.app/Contents/Java 
- [COMPUTED]           ARDMK_DIR = /usr/local/Cellar/arduino-mk/1.5 (relative to Common.mk)
- [AUTODETECTED]       ARDUINO_VERSION = 165 
- [DEFAULT]            ARCHITECTURE = avr 
- [DEFAULT]            VENDOR = arduino 
- [AUTODETECTED]       ARDUINO_PREFERENCES_PATH = /Users/leo/Library/Arduino15/preferences.txt 
- [AUTODETECTED]       ARDUINO_SKETCHBOOK = /Users/leo/Documents/Source/ubirch/ubirch-sketches (from arduino preferences file)
- [BUNDLED]            AVR_TOOLS_DIR = /Applications/Arduino.app/Contents/Java/hardware/tools/avr (in Arduino distribution)
- [COMPUTED]           ARDUINO_LIB_PATH = /Applications/Arduino.app/Contents/Java/libraries (from ARDUINO_DIR)
- [COMPUTED]           ARDUINO_PLATFORM_LIB_PATH = /Applications/Arduino.app/Contents/Java/hardware/arduino/avr/libraries (from ARDUINO_DIR)
- [COMPUTED]           ARDUINO_VAR_PATH = /Applications/Arduino.app/Contents/Java/hardware/arduino/avr/variants (from ARDUINO_DIR)
- [COMPUTED]           BOARDS_TXT = /Applications/Arduino.app/Contents/Java/hardware/arduino/avr/boards.txt (from ARDUINO_DIR)
- [DEFAULT]            USER_LIB_PATH = /Users/leo/Documents/Source/ubirch/ubirch-sketches/libraries (in user sketchbook)
- [DEFAULT]            PRE_BUILD_HOOK = pre-build-hook.sh 
- [DEFAULT]            BOARD_TAG = uno 
- [COMPUTED]           CORE = arduino (from build.core)
- [COMPUTED]           VARIANT = standard (from build.variant)
- [COMPUTED]           OBJDIR = build-uno (from BOARD_TAG)
- [COMPUTED]           ARDUINO_CORE_PATH = /Applications/Arduino.app/Contents/Java/hardware/arduino/avr/cores/arduino (from ARDUINO_DIR, BOARD_TAG and boards.txt)
- [DETECTED]           MONITOR_BAUDRATE = 115200  (in sketch)
- [DEFAULT]            OPTIMIZATION_LEVEL = s 
- [DEFAULT]            MCU_FLAG_NAME = mmcu 
- [DEFAULT]            CFLAGS_STD =  
- [DEFAULT]            CXXFLAGS_STD =  
- [COMPUTED]           DEVICE_PATH = /dev/cu.SLAB_USBtoUART (from MONITOR_PORT)
- [DEFAULT]            FORCE_MONITOR_PORT =  
- [AUTODETECTED]       Size utility: AVR-aware for enhanced output
- [COMPUTED]           BOOTLOADER_PARENT = /Applications/Arduino.app/Contents/Java/hardware/arduino/avr/bootloaders (from ARDUINO_DIR)
- [COMPUTED]           ARDMK_VERSION = 1.5 
- [COMPUTED]           CC_VERSION = 4.8.1 (avr-gcc)
-------------------------
mkdir -p build-uno
/Applications/Arduino.app/Contents/Java/hardware/tools/avr/bin/avr-g++ -x c++ -include Arduino.h -MMD -c -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=165 -DARDUINO_ARCH_AVR -D__PROG_TYPES_COMPAT__ -I/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/cores/arduino -I/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/variants/standard    -Wall -ffunction-sections -fdata-sections -Os -fno-exceptions  hc-sr04-led.ino -o build-uno/hc-sr04-led.o
/Applications/Arduino.app/Contents/Java/hardware/tools/avr/bin/avr-gcc -MMD -c -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=165 -DARDUINO_ARCH_AVR -D__PROG_TYPES_COMPAT__ -I/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/cores/arduino -I/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/variants/standard    -Wall -ffunction-sections -fdata-sections -Os  /Applications/Arduino.app/Contents/Java/hardware/arduino/avr/cores/arduino/WInterrupts.c -o build-uno/core/WInterrupts.o
/Applications/Arduino.app/Contents/Java/hardware/tools/avr/bin/avr-gcc -MMD -c -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=165 -DARDUINO_ARCH_AVR -D__PROG_TYPES_COMPAT__ -I/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/cores/arduino -I/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/variants/standard    -Wall -ffunction-sections -fdata-sections -Os  /Applications/Arduino.app/Contents/Java/hardware/arduino/avr/cores/arduino/hooks.c -o build-uno/core/hooks.o
/Applications/Arduino.app/Contents/Java/hardware/tools/avr/bin/avr-gcc -MMD -c -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=165 -DARDUINO_ARCH_AVR -D__PROG_TYPES_COMPAT__ -I/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/cores/arduino -I/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/variants/standard    -Wall -ffunction-sections -fdata-sections -Os  /Applications/Arduino.app/Contents/Java/hardware/arduino/avr/cores/arduino/wiring.c -o build-uno/core/wiring.o
/Applications/Arduino.app/Contents/Java/hardware/tools/avr/bin/avr-gcc -MMD -c -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=165 -DARDUINO_ARCH_AVR -D__PROG_TYPES_COMPAT__ -I/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/cores/arduino -I/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/variants/standard    -Wall -ffunction-sections -fdata-sections -Os  /Applications/Arduino.app/Contents/Java/hardware/arduino/avr/cores/arduino/wiring_analog.c -o build-uno/core/wiring_analog.o
/Applications/Arduino.app/Contents/Java/hardware/tools/avr/bin/avr-gcc -MMD -c -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=165 -DARDUINO_ARCH_AVR -D__PROG_TYPES_COMPAT__ -I/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/cores/arduino -I/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/variants/standard    -Wall -ffunction-sections -fdata-sections -Os  /Applications/Arduino.app/Contents/Java/hardware/arduino/avr/cores/arduino/wiring_digital.c -o build-uno/core/wiring_digital.o
[... cut ...] g.o  build-uno/core/abi.o  build-uno/core/main.o  build-uno/core/new.o         
/Applications/Arduino.app/Contents/Java/hardware/tools/avr/bin/avr-gcc -mmcu=atmega328p -Wl,--gc-sections -Os -o build-uno/hc-sr04-led.elf build-uno/hc-sr04-led.o build-uno/libcore.a  -lc -lm
build-uno/libcore.a(wiring_pulse.o): In function `pulseIn':
wiring_pulse.c:(.text.pulseIn+0x32): undefined reference to `countPulseASM'
collect2: error: ld returned 1 exit status
make: *** [build-uno/hc-sr04-led.elf] Error 1
```

sej7278, 2015-07-08:
i'm not sure, i've got a feeling they made some changes to pulsein in 1.6.5, maybe try 1.6.4 or somesuch?

thinkberg, 2015-07-09:
just tested as well, yes 1.6.4 works.

sej7278, 2015-07-09:
i can't figure out why the .S file is being ignored by our Makefile, I thought it might be that we were only building .S files in libraries not the core, but this makes no difference:

```diff
diff --git a/Arduino.mk b/Arduino.mk
index aabe8e7..5007cda 100644
--- a/Arduino.mk
+++ b/Arduino.mk
@@ -797,6 +797,7 @@ ifeq ($(strip $(NO_CORE)),)
         CORE_C_SRCS     = $(wildcard $(ARDUINO_CORE_PATH)/*.c)
         CORE_C_SRCS    += $(wildcard $(ARDUINO_CORE_PATH)/avr-libc/*.c)
         CORE_CPP_SRCS   = $(wildcard $(ARDUINO_CORE_PATH)/*.cpp)
+        CORE_AS_SRCS    = $(wildcard $(ARDUINO_CORE_PATH)/*.S)
 
         ifneq ($(strip $(NO_CORE_MAIN_CPP)),)
             CORE_CPP_SRCS := $(filter-out %main.cpp, $(CORE_CPP_SRCS))
```

peplin, 2015-07-09:
> i guess its a problem with wiring_pulse.c and wiring_pulse.S becoming the same object file wiring_pulse.o

I ran into that problem back in #255 so it's quite possible that's also the root cause here.

ladislas, 2015-07-09:
May be we should reconsider merging #266?

sej7278, 2015-07-09:
i'm not convinced its the same filename problem, i might have a go at rebasing it. 

the .S file never seems to make it into CORE_OBJS or CORE_OBJ_FILES even though CORE_AS_SRCS seems to list it.

sej7278, 2015-07-09:
this diff seems to fix the pulseIn() issue but breaks SoftwareSerial, any ideas?

https://gist.github.com/sej7278/0df5220133f40ed0a368

sej7278, 2015-07-09:
I've rebased and edited @peplin's PR #266 into PR #365 (been a while!)

This seems to fix the issue with wiring_pulse.S

Could we test if it fixes the chipKIT issue too from #255 please?

sej7278, 2015-07-09:
doesn't allow mpide-0023-linux64-20140821 to build, is there anything else needed for the newer chipkit installs?

```
build-mega_pic32/libcore.a(WSystem.c.o): In function `initIntVector':
WSystem.c:(.text.initIntVector+0x0): undefined reference to `_image_header_info'
WSystem.c:(.text.initIntVector+0x4): undefined reference to `_image_header_info'
collect2: ld returned 1 exit status
/data3/useful_software/arduino/arduino-mk/my_git/Arduino-Makefile/Arduino.mk:1410: recipe for target 'build-mega_pic32/blink.elf' failed
make: *** [build-mega_pic32/blink.elf] Error 1
```

sej7278, 2015-07-09:
chipkit mpide-0023-linux64-20140821 now works (had to add space between -T and linker script)

closed 2015-07-12 by sudar

thinkberg, 2015-07-13:
Awesome, thanks!

tschannerl, 2017-01-17:
Hi, exactily, I change version 1.6.9 to 1.6.4 and working fine.... Worse than in new version 1.8.1 is with the same problem.....

chrisspen, 2017-02-06:
Why was this closed? I'm still getting this error with Arduino 1.8.1 and the current 1.5 release of Arduino-Makefile. Is there a workaround?

mdasyg, 2018-10-25:
I am still getting the error on Arduino 1.8.7 . I made a temporary fix:

Append the following lines to the file /opt/arduino/hardware/arduino/avr/cores/arduino/wiring_pulse.c 

---CUT HERE---
#include <Arduino.h>
#include <stdlib.h>

unsigned long countPulseASM(volatile uint8_t *port, uint8_t bit, uint8_t stateMask, unsigned long maxloops)
 {
     unsigned long width = 0;
     // wait for any previous pulse to end
     while ((*port & bit) == stateMask)
         if (--maxloops == 0)
             return 0;

     // wait for the pulse to start
     while ((*port & bit) != stateMask)
         if (--maxloops == 0)
             return 0;

      // wait for the pulse to stop
      while ((*port & bit) == stateMask) {
          if (++width == maxloops)
              return 0;
      }
      return width;
 }

---CUT HERE---

With this snipset of code I was able to compile my Nanpy firmware with 1.8.7 on Linux AMD64 via command line (arduino-mk) and resolved the problem of undefined reference to `countPulseASM'.

Hope that this helps you.

commit 3dd06a6e (2015-07-12): Merge pull request #365 from sej7278/master

Preserve source extension for object files.

Fix #256 
Fix #266 
Fix #364