sudar/Arduino-Makefile #366: No Rule to Make Target

abferm, 2015-07-14:
Hey guys, I’m having some issues. I switched from the debian version to directly pulling the git repo, and am having issues now that I didn’t have before. I am trying to compile .ino files that I had working before. I forked at commit 7b11860ba8f590ceba1fa46dddf45d8d8b6d6c95 to https://github.com/abferm/Arduino-Makefile in order to implement the ability to specify optimization level. I am running this on Debian 8 32-bit.

————————-
Arduino.mk Configuration:
– [AUTODETECTED] CURRENT_OS = LINUX
– [USER] ARDUINO_DIR = /usr/share/arduino
– [USER] ARDMK_DIR = ../../Arduino-Makefile
– [AUTODETECTED] ARDUINO_VERSION = 105
– [DEFAULT] ARCHITECTURE =
– [DEFAULT] VENDOR = arduino
– [DEFAULT] ARDUINO_SKETCHBOOK =
– [USER] AVR_TOOLS_DIR = /usr
– [COMPUTED] ARDUINO_LIB_PATH = /usr/share/arduino/libraries (from ARDUINO_DIR)
– [COMPUTED] ARDUINO_VAR_PATH = /usr/share/arduino/hardware/arduino//variants (from ARDUINO_DIR)
– [COMPUTED] BOARDS_TXT = /usr/share/arduino/hardware/arduino//boards.txt (from ARDUINO_DIR)
– [USER] USER_LIB_PATH = `pwd`/lib
– [DEFAULT] PRE_BUILD_HOOK = pre-build-hook.sh
– [USER] BOARD_TAG = uno
– [COMPUTED] CORE = arduino (from build.core)
– [COMPUTED] VARIANT = standard (from build.variant)
– [USER] OBJDIR = `pwd`/bin/uno/Code
– [COMPUTED] ARDUINO_CORE_PATH = /usr/share/arduino/hardware/arduino//cores/arduino (from ARDUINO_DIR, BOARD_TAG and boards.txt)
– [USER] MONITOR_BAUDRATE = 115200
– [DEFAULT] OPTIMIZATION_LEVEL = s
– [DEFAULT] MCU_FLAG_NAME = mmcu
– [USER] CFLAGS_STD = -std=gnu11
– [USER] CXXFLAGS_STD = -std=gnu++11
– [COMPUTED] DEVICE_PATH = /dev/ttyACM0 (from MONITOR_PORT)
– [DEFAULT] FORCE_MONITOR_PORT =
– [AUTODETECTED] Size utility: AVR-aware for enhanced output
– [COMPUTED] BOOTLOADER_PARENT = /usr/share/arduino/hardware/arduino//bootloaders (from ARDUINO_DIR)
– [COMPUTED] ARDMK_VERSION = 1.5
– [COMPUTED] CC_VERSION = 4.8.1 (avr-gcc)
————————-
make: *** No rule to make target ‘elseif.s’, needed by ‘`pwd`/bin/uno/Code/elseif.o’. Stop.

abferm, 2015-07-14:
Solved by running tests/script/bootstrap.sh.

closed 2015-07-14 by abferm

sej7278, 2015-07-14:
don't know where all the pwd's are coming from, but you certainly shouldn't have run bootstrap.sh