sudar/Arduino-Makefile #372: Allow spaces in project path.

MeanEYE, 2015-08-07:
If I save my project in `LCD Test` directory, `makefile` will fail to compile and upload. However if I rename directory to `LCD_Test` everything works properly.

sudar, 2015-08-08:
This Makefile doesn't support directories with spaces, since it breaks lot of scripts that the makefile uses.

Also it is generally a bad practice to have space in directory names anyways :) So would it be possible for you to change it?

MeanEYE, 2015-08-08:
Well, of course it's possible, but it's a bad excuse on their part. Should we limit file names to 8 characters now? Since we are already making an effort of living in the past. Makefile should respect specs of the system in question, and not invent their own rules. Even DOS supported spaces. Gah, I love when developers come up with retarded ideas.

Thanks anyway, I can see this is a wide spread problem and that I should probably start looking for another solution as `make` is not going to cut it anymore. Also, since this is invalid issue report related to this project, am closing it. Thanks for clearing this up.

closed 2015-08-08 by MeanEYE

sudar, 2015-08-08:
> I love when developers come up with retarded ideas.

Well, I can give you the reason why we don't currently support directories with spaces (or other special characters) and you can decide whether it is _retarded_ or not.

This Makefile uses lot of shell scripts and spaces (or other special characters) in the directory names breaks those shell scripts. It would takes us huge amount of effort to go over all those shell scripts and properly quote them and then test them in all the different operating systems that we support. Instead we choose to spend that time to add new features and gently (without calling them _retarded_) tell the users to have rename their directories.

Ohh.. by the way anyone is welcomed to help us the developers (who have come up with _retarded_ ideas) by coming up with a clever idea and/or a (well tested) pull request to add support for directories with spaces (or any other special characters)

ladislas, 2015-08-08:
@sudar :+1: 

@MeanEYE :-1:

MeanEYE, 2015-08-08:
I thought `make` tool itself was responsible for lack of support, but it looks like that's not the case. _Retarded_ might be too harsh of a word to use to describe this issue but telling people to adapt and not fix the issue because it's too much work is not a good policy in my opinion.

If you are, by any chance, offended by my statement, I welcome you to submit a patch to any open source project which removes support for spaces in paths and see how that goes and if they are willing to accept it. Should such a patch be accepted I am more than willing to publicly apologize for my choice of words. Until then I stand by my statement as we do live in 21st century which tends to have support for many things, including spaces in paths, regardless if it's a bad policy or not.

I guess I have no other choice but to try and maintain solution with nest of symlinks hoping that one day you don't find those too hard as well or remove support for underscores and dashes. At that point I will have no other choice but to give up on this project.

You should note though that such approach of "adapt or gtfo" greatly reduces chances of people contributing especially considering this project wasn't copy pasted and source came to be line by line. So choice not to support spaces was either a neglect from day one or deliberate choice. But what do I know, I still use spaces in my paths.

All the best!

kautzz, 2015-08-08:
the name says it all

ladislas, 2015-08-08:
> If you are, by any chance, offended by my statement, I welcome you to submit a patch to any open source project which removes support for spaces in paths and see how that goes and if they are willing to accept it.

That's not about removing stuff, it's about adding stuff that are not there yet. Spaces are not supported yet, but you are more than welcome to contribute to this project by adding this feature.

The guidelines for contributing are pretty explicit and a lot of people have already done it, so it should not be too difficult for you.

As for changing the name of your directory, I really think it should be a little less difficult and time consuming than re-writing all the scripts and Makefiles. But feel free to start from scratch and let us know when you come up with a Makefile as robust as this one...

sej7278, 2015-08-10:
we have tried before and adding support for spaces on windows broke other platforms due to the way quoting works. it doesn't really make sense to uses spaces in paths and its an easy to fix by removing the space, but we welcome PR's, not abuse.

GreyGnome, 2015-08-11:
Gee, give @MeanEYE his money back so he stops complaining. He obviously paid too much for the software which is not meeting his expectations. Isn't there a money-back guarantee with this thing? Let me speak to the manager!

And quit your day jobs and give me everything I want. Now. I don't care how much time you have or how you feed yourself. You can always stop sleeping, and dig for food in the dumpster. Geez, what's the matter with you people?

...Oh yeah:  :-P