sudar/Arduino-Makefile #373: How to Compile with DMellis' ATtiny work?

GreyGnome, 2015-08-20:
Hello,
I am using DMellis' work at https://github.com/damellis/attiny and I am unable to compile. I am in a twisty maze of ugly hacking and I wonder if there isn't a few couple of simple steps that I should be doing instead to get this to work? I don't understand what is required, even after Googling. Below is my Makefile. Any help is appreciated.
```
# Arduino Make file. Refer to https://github.com/sudar/Arduino-Makefile

AVR_TOOLS_DIR = /usr
CFLAGS_STD = -g -I /home/schwager/.arduino15/packages/attiny/hardware/avr/1.0.1/variants/tiny8
CXXFLAGS_STD = -g -I /home/schwager/.arduino15/packages/attiny/hardware/avr/1.0.1/variants/tiny8

# if you have placed the alternate core in your sketchbook directory, then you can just mention the core name alone.
ALTERNATE_CORE = attiny
# If not, you might have to include the full path.
#ALTERNATE_CORE_PATH = /home/schwager/bin/arduino-1.6.4/hardware/arduino/avr/cores/
ALTERNATE_CORE_PATH = /home/schwager/.arduino15/packages/attiny/hardware/avr/1.0.1
ARDUINO_VAR_PATH = /home/schwager/Projects/Arduino/attiny-ide-1.6.x/attiny/avr/variants
ARDUINO_HEADER = /home/schwager/bin/arduino-1.6.4/hardware/arduino/avr/cores/arduino/Arduino.h

BOARD_TAG    = attiny85
ISP_PORT = /dev/ttyACM*

#include $(ARDMK_DIR)/Arduino.mk  # the variable is unrecognized.
include /usr/share/arduino/Arduino.mk

# !!! Important. You have to use make ispload to upload when using ISP programmer

```

The error is:
```
In file included from /home/schwager/bin/arduino-1.6.4/hardware/arduino/avr/cores/arduino/USBAPI.h:27:0,
                 from /home/schwager/bin/arduino-1.6.4/hardware/arduino/avr/cores/arduino/Arduino.h:224,
                 from <command-line>:0:
/usr/avr/include/util/delay.h: In function 'void _delay_ms(double)':
/usr/avr/include/util/delay.h:151:17: error: expected primary-expression before ')' token
  __tmp = ((F_CPU) / 1e3) * __ms;
                 ^
```

GreyGnome, 2015-08-20:
...Whoops! Looks like all that remained was to include "F_CPU=100000" in the Makefile, and I'm good to go! Sorry!

closed 2015-08-20 by GreyGnome

sej7278, 2015-08-20:
its actually all explained in the Blink example:

https://github.com/sudar/Arduino-Makefile/blob/master/examples/Blink/Makefile#L101

GreyGnome, 2015-08-20:
Nice, thanks for the update! That's even better :-)