sudar/Arduino-Makefile #374: ATtiny85 creates a .elf file named "ATtiny.elf"

GreyGnome, 2015-08-20:
Hello,
When I compile a sketch for ATtiny85, the file it creates in the build directory is named "ATtiny.elf". The sketch is named "ATtinyBlink.ino" so I would hope it would create "ATtinyBlink.elf". I have a utility that I use to dump the object file and look at the assembler, and it works fine on the ATmega328 and 2560 but here it breaks because of the file name difference. Is it something I've done? Here's some of the gcc output which shows where it's creating the elf file:
```
/usr/bin/avr-gcc -mmcu=attiny85 -Wl,--gc-sections -Os -o build-attiny85/ATtiny.elf build-attiny85/ATtinyBlin
k.o build-attiny85/libcore.a  -lc -lm
/usr/bin/avr-objcopy -j .eeprom --set-section-flags=.eeprom='alloc,load' \
        --change-section-lma .eeprom=0 -O ihex build-attiny85/ATtiny.elf build-attiny85/ATtiny.eep
/usr/bin/avr-objcopy: --change-section-lma .eeprom=0x0000000000000000 never used
/usr/bin/avr-objcopy -O ihex -R .eeprom build-attiny85/ATtiny.elf build-attiny85/ATtiny.hex

/usr/bin/avr-size --mcu=attiny85 -C --format=avr build-attiny85/ATtiny.elf

```
Here is my Makefile:

```
# Arduino Make file. Refer to https://github.com/sudar/Arduino-Makefile

AVR_TOOLS_DIR = /usr
CFLAGS_STD = -g -I /home/schwager/.arduino15/packages/attiny/hardware/avr/1.0.1/variants/tiny8
CXXFLAGS_STD = -g -I /home/schwager/.arduino15/packages/attiny/hardware/avr/1.0.1/variants/tiny8

# if you have placed the alternate core in your sketchbook directory, then you can just mention the core name alone.
ALTERNATE_CORE = attiny
# If not, you might have to include the full path.
#ALTERNATE_CORE_PATH = /home/schwager/bin/arduino-1.6.4/hardware/arduino/avr/cores/
ALTERNATE_CORE_PATH = /home/schwager/.arduino15/packages/attiny/hardware/avr/1.0.1
ARDUINO_VAR_PATH = /home/schwager/Projects/Arduino/attiny-ide-1.6.x/attiny/avr/variants
ARDUINO_HEADER = /home/schwager/bin/arduino-1.6.4/hardware/arduino/avr/cores/arduino/Arduino.h

BOARD_TAG    = attiny85
ISP_PORT = /dev/ttyACM*

F_CPU = 1000000L

#include $(ARDMK_DIR)/Arduino.mk  # the variable is unrecognized.
include /usr/share/arduino/Arduino.mk

# !!! Important. You have to use make ispload to upload when using ISP programmer
```

sej7278, 2015-08-20:
here's my Makefile:

```
ARDUINO_DIR = /home/user/arduino-1.6.5

ISP_PROG = usbasp
BOARD_TAG = attiny
BOARD_SUB = attiny85
ALTERNATE_CORE = attiny
F_CPU = 16000000L

include /usr/share/arduino/Arduino.mk
````

an that creates build-attiny-attiny85/blink.elf

GreyGnome, 2015-08-21:
Thanks sej. I will give that a try.

sej7278, 2015-08-28:
can we close this, i can't reproduce. did you try with BOARD_SUB defined?

sudar, 2015-08-29:
@GrayGnome,

I am closing it for now.

Feel free to open it if needed.

closed 2015-08-29 by sudar

sej7278, 2015-09-06:
it seems we use the directory name not the sketch name as the output target.

GreyGnome, 2015-09-18:
...It's been a while, I was buys on other projects (ehm, World of Tanks Blitz is a project isn't it? ;-) ), but I discovered that indeed my directory did not share the name of the sketch. I had created "ATtiny" for my "ATtinyBlink.ino" sketch, and I should have called the directory "ATtinyBlink". Once I did that, all was fixed.

The line in the Makefile which sets the TARGET name is:

    TARGET  = $(notdir $(CURDIR))