sudar/Arduino-Makefile #376: Makefile miscompiles IRremote

audetto, 2015-08-30:
Hi,

I am trying to compile IRremote from 

https://github.com/z3t0/Arduino-IRremote

with the Makefile

(the Makefile I use is
========================================================
BOARD_TAG=uno
ARDUINO_DIR=/home/andrea/projects/pi/ide/arduino-1.6.5
ARDUINO_SKETCHBOOK=/home/andrea/Arduino

include /home/andrea/projects/cvs/3rdParty/Arduino-Makefile/Arduino.mk
========================================================
)

and the library IRremote is in /home/andrea/Arduino

(the sketch is
#include <IRremote.h>

int RECV_PIN = 2;
int LED = A5;

IRrecv irrecv(RECV_PIN);

decode_results results;

void setup()
{
  Serial.begin(9600);
  Serial.println("Here we are");
  delay(100);

  pinMode(LED, OUTPUT);
  digitalWrite(LED, HIGH);
  delay(100);

  irrecv.enableIRIn(); // Start the receiver
  delay(100);
}

void loop() {
  if (irrecv.decode(&results))
  {
    Serial.println(results.value, HEX);
    Serial.println(results.decode_type);
    irrecv.resume(); // Receive the next value
  }
  delay(100);
}
)

which is based on one of the examples of the library.
But if I compile with the Makefile nothing works, with arduino 1.6.5 it is good.
With the Makefile it compiles, links, uploads, but it is like some parts of the code are not there.

This started happening at this revision

https://github.com/z3t0/Arduino-IRremote/commit/88e243fe068e06d2ed602fefe6bd5effd07006e1

which seems to have split the library in many pieces.

Compilation log with Arduino 1.6.5

http://pastebin.com/ZZYU2dtC

with this Makefile

http://pastebin.com/3kv4HznZ

sej7278, 2015-08-31:
i suspect this is related to #93 where libraries that include other libraries aren't detected.

you probably need to look for all the #include lines and add them to your main sketch, for example IRremote.h calls ```#include "IRremoteInt.h"```

audetto, 2015-08-31:
The compilation is not the problem.
Nor is linking.

IRremoteInt.h is not an other library, is just an other header file in the same library.
There are no other libraries used.

The problem can only be detected at run time when the sketch fails to run properly.
It gets stuck probably in the call

irrecv.enableIRIn();

and never gets to the end of setup().

The issue #93 seems to be with compilation issues.

audetto, 2015-08-31:
The sketch was the same, just in separate folders just to void inteference.

Till the last rev working the sizes of arduino vs makefile are very similar
8622 vs 8628
and variables are well
445 vs 445

with master on the other hand
10334 vs 9876
451 vs 451

something is different.
But all files seems to be included in both builds, and files cannot be excluded from the build without a linker error.
I will try to ask and replicate the 2 builds and see exactly if the flags passed to gcc are the same.

This is the output from the file in the same location

http://pastebin.com/n2ySD6jJ
http://pastebin.com/gp4GDPY1

audetto, 2015-09-01:
Found it.

The IDE and the Makefile link the sketch i 2 different ways
* IDE: create static lib with Arduino CORE source code, then link all the source code of the sketch AND the used libraries with the static lib
* Makefile: create static lib of Arduino CORE and all libraries used, then link the sketch to this static lib

The difference is where the code for the libraries goes:
* IDE: to the final g++ link command line
* Makefile: to the static lib

Apparently the linker will include all objects explicitly mentioned and pick from the static lib only objects really needed.

In IRremote there is a file isISR.cpp which exports no symbols, but defines the timer/interrupt, so if the linker is not forced it will NOT link it.

Probably this is a weak feature of the linker to rely on, but at the same time the IDE is "the" reference.
What do you guys think?

I will propose to IRremote a change to fix this.

sej7278, 2015-09-01:
ah well done! 

i suspect this is a side effect of the preprocessing that the ide does - basically concatenates all of the libraries and the sketch into one giant c++ file, i guess it guarantees that everything will get linked and users don't have to declare their methods before calling them etc. as you do with "proper" c++.

see https://github.com/arduino/Arduino/wiki/Build-Process and #59, i don't think its something we can simulate with the makefile.

audetto, 2015-09-12:
The fix has been applied to IRremote. Works well now.

closed 2015-09-12 by audetto