sudar/Arduino-Makefile #377: Linking error "collect2.exe: error: ld returned 5 exit statu" when compile Blink project

sunweylee, 2015-09-04:
My working env is WIN8+Cygwin+Your Arduino-Makefile

When I try to build example Blink, I encounter link error like below. Could you please help me out, Thanks a lot! 

../../../../../RaspberryPi/arduino-1.6.5-r2//hardware/tools/avr/bin/avr-gcc -mmcu=atmega328p -Wl,--gc-sections -Os -o build-uno/Blink.elf build-uno/Blink.ino.o build-uno/libcore.a  -lc -lm
collect2.exe: error: ld returned 5 exit status
../../Arduino.mk:1410: recipe for target 'build-uno/Blink.elf' failed

sej7278, 2015-09-04:
i'm not much good with windows, but we need to see your Makefile

also there's seems to be a lot of those type errors here: http://forum.arduino.cc/index.php?topic=310950.0

sunweylee, 2015-09-04:
Thanks for your advice, I just read carefully the post you provided. And try to replace ld.exe with the old version 1.0.5. The linker issue gone. Tricky thing!!!

closed 2015-09-04 by sunweylee