sudar/Arduino-Makefile #378: MONITOR_PORT not working with network connections

Ulli2k, 2015-09-05:
I want to execute the following flash command with your make file
avrdude -b 115200 -p m328p -c arduino -P net:beagle:4000 -Uflash:w:file.hex

But MONITOR_PORT does not support net:beagle:4000!

I additionally want to use make monitor with a different port "net:beagle:4001"

Would it be possible to include that or is there a workaround?

sej7278, 2015-09-05:
what is the error message and can we see your Makefile?

i think it should work if you set:

```Make
FORCE_MONITOR_PORT=1
MONITOR_PORT=net:beagle:4001
```

Ulli2k, 2015-09-05:
With your flags i get the following output

make reset
make[1]: Entering directory '/home/ulli/svn/firmware/src/RadioNetwork'
/usr/bin/ard-reset-arduino  net:beagle:4001
Traceback (most recent call last):
  File "/usr/bin/ard-reset-arduino", line 34, in <module>
    ser = serial.Serial(args.port[0], 115200)
  File "/usr/lib/python2.7/dist-packages/serial/serialutil.py", line 261, in __init__
    self.open()
  File "/usr/lib/python2.7/dist-packages/serial/serialposix.py", line 277, in open
    raise SerialException("could not open port %s: %s" % (self._port, msg))
serial.serialutil.SerialException: could not open port net:beagle:4001: [Errno 2] No such file or directory: 'net:beagle:4001'
/home/ulli/svn/firmware/Arduino-Makefile/Arduino.mk:1477: recipe for target 'reset' failed
make[1]: *** [reset] Error 1
make[1]: Leaving directory '/home/ulli/svn/firmware/src/RadioNetwork'
/home/ulli/svn/firmware/Arduino-Makefile/Arduino.mk:1453: recipe for target 'upload' failed
make: *** [upload] Error 2

sej7278, 2015-09-05:
hmm, yes its the reset routine expecting a serial device. have you tried setting the net:beagle:4001 thing as an isp? what is this anyway, does the ide support it or is there some sort of addon like with teensy?

Ulli2k, 2015-09-05:
Unfortunetly not, with ISP_PORT I get the following:
make[1]: Entering directory '/home/ulli/svn/firmware/src/RadioNetwork'
/home/ulli/svn/firmware/Arduino-Makefile/Arduino.mk:1477: *** Arduino port  not found!.  Schluss.
make[1]: Leaving directory '/home/ulli/svn/firmware/src/RadioNetwork'
/home/ulli/svn/firmware/Arduino-Makefile/Arduino.mk:1453: recipe for target 'upload' failed
make: *** [upload] Error 2

Ulli2k, 2015-09-05:
I modified the file "/usr/bin/ard-reset-arduino" and added in line 34 the following line
if "net:" not in args.port[0]:
Would it be possible to add it in the svn?

sej7278, 2015-09-05:
your line references make no sense to me (maybe 1502 not 1532?) perhaps you could make a PR for us to look at? we'd need to know what device it is you're using also to document it.

sej7278, 2015-12-18:
re-closing as no response

closed 2015-12-18 by sej7278