sudar/Arduino-Makefile #381: AVRDUDE_CONF not properly set when AVR_TOOLS_DIR specified

theicfire, 2015-09-08:
Looking at `Arduino.mk`, I see that if you're not on Windows and have `AVR_TOOLS_DIR` specified, AVRDUDE_CONF will not be set.

Repro:
- Set `AVR_TOOLS_DIR`
- Run `make upload` not on windows

The final upload command is something like
`/Applications/Arduino.app/Contents/Java/hardware/tools/avr/bin/avrdude -q -V -p atmega328p -D -c arduino -b 57600 -P /dev/cu.usbserial-AL00EZ5A -U flash:w:build-pro-16MHzatmega328/Blink.hex:i`

But if I remove `AVR_TOOLS_DIR`, I get something like:
`/Applications/Arduino.app/Contents/Java/hardware/tools/avr/bin/avrdude -q -V -p atmega328p -C /Applications/Arduino.app/Contents/Java/hardware/tools/avr/etc/avrdude.conf -D -c arduino -b 57600 -P /dev/cu.usbserial-AL00EZ5A -U flash:w:build-pro-16MHzatmega328/Blink.hex:i`

This second one actually works in that it has avrdude.conf passed over. The first one does not.

sej7278, 2015-09-09:
whilst i agree that setting ```AVR_TOOLS_DIR``` on non-windows does seem to cause ```AVRDUDE_CONF``` not to be found, it seems the fix is over-simplistic as it totally breaks linux:

```
/usr/share/arduino/hardware/tools/avr/bin/avrdude -q -V -p atmega2560  -C /usr/share/arduino/hardware/tools/avr/etc/avrdude.conf
...
avrdude: can't open config file "/usr/share/arduino/hardware/tools/avr/etc/avrdude.conf": No such file or directory
avrdude: error reading system wide configuration file "/usr/share/arduino/hardware/tools/avr/etc/avrdude.conf"
```

on linux it resides in /usr/share/arduino/hardware/tools/avr/../avrdude.conf which is a symlink to /etc/avrdude.conf

theicfire, 2015-09-10:
Ah, good catch. Looks like the if statement then should be "if windows or mac". I'm just not sure how to do that! It looks surprisingly complicated in Makefiles. Here's what I could find:

http://stackoverflow.com/questions/7656425/makefile-ifeq-logical-or

sej7278, 2015-09-10:
i think platform needs to be ignored, and a different check put in place, as we also support freebsd.

theicfire, 2015-09-10:
Hmm, I'm not sure what to do about this but I'll look at it later

sej7278, 2015-09-11:
also there's a version issue too it seems.....

1.0.6 : hardware/tools/avrdude.conf
1.6.5 : hardware/tools/**avr/etc**/avrdude.conf

theicfire, 2015-09-11:
Hmmm

`find $(AVR_TOOLS_DIR) -name avrdude.conf`  ¯\\_(ツ)_/¯

sej7278, 2015-09-11:
@theicfire its getting a bit like that yes with the random way the arduino devs lay things out, no concern for backwards compatibility or portability.

i really wish they'd split the cores/libraries/tools into a separate project to the ide.

ladislas, 2015-09-12:
> i really wish they'd split the cores/libraries/tools into a separate project to the ide.

@sej7278 could it not be possible to do this ourselves with a simple script pushing what we want to another repo?

sej7278, 2015-09-12:
@ladislas well yes, but then we end up choosing ide users or makefile users, i.e. you can't just use our makefile with the ide, you have to have a whole repo, and it'll also break compatibility with linux distro packages etc.

sej7278, 2015-09-12:
does this work on your mac (and what version ide are you using?) it works on linux, although it seems to be duplicating a lot of stuff from just above it, dunno if it works on cygwin

```diff
diff --git a/Arduino.mk b/Arduino.mk
index 43e1bb8..68c7e24 100644
--- a/Arduino.mk
+++ b/Arduino.mk
@@ -454,9 +454,11 @@ else
 
     # Check in Windows as Cygwin is being used, that the configuration file for the AVRDUDE is set
     # Check if it works on MAC
-    ifeq ($(CURRENT_OS),WINDOWS)
-        ifndef AVRDUDE_CONF
-            AVRDUDE_CONF  = $(AVR_TOOLS_DIR)/etc/avrdude.conf
+    ifndef AVRDUDE_CONF
+        ifeq ($(shell expr $(ARDUINO_VERSION) '>' 157), 1)
+            AVRDUDE_CONF = $(AVR_TOOLS_DIR)/etc/avrdude.conf
+        else
+            AVRDUDE_CONF = $(AVR_TOOLS_DIR)/../avrdude.conf
         endif
     endif
```

theicfire, 2015-09-13:
My version is 1.6.5-r5.

I don't have something to program with me but this looks like it should totally work

sej7278, 2015-09-13:
ok i've made a PR #385 we'll see about merging it when some people have tested it, it certainly works on linux 105, 106 and 165, but we need a mac test at least.

sej7278, 2015-09-14:
i've merged the PR after some more testing, thanks for the report

closed 2015-09-14 by sej7278