sudar/Arduino-Makefile #383: Teensy.mk no longer works with Arduino IDE 1.6.5 and Teensyduino 1.25 (and possibly earlier)

TJC, 2015-09-08:
Two problems.

## First problem

The variables in which build flags and the like are stored in boards.txt changed at some point. Teensy.mk no longer finds the variables it needs to build.

## Second problem

Teensy.mk looks for boards.txt without using the architecture type in the path, and thus does not locate it at all, on recent arduino ide installations.
ie. $ARDUINO/hardware/teensy/boards.txt vs $ARDUINO/hardware/teensy/avr/boards.txt

ladislas, 2015-09-08:
I've got the same issue. Best way I found was to use 1.0.x until settles and gets fixed.

sej7278, 2015-09-08:
lol, we should probably use that as a general disclaimer @ladislas "if you have problems with 1.6.5 try 1.0.5 !"

ladislas, 2015-09-08:
:D

sej7278, 2015-09-09:
i had a bit of a play with teensyduino 1.24 (1.25 is beta) and a lot of the problem seems to be that its an arm chip but the directory structure is avr, so its not just a case of adding ```$(ARCHITECTURE)``` or ```$(VENDOR)``` to the paths.

also there's a lot of 1.6 submenu stuff in boards.txt and platform.txt not just cpu like most chips, so i can't see us fixing this any time soon, unless the original Teensy.mk author @stepcut has any ideas....?

stepcut, 2015-09-20:
I'm still using 1.0.x myself. I still care about Teensy.mk, but I am unlikely to have time to investigate 1.6 any time soon.

maze735, 2016-07-27:
**I was setting up a Makefile based project for teensy 3.1. 
My projects makefile looks like that:**

ARDUINO_DIR  = /usr/share/arduino-1.6.9

# The name of your project (used to name the compiled .hex file)
TARGET = main

OPTIONS += -DTEENSYDUINO=129 -D__MK20DX256__
CPPFLAGS = -Wall -g -Os -mcpu=cortex-m4 -mthumb -MMD $(OPTIONS) -I.
CXXFLAGS =  -fno-exceptions -fno-rtti

# linker options
LDFLAGS = -Wl,--gc-sections,--defsym=__rtc_localtime=0\
    --specs=nano.specs\
    -mcpu=cortex-m4 -mthumb

BOARD_TAG = teensy31

# Avrdude code
ARDUINO_PORT = /dev/ttyACM0
AVRDUDE_ARD_PROGRAMMER = arduino
AVRDUDE_ARD_BAUDRATE = 115200

include /home/winkler/Desktop/helloArdu/Arduino-Makefile/Teensy.mk

**in Teensy.mk fix, otherwise boards.txt wont be found:**
-ARDUINO_CORE_PATH   = $(ARDUINO_DIR)/hardware/teensy/cores/teensy3
+ARDUINO_CORE_PATH   = $(ARDUINO_DIR)/hardware/teensy/**avr**/cores/teensy3

**in Boards.txt added:**
teensy31.build.architecture=arm
teensy31.build.gnu0x=true
teensy31.build.elide_constructors=true
teensy31.build.linkscript=mk20dx256.ld

maze735, 2016-07-28:
After some sleep i come to the conclusion that its better NOT to change anything in boards.txt because its part of the official toolchain. It should be better to only modify the  but map the options mentioned there manually to my Makefile. Chnages should only be made to the files of our "Arduino-Makefile" project.

That ends me up with my Makefile:
```

ARDUINO_DIR  = /usr/share/arduino-1.6.9
ARDUINO_CORE_PATH   = $(ARDUINO_DIR)/hardware/teensy/avr/cores/teensy3

BOARD_TAG = teensy31
ARCHITECTURE = arm
MCU=cortex-m4

# The name of your project (used to name the compiled .hex file)
TARGET = main

# Avrdude code
ARDUINO_PORT = /dev/ttyACM0
AVRDUDE_ARD_PROGRAMMER = arduino
AVRDUDE_ARD_BAUDRATE = 115200

OPTIONS += -DTEENSYDUINO=129 -D__MK20DX256__

# CPPFLAGS = compiler options for C and C++
CPPFLAGS = -Wall -g -ffunction-sections -fdata-sections -nostdlib \
    -Os \
    -mthumb -fsingle-precision-constant \
    -MMD $(OPTIONS) -I.

# compiler options for C++ only
CXXFLAGS = -fno-exceptions -felide-constructors -std=gnu++0x -fno-rtti

LDFLAGS = -Wl,--gc-sections,--relax,--defsym=__rtc_localtime=0\
    --specs=nano.specs\
    -mthumb \
    -T$(ARDUINO_CORE_PATH)/mk20dx256.ld

include /home/winkler/Desktop/helloArdu/Arduino-Makefile/Teensy.mk
```

sej7278, 2016-07-28:
could we use version checking perhaps as per Arduino.mk, as this diff seems to require 1.0 users to manually define everything or they'll get 1.6 settings

as far as a more sophisticated parser goes, i think we're going to need one at some point, probably written in python, although even then people are just making up random submenu's, which don't really map to Arduino.mk variables, purely to make the ide boards list shorter on screen!

maze735, 2016-07-29:
@sej7278: I have gone over everything again and fixed it so:
- it will leave old code intact
- adopts to new layout and extracts build flags from board.txt
- does a basic check on the paths
- only modifes Teensy.mk
- has minimal Makefile

TODO:
I works well for me under linux, but i am not sure if the advanced paramter extraction in PARSE_TEENSY would work flawless with OS X. To be tested!

My Makefile now looks clean again:
```
ARDUINO_DIR  = /usr/share/arduino-1.6.9

BOARD_TAG = teensy31
ARCHITECTURE = arm

# Avrdude code
ARDUINO_PORT = /dev/ttyACM0
AVRDUDE_ARD_PROGRAMMER = arduino
AVRDUDE_ARD_BAUDRATE = 115200

# The name of your project (used to name the compiled .hex file)
TARGET = main

include /home/winkler/Desktop/helloArdu/Arduino-Makefile/Teensy.mk
```

and here the final Teensy.mk:
```
diff --git a/Teensy.mk b/Teensy.mk
index 3676c78..4d3b430 100644
--- a/Teensy.mk
+++ b/Teensy.mk
@@ -37,18 +37,68 @@ ARDMK_VENDOR        = teensy
 ARDUINO_CORE_PATH   = $(ARDUINO_DIR)/hardware/teensy/cores/teensy3
 BOARDS_TXT          = $(ARDUINO_DIR)/hardware/$(ARDMK_VENDOR)/boards.txt
 
+#
+# teensiduino 1.29: try alternative paths if others are not found
+#
+ifeq ("$(wildcard $(ARDUINO_CORE_PATH))","")
+    ARDUINO_CORE_PATH   = $(ARDUINO_DIR)/hardware/teensy/avr/cores/teensy3
+endif
+
+ifeq ("$(wildcard $(BOARDS_TXT))","")
+    BOARDS_TXT   = $(ARDUINO_DIR)/hardware/teensy/avr/boards.txt
+endif
+#
+# End of teensiduino 1.29:
+#
+
 ifndef F_CPU
     F_CPU=96000000
 endif
 
 ifndef PARSE_TEENSY
     # result = $(call READ_BOARD_TXT, 'boardname', 'parameter')
-    PARSE_TEENSY = $(shell grep -v "^\#" "$(BOARDS_TXT)" | grep $(1).$(2) | cut -d = -f 2,3 )
+    # NOTE:
+    #  -grep searches for parameter-name starting with (^) the name. lines sartings
+    #  the comment sign # or other will be ignored. also it checks for = after the parameter name.
+    #  -sed will extract anything right of the first = char as epxression \2. the parameter name will
+    #   be stored in \1. try this on the shell: cat boards.txt | sed -e 's/\([^=]*\)=\(.*\)/\1 == \2/'
+    #  -two last lines fixes the tags: {extra.time.local} and {build.core.path}
+    #  -last sed uses | instead of / as speration charater otherwise the unsecaped path would mess up sed command
+    PARSE_TEENSY = $(shell \
+        grep ^$(1).$(2)= "$(BOARDS_TXT)" | \
+        sed -e 's/\([^=]*\)=\(.*\)/\2/'  | \
+        sed -e 's/{extra.time.local}/0/' | \
+        sed -e 's|{build.core.path}|$(ARDUINO_CORE_PATH)|' \
+        )
 endif
 
-ARCHITECTURE  = $(call PARSE_TEENSY,$(BOARD_TAG),build.architecture)
+
+ifndef ARCHITECTURE
+    ARCHITECTURE  = $(call PARSE_TEENSY,$(BOARD_TAG),build.architecture)
+endif
 AVR_TOOLS_DIR = $(call dir_if_exists,$(ARDUINO_DIR)/hardware/tools/$(ARCHITECTURE))
 
+
+########################################################################
+# check if some basic settings are set, otherwise stop
+
+ifeq ("$(wildcard $(ARDUINO_DIR))","")
+  $(error Arduino path not found: $(ARDUINO_DIR) )
+endif
+
+ifeq ("$(wildcard $(ARDUINO_CORE_PATH))","")
+  $(error Core path not found: $(ARDUINO_CORE_PATH) )
+endif
+
+ifeq ("$(wildcard $(BOARDS_TXT))","")
+  $(error boards.txt not found: $(BOARDS_TXT) )
+endif
+
+ifeq ("$(ARCHITECTURE)","")
+  $(error ARCHITECTURE is not set)
+endif
+
+
 ########################################################################
 # command names
 
@@ -129,6 +179,21 @@ ifndef MCU
     MCU := $(call PARSE_TEENSY,$(BOARD_TAG),build.cpu)
 endif
 
+#
+# teensiduino 1.29
+# MCU is part of the cpu options now: extract it as Arduino.mk (still) expects a seperate the mcu-switch
+ifndef MCU
+    CPU_FLAGS := $(call PARSE_TEENSY,$(BOARD_TAG),build.flags.cpu)
+    MCU = $(shell \
+    echo $(CPU_FLAGS) | \
+    sed -n -e 's/^.*mcpu=\([^ ]*\).*/\1/p' \
+    )
+endif
+
+#
+# End of teensiduino 1.29:
+#
+
 ifndef MCU_FLAG_NAME
     MCU_FLAG_NAME=mcpu
 endif
@@ -154,10 +219,32 @@ endif
 
 LDFLAGS +=  $(call PARSE_TEENSY,$(BOARD_TAG),build.linkoption) $(call PARSE_TEENSY,$(BOARD_TAG),build.additionalobject)
 
-ifneq ("$(call PARSE_TEENSY,$(BOARD_TAG),build.linkscript)",)
+# only set if if we really find the tag:
+ifneq ("$(call PARSE_TEENSY,$(BOARD_TAG),build.linkscript)","")
     LDFLAGS   += -T$(ARDUINO_CORE_PATH)/$(call PARSE_TEENSY,$(BOARD_TAG),build.linkscript)
 endif
 
+#
+# teensiduino 1.29
+#
+# usual flags
+CPPFLAGS += $(call PARSE_TEENSY,$(BOARD_TAG),build.flags.common)
+
+# defines
+CPPFLAGS += $(call PARSE_TEENSY,$(BOARD_TAG),build.flags.defs)
+CXXFLAGS += $(call PARSE_TEENSY,$(BOARD_TAG),build.flags.cpp)
+# plain linker flags
+LDFLAGS  += $(call PARSE_TEENSY,$(BOARD_TAG),build.flags.ld)
+
+# finally the cpu flags
+CPPFLAGS += $(call PARSE_TEENSY,$(BOARD_TAG),build.flags.cpu)
+LDFLAGS  += $(call PARSE_TEENSY,$(BOARD_TAG),build.flags.cpu)
+
+#
+# End of teensiduino 1.29:
+#
+
+
 ########################################################################
 # some fairly odd settings so that 'make upload' works
 #
```

sej7278, 2016-07-29:
its not working for me on 1.0.6 as it isn't defining ```AVR_TOOLS_DIR``` properly

```
ARCHITECTURE = arm
AVR_TOOLS_DIR = $(call dir_if_exists,$(ARDUINO_DIR)/hardware/tools/$(ARCHITECTURE))
```

would expand to arduino-1.0.6/hardware/tools/arm which simply doesn't exist, its arduino-1.0.6/hardware/tools/arm-none-eabi but setting arm-none-eabi as ```ARCHITECTURE``` doesn't help either as then you end up with arduino/teensy/arduino-1.0.6/hardware/teensy/arm-none-eabi/variants as ```ARDUINO_VAR_PATH``` which doesn't exist

if it helps this is the working 1.0.6 output without your patch:

```
Arduino.mk Configuration:
- [AUTODETECTED]       CURRENT_OS = LINUX 
- [USER]               ARDUINO_DIR = /home/user/arduino/teensy/arduino-1.0.6 
- [USER]               ARDMK_DIR = /usr/share/arduino 
- [AUTODETECTED]       ARDUINO_VERSION = 106 
- [USER]               ARCHITECTURE = arm-none-eabi 
- [USER]               ARDMK_VENDOR = teensy 
- [AUTODETECTED]       ARDUINO_PREFERENCES_PATH = /home/user/.arduino/preferences.txt 
- [AUTODETECTED]       ARDUINO_SKETCHBOOK = /home/user/arduino (from arduino preferences file)
- [USER]               AVR_TOOLS_DIR = /home/user/arduino/teensy/arduino-1.0.6/hardware/tools/arm-none-eabi 
- [COMPUTED]           ARDUINO_LIB_PATH = /home/user/arduino/teensy/arduino-1.0.6/libraries (from ARDUINO_DIR)
- [COMPUTED]           ARDUINO_VAR_PATH = /home/user/arduino/teensy/arduino-1.0.6/hardware/teensy/arm-none-eabi/variants (from ARDUINO_DIR)
- [USER]               BOARDS_TXT = /home/user/arduino/teensy/arduino-1.0.6/hardware/teensy/boards.txt 
- [DEFAULT]            USER_LIB_PATH = /home/user/arduino/libraries (in user sketchbook)
- [DEFAULT]            PRE_BUILD_HOOK = pre-build-hook.sh 
- [USER]               BOARD_TAG = teensy31 
- [COMPUTED]           CORE = teensy3 (from build.core)
- [COMPUTED]           VARIANT =  (from build.variant)
- [COMPUTED]           OBJDIR = build-teensy31 (from BOARD_TAG)
- [USER]               ARDUINO_CORE_PATH = /home/user/arduino/teensy/arduino-1.0.6/hardware/teensy/cores/teensy3 
- [ASSUMED]            MONITOR_BAUDRATE = 9600 
- [DEFAULT]            OPTIMIZATION_LEVEL = s 
- [USER]               MCU_FLAG_NAME = mcpu 
- [DEFAULT]            CFLAGS_STD =  
- [USER]               CXXFLAGS_STD = -std=gnu++0x 
- [AUTODETECTED]       DEVICE_PATH =  
- [DEFAULT]            FORCE_MONITOR_PORT =  
- [AUTODETECTED]       Size utility: Basic (not AVR-aware)
- [COMPUTED]           BOOTLOADER_PARENT = /home/user/arduino/teensy/arduino-1.0.6/hardware/teensy/arm-none-eabi/bootloaders (from ARDUINO_DIR)
- [COMPUTED]           ARDMK_VERSION = 1.5 
- [COMPUTED]           CC_VERSION = 4.7.2 (arm-none-eabi-gcc)
```

and just a simple Makefile:
```make
BOARD_TAG 	= teensy31
ARDUINO_DIR	= /home/user/arduino/teensy/arduino-1.0.6
include /usr/share/arduino/Teensy.mk
```

maze735, 2016-07-29:
Hi Simon,

the code old code seems wrong already as ARCHITECTURE is expected to result in an empty string even for your old stuff:
ARCHITECTURE  = $(call PARSE_TEENSY,$(BOARD_TAG),build.architecture)
AVR_TOOLS_DIR = $(call dir_if_exists,$(ARDUINO_DIR)/hardware/tools/$(ARCHITECTURE))

I will comeup with sth which fixes that too.
Where can I commit it?

cheers

sej7278, 2016-11-10:
Fixed with commit 2c1155e156ac0c

closed 2016-11-10 by sej7278