sudar/Arduino-Makefile #384: How can I exit from the monitor target?

daitangio, 2015-09-08:
Hi, under macosx when I issue "make monitor" I did not know what is the best way to exit from it.
If I detach screen, it will continue to run.
Also if I quit screen I need to reset my terminal.
What is the best way of doing it?

sej7278, 2015-09-08:
ctrl-a, then k, then "yes"

sej7278, 2015-09-09:
@daitangio does that work for you? 

i don't understand about quitting screen requiring you to reset your terminal. how are you quitting?

ladislas, 2015-09-09:
@sej7278 if you unplug the usb cable, you exit from screen but can't using again if you don't close and reopen the terminal

sej7278, 2015-09-09:
@ladislas well yes, because you've not disconnected from screen, you've just yanked the cable and killed the serial connection, its still running within screen, but that's to be expected.

daitangio, 2015-09-10:
Hi thank you!! If you detach screen you may have issue

closed 2015-09-10 by daitangio

sej7278, 2015-09-15:
oddly enough i just accidentally pulled the usb cable in the middle of a "make monitor" and when i reran "make monitor" it reconnected fine, but that's linux

ladislas, 2015-09-15:
[image]

sej7278, 2015-09-15:
;)

santids, 2017-02-10:
ctrl-a, then k, then "yes", works, though I think is not very elegant. Should be in the documentation maybe

sej7278, 2017-02-10:
it defaults to gnu screen which is the standard terminal emulator for most unixen, but as it can be overridden and replaced with minicom or whatever its difficult to document. PR's are always welcome