sudar/Arduino-Makefile #386: VENDOR variable conflicts with tcsh

stefanrueger, 2015-09-14:
Arduino.mk has stopped working under tcsh. It would roll out numerous grep errors of the kind

  grep: /usr/share/arduino-m/hardware/unknown/avr/boards.txt: No such file or directory

It took me a while to figure out that the problem is that VENDOR is a default environment name in the tcsh, which is wrongly taken to be a USER set value by Arduino.mk and makes it look in the wrong directories.

There are a number of things that can be done to improve this situation (also for bash users):
 - Test for the existence of the directory and make the error message more meaningful
 - Explain workarounds (set VENDOR explicitly in your Makefile, unsetenv VENDOR in tcsh, use bash, ...)
 - Rename the VENDOR variable in Makefile to something else, eg, ARDUINO_VENDOR, ARDMK_VENDOR, etc.

Obviously, I prefer the latter, as this avoids giving unwanted trouble to unsuspecting tcsh users, and descriptive variable names are always neat.

sej7278, 2015-09-14:
I've created PR #388 for you to test

stefanrueger, 2015-09-14:
That's cool. I can confirm this version works like a charm.

closed 2015-09-15 by sej7278