sudar/Arduino-Makefile #387: Option -I. dropped for 1.6.x arduino

stefanrueger, 2015-09-14:
When I switched from arduino 1.03 to 1.65, I noted that -I. was no longer  added to the compiler options by default. This is needed in my environment, because every of my libraries includes a "config.h" that is meant to be located in the project home directory. config.h holds hardware pin numbers, library and project configuration directives that influence how my libraries would compile for this particular project.  With -I. now missing, I get the fatal error "config.h: No such file or directory" when libraries are compiled for a project.

I found a few workarounds. One is to add the project directory relative to the USER_LIB_PATH, for example

   USER_LIB_PATH = /my/libs/are/somewhere/here
   ARDUINO_LIBS = ../../../../../where/my/project/is lib1 lib2 

Another is to add -I. to CPPFLAGS, eg,

CPPFLAGS += -I.

Yes, I can effect changes in all my Makefiles, but it's been a bit of a journey to figure out what needs doing, particularly as the EXTRA_FLAGS variable (that I had been using for years) has been withdrawn recently. 

In general, I would have thought that -I. is pretty useful and should remain to be included per default. I realise that (apparently) the arduino 1.5 or 1.6 IDE thinks the sketch source should be going to the src subdirectory. I also realise that Arduino.mk adds -Isrc (and recursions from there) when the src dir exists. 

Still, I argue it would be great if -I. was kept, if only for some sort of backward compatibility.

sej7278, 2015-09-14:
explanation for this is here:

https://github.com/sudar/Arduino-Makefile/commit/d998de8d84a29f1c7d5e1d9c541051ab29950b47

stefanrueger, 2015-09-14:
Thanks for the explanation.
  
OK, no -I. per default, but why remove the flexibility of appending user options that was possible with EXTRA_FLAGS? It's kind of neater to append -I. that have it up front as with the CPPFLAGS workaround.

sej7278, 2015-09-15:
```EXTRA_FLAGS``` removal explained in issue https://github.com/sudar/Arduino-Makefile/pull/234, essentially we couldn't figure out what it was for and decided it was fine to append to ```CXX_FLAGS```

stefanrueger, 2015-09-15:
Yes, I can confirm

CPPFLAGS += I.
CXXFLAGS += ...

in my own Makefile now allow me to include -I. and other options (such as my favourite -mcall-prologues -fno-tree-switch-conversion) at the beginning or end of the options depending whether I put these lines before or after
 
include $(ARDMK_DIR)/Arduino.mk

That's good and closes the issue for me. Thanks for your hard work and support.

closed 2015-09-15 by sej7278