sudar/Arduino-Makefile #389: Adafruit Bluefruit LE Micro uploading fails

jonblack, 2015-09-19:
The [Adafruit Bluefruit LE Micro](https://www.adafruit.com/products/2661) is an Arduino compatible board which contains a Bluetooth LE chip and an ATmega32u4.

The documentation says the following:

>You can program the ATmega32u4 over USB using the built-in USB bootloader, either directly with avrdude or using the Arduino IDE (it's the same as used as in the Adafruit Flora or Arduino Leonardo).

When selecting `Leonardo` in the Arduino IDE, compilation and uploading of the `blink` example works fine. Using a Makefile fails to upload with the following error:

    mkdir -p build-leonardo
    make reset
    make[1]: Entering directory '/home/jon/arduino/example'
    /usr/bin/ard-reset-arduino --caterina  /dev/ttyACM0
    Traceback (most recent call last):
      File "/usr/bin/ard-reset-arduino", line 23, in <module>
        ser.setDTR(False) # (see Arduino IDE source code)
      File "/usr/lib/python2.7/dist-packages/serial/serialposix.py", line 564, in setDTR
        fcntl.ioctl(self.fd, TIOCMBIC, TIOCM_DTR_str)
    IOError: [Errno 32] Broken pipe
    /usr/share/arduino/Arduino.mk:1432: recipe for target 'reset' failed
    make[1]: *** [reset] Error 1
    make[1]: Leaving directory '/home/jon/arduino/example'
    /usr/share/arduino/Arduino.mk:1408: recipe for target 'upload' failed
    make: *** [upload] Error 2

The Makefile contains:

    BOARD_TAG = leonardo
    ARDUINO_SKETCHBOOK = /home/jon/arduino
    include /usr/share/arduino/Arduino.mk

I'm using version 1.6.5 of the Arduino IDE and am on Debian.

sej7278, 2015-09-19:
i saw this recently too, i think it was a busted 32u4 bootloader on my pro micro, that was using 1.0.5 on debian

jonblack, 2015-09-27:
I actually uploaded the blink sketch successfully. All subsequence uploads produce the error. Do you still think it could be a wiring issue? In any case, I'm getting the same problems in the Arduino IDE so it's not a problem with this project.

ladislas, 2015-09-27:
have you tried manually resetting the board just before uploading?

sej7278, 2015-09-27:
i assume its not the issue on their product page?

https://learn.adafruit.com/bluefruit-le-micro-atmega32u4-microcontroller-usb-bluetooth-le-in-one/using-with-arduino-ide#ubuntu-and-linux-issue-fix

sej7278, 2015-09-28:
seems they have a few problems with that board, as @ladislas says, pressing reset just as you upload may allow you to upload, apparently it only needs to work once to get the bootloader to kick in

https://forums.adafruit.com/viewtopic.php?f=22&t=81094

jonblack, 2015-09-28:
@ladislas 
I've tried resetting the board before uploading and still not joy.

@sej7278
'm not having bluetooth issues but I applied the udev rules to check but again, no joy. Nice find on the forum, it's good to see someone else having the same problem, although at the same time, that doesn't bode well. I'm also [in touch with adafruit support in the forum](https://forums.adafruit.com/viewtopic.php?f=53&t=81109&p=410559#p410559) and they've already made me aware of something that wasn't in the tutorial linked from the product page: there is a separate board profile for the Arduino IDE. This doesn't work because the board isn't being detected, but on a slightly off topic note, I was wondering if these boards can be added to Arduino-Makefile?

In the meantime, I'm gonna go crazy with the reset/upload hack and see if I'm lucky.

jonblack, 2015-09-28:
Ok it seems I am lucky. The hack worked, but only from within the Arduino IDE with the Adafruit Bluefruit Micro board selected. Thanks for the tips.

sej7278, 2015-09-28:
you should be able to just copy this [boards.txt](
https://github.com/adafruit/Adafruit_Arduino_Boards/blob/master/hardware/adafruit/avr/boards.txt) to your hardware directory and point your Makefile at it:

```
BOARD_TAG = bluefruitmicro
BOARDS_TXT = /home/jon/arduino/hardware/adafruit/boards.txt 
include /usr/share/arduino/Arduino.mk
```

jonblack, 2015-09-29:
I got the following error when I tried the above:

    mkdir -p build-bluefruitmicro
    /usr/share/arduino/hardware/tools/avr/bin/avr-g++ -x c++ -include Arduino.h -MMD -c -mmcu=atmega32u4 -DF_CPU=8000000L -DARDUINO=105  -D__PROG_TYPES_COMPAT__ -I/usr/share/arduino/hardware/arduino//cores/arduino -I/usr/share/arduino/hardware/arduino//variants/bluefruitmicro    -Wall -ffunction-sections -fdata-sections -Os -DUSB_VID=0x239A -DUSB_PID=0x800A -fno-exceptions  blink.ino -o build-bluefruitmicro/blink.o
    In file included from <command-line>:0:0:
    /usr/share/arduino/hardware/arduino//cores/arduino/Arduino.h:213:26: fatal error: pins_arduino.h: No such file or directory
     #include "pins_arduino.h"

The `pins_arduino.h` file is also on the adafruit site, but it has to be in `/usr/share/arduino/hardware/arduino//variants/bluefruitmicro` since I've installed Arduino-Makefile via the package. I wonder if there's a nicer way? If I upgrade the package, or update my system, I'll need to do this again.

sej7278, 2015-09-29:
yeah, i'll have a play probably tomorrow now, you should be able to download [this lot](https://github.com/adafruit/Adafruit_Arduino_Boards/tree/master/hardware/adafruit/avr) to your ~/arduino/hardware/adafruit/avr/ directory and just set ```ALTERNATE_CORE``` like we do for attiny or promicro

i've also got to check the 1.6.5-r5 versioning scheme doesn't break things, shouldn't do.

sej7278, 2015-09-29:
actually didn't take that long, i can't test it uploads, but it sure builds fine (and the versioning is ok):

Makefile:

```make
ARDUINO_DIR = $(HOME)/arduino-1.6.5-r5

#MONITOR_PORT = /dev/ttyACM0
ALTERNATE_CORE = adafruit
BOARD_TAG = bluefruitmicro

include /usr/share/arduino/Arduino.mk
```

Directory structure inside my sketchbook:
```
hardware/
├── adafruit
│   └── avr
│       ├── boards.txt
│       ├── bootloaders
│       ├── platform.txt
│       └── variants
```

sej7278, 2015-10-04:
Can we close this now? Did you try my setup above, did it flash ok?

jonblack, 2015-10-04:
Yup, you can close this. It works great, thanks.

closed 2015-10-04 by jonblack