sudar/Arduino-Makefile #392: Link errors in Ethernet library using mega2560 - no errors in Arduino IDE

hwstar, 2015-10-23:
I just recently upgraded to Arduino 1.6.5, and I'm trying to get a sucessful compile and link with the Makefile. Previously, I was using Arduino 1.5, and the Makefile worked perfectly.

I can compile my application using the Arduino IDE, but not with the Makefile. I'm getting some link errors. These are the errors I'm seeing:

/home/srodgers/bin/arduino-1.6.5-r5/hardware/tools/avr/bin/avr-gcc -mmcu=atmega2560 -Wl,--gc-sections -Os -o build-mega-atmega2560/gatewaymq24.elf build-mega-atmega2560/jsmn.c.o build-mega-atmega2560/gatewaymq24.ino.o build-mega-atmega2560/libcore.a  -lc -lm 
build-mega-atmega2560/libcore.a(Ethernet.cpp.o): In function `EthernetClass::begin(unsigned char*)':
Ethernet.cpp:(.text._ZN13EthernetClass5beginEPh+0x2c): undefined reference to `__cxa_guard_acquire'
Ethernet.cpp:(.text._ZN13EthernetClass5beginEPh+0x40): undefined reference to `__cxa_guard_release'
collect2: error: ld returned 1 exit status
make: *** [build-mega-atmega2560/gatewaymq24.elf] Error 1

System configuration:

@srodgers-Precision-WorkStation-T3500 ~/projects/arduino/sketchbook/gatewaymq24 $ 
srodgers@srodgers-Precision-WorkStation-T3500 ~/projects/arduino/sketchbook/gatewaymq24 $ make clean
-------------------------
Arduino.mk Configuration:
- [AUTODETECTED]       CURRENT_OS = LINUX 
- [USER]               ARDUINO_DIR = /home/srodgers/bin/arduino-1.6.5-r5 
- [COMPUTED]           ARDMK_DIR = /home/srodgers/projects/arduino/Arduino-Makefile (relative to Common.mk)
- [AUTODETECTED]       ARDUINO_VERSION = 165 
- [USER]               ARCHITECTURE = avr 
- [DEFAULT]            ARDMK_VENDOR = arduino 
- [AUTODETECTED]       ARDUINO_PREFERENCES_PATH = /home/srodgers/.arduino15/preferences.txt 
- [AUTODETECTED]       ARDUINO_SKETCHBOOK = /home/srodgers/Arduino (from arduino preferences file)
- [BUNDLED]            AVR_TOOLS_DIR = /home/srodgers/bin/arduino-1.6.5-r5/hardware/tools/avr (in Arduino distribution)
- [COMPUTED]           ARDUINO_LIB_PATH = /home/srodgers/bin/arduino-1.6.5-r5/libraries (from ARDUINO_DIR)
- [COMPUTED]           ARDUINO_PLATFORM_LIB_PATH = /home/srodgers/bin/arduino-1.6.5-r5/hardware/arduino/avr/libraries (from ARDUINO_DIR)
- [COMPUTED]           ARDUINO_VAR_PATH = /home/srodgers/bin/arduino-1.6.5-r5/hardware/arduino/avr/variants (from ARDUINO_DIR)
- [COMPUTED]           BOARDS_TXT = /home/srodgers/bin/arduino-1.6.5-r5/hardware/arduino/avr/boards.txt (from ARDUINO_DIR)
- [DEFAULT]            USER_LIB_PATH = /home/srodgers/Arduino/libraries (in user sketchbook)
- [DEFAULT]            PRE_BUILD_HOOK = pre-build-hook.sh 
- [USER]               BOARD_SUB = atmega2560 
- [USER]               BOARD_TAG = mega 
- [COMPUTED]           CORE = arduino (from build.core)
- [COMPUTED]           VARIANT = mega (from build.variant)
- [COMPUTED]           OBJDIR = build-mega-atmega2560 (from BOARD_TAG)
- [COMPUTED]           ARDUINO_CORE_PATH = /home/srodgers/bin/arduino-1.6.5-r5/hardware/arduino/avr/cores/arduino (from ARDUINO_DIR, BOARD_TAG and boards.txt)
- [DETECTED]           MONITOR_BAUDRATE = 57600  (in sketch)
- [DEFAULT]            OPTIMIZATION_LEVEL = s 
- [DEFAULT]            MCU_FLAG_NAME = mmcu 
- [DEFAULT]            CFLAGS_STD =  
- [DEFAULT]            CXXFLAGS_STD =  
- [COMPUTED]           DEVICE_PATH = /dev/ttyACM0 (from MONITOR_PORT)
- [DEFAULT]            FORCE_MONITOR_PORT =  
- [AUTODETECTED]       Size utility: AVR-aware for enhanced output
-
-                      ARDUINO_LIBS =
- [USER]                 PubSubClient
- [USER]                 RadioHead
- [SYSTEM]               Ethernet
- [PLATFORM]             SPI
- [COMPUTED]           BOOTLOADER_PARENT = /home/srodgers/bin/arduino-1.6.5-r5/hardware/arduino/avr/bootloaders (from ARDUINO_DIR)
- [COMPUTED]           ARDMK_VERSION = 1.5 
- [COMPUTED]           CC_VERSION = 4.8.1 (avr-gcc)
-------------------------

mikeS7, 2015-11-10:
fix from esp8266/Arduino https://github.com/esp8266/Arduino/issues/574#issuecomment-123416888 in Ethernet.cpp

sej7278, 2015-11-10:
oh interesting, thanks for the link to the workaround. not a lot we can do unless arduino merge the fix to the library, obviously their mangler is working around it.

closed 2015-11-14 by hwstar