sudar/Arduino-Makefile #394: Sublime Text 2 Fails to Compile and Run

domi-a-philip, 2015-11-14:
Hello,

I'm using Sublime Text 2 to write code for my Arduino Uno. I'm using a custom build system for Sublime Text 2 to compile and run the Arduino program. When I just build the program through Sublime, it works but when I build and run It fails with an ImportError: No module named serial, even though I have the serial module for python. But when I use the terminal, everything works just fine. I am on Mac OS X 10.10.4, Python 2.7.10, Arduino IDE version 1.6.5, Makefile Version 1.5

This is my build and run result while using Sublime Text 2
```-------------------------
Arduino.mk Configuration:
- [AUTODETECTED]       CURRENT_OS = MAC 
- [AUTODETECTED]       ARDUINO_DIR = /Applications/Arduino.app/Contents/Java 
- [COMPUTED]           ARDMK_DIR = /usr/local/Cellar/arduino-mk/1.5 (relative to Common.mk)
- [AUTODETECTED]       ARDUINO_VERSION = 165 
- [DEFAULT]            ARCHITECTURE = avr 
- [DEFAULT]            VENDOR = arduino 
- [AUTODETECTED]       ARDUINO_PREFERENCES_PATH = /Users/dominicphilip/Library/Arduino15/preferences.txt 
- [AUTODETECTED]       ARDUINO_SKETCHBOOK = /Users/dominicphilip/Documents/Arduino (from arduino preferences file)
- [BUNDLED]            AVR_TOOLS_DIR = /Applications/Arduino.app/Contents/Java/hardware/tools/avr (in Arduino distribution)
- [COMPUTED]           ARDUINO_LIB_PATH = /Applications/Arduino.app/Contents/Java/libraries (from ARDUINO_DIR)
- [COMPUTED]           ARDUINO_PLATFORM_LIB_PATH = /Applications/Arduino.app/Contents/Java/hardware/arduino/avr/libraries (from ARDUINO_DIR)
- [COMPUTED]           ARDUINO_VAR_PATH = /Applications/Arduino.app/Contents/Java/hardware/arduino/avr/variants (from ARDUINO_DIR)
- [COMPUTED]           BOARDS_TXT = /Applications/Arduino.app/Contents/Java/hardware/arduino/avr/boards.txt (from ARDUINO_DIR)
- [DEFAULT]            USER_LIB_PATH = /Users/dominicphilip/Documents/Arduino/libraries (in user sketchbook)
- [DEFAULT]            PRE_BUILD_HOOK = pre-build-hook.sh 
- [USER]               BOARD_TAG = uno 
- [COMPUTED]           CORE = arduino (from build.core)
- [COMPUTED]           VARIANT = standard (from build.variant)
- [COMPUTED]           OBJDIR = build-uno (from BOARD_TAG)
- [COMPUTED]           ARDUINO_CORE_PATH = /Applications/Arduino.app/Contents/Java/hardware/arduino/avr/cores/arduino (from ARDUINO_DIR, BOARD_TAG and boards.txt)
- [DETECTED]           MONITOR_BAUDRATE = 9600  (in sketch)
- [DEFAULT]            OPTIMIZATION_LEVEL = s 
- [DEFAULT]            MCU_FLAG_NAME = mmcu 
- [DEFAULT]            CFLAGS_STD =  
- [DEFAULT]            CXXFLAGS_STD =  
- [AUTODETECTED]       DEVICE_PATH = /dev/tty.usbmodem1a11 
- [DEFAULT]            FORCE_MONITOR_PORT =  
- [AUTODETECTED]       Size utility: AVR-aware for enhanced output
- [COMPUTED]           BOOTLOADER_PARENT = /Applications/Arduino.app/Contents/Java/hardware/arduino/avr/bootloaders (from ARDUINO_DIR)
- [COMPUTED]           ARDMK_VERSION = 1.5 
- [COMPUTED]           CC_VERSION = 4.8.1 (avr-gcc)
-------------------------
mkdir -p build-uno
/Applications/Xcode.app/Contents/Developer/usr/bin/make reset
/usr/local/Cellar/arduino-mk/1.5/bin/ard-reset-arduino  /dev/tty.usbmodem1a11
Traceback (most recent call last):
  File "/usr/local/Cellar/arduino-mk/1.5/bin/ard-reset-arduino", line 4, in <module>
    import serial
ImportError: No module named serial
make[1]: *** [reset] Error 1
make: *** [upload] Error 2
```

This is my Makefile
```make
BOARD_TAG    = uno

include /usr/local/opt/arduino-mk/Arduino.mk

monitor:
	@echo " ---- open serial ---- "
	osascript -e 'tell application "Terminal" to do script "screen /dev/tty.usbmodem* 9600"'

killmonitor:
	@echo " ---- close serial ---- "
	osascript -e 'tell application "Terminal" to do script "screen -X quit"'
```

This is my Sublime Build File
```JSON
{
	"cmd": ["make"],
	"shell": true,
	"selector": "source.arduino",

	"variants": [

		{
			"name": "Run",
			"cmd": ["make upload"],
			"shell": true,
			"quiet": true
		}

	]
}
```

sej7278, 2015-11-14:
probably due to multiple versions of python installed and the one called from ```/usr/bin/env python``` doesn't have the serial module installed, you can test that by running:

```
/usr/bin/env python -c 'import serial'
``` 

it should return nothing if the module is installed in the system default python. if you get an importerror then you've not installed the serial module to the correct place.

so you can fix that by either installing the serial module to the correct location of python, or change the shebang in ard-reset-arduino to point to wherever you've installed the version of python with the serial module, instead of using env.

the whole point of env is to point everything to the correct version of python, so whatever you used to install the serial module (xcode, homebrew, pip, pypy?) has ignored that which is very bad form.

domi-a-philip, 2015-11-15:
I fixed the problem by changing the shebang line to where it was installed. Even though when I ran the command it returned nothing, only after I changed the shebang line, it was fixed.

Thanks for the help!

closed 2015-11-15 by sudar