sudar/Arduino-Makefile #395: Make Into Build Folder

domi-a-philip, 2015-11-26:
Hello,

I was wondering if it was possible to keep the Makefile and .ino files in the src folder and when make or make upload is executed, it builds it into the build folder instead of the src folder. Is there a flag to do this? I believe it is somewhat similar to #267 

This is my Project Structure
```
├── build
├── docs
├── ext
│   ├── include
│   ├── libs
└── src
    └── blink.ino
    └── Makefile
```

sej7278, 2015-11-27:
have a look at https://github.com/sudar/Arduino-Makefile/blob/master/arduino-mk-vars.md#objdir

this works for me:

```
BOARD_TAG = uno
MONITOR_PORT = /dev/ttyUSB0
OBJDIR = ../build
USER_LIB_PATH = ../ext/libs
include /usr/share/arduino/Arduino.mk
```

with this directory structure:

```
├── build/
│   ├── core/
│   │   ├── avr-libc/
│   │   ├── CDC.cpp.d
│   │   ├── blah...
│   │   └── WString.cpp.o
│   ├── libcore.a
│   ├── libs/
│   │   └── Servo/
│   ├── obstacle_bot.ino.d
│   ├── obstacle_bot.ino.o
│   ├── src.eep
│   ├── src.elf
│   ├── src.hex
│   ├── src.hex.sizeok
│   └── userlibs/
│       └── DCMotor/
├── docs/
├── ext/
│   └── libs/
│       └── DCMotor/
└── src/
    ├── Makefile
    └── obstacle_bot.ino
```

domi-a-philip, 2015-11-27:
Works perfectly, thank you!

closed 2015-11-27 by domi-a-philip