sudar/Arduino-Makefile #396: Over shh on a Raspberry Pi , error when uploading

TheeeeeeeeeeeeeeeBM, 2015-12-01:
it gives me this error 

make: /usr/share/arduino/ard-parse-boards: Command not found
make: /usr/share/arduino/ard-parse-boards: Command not found
/usr/bin/avr-gcc -c -mmcu=atmega328 -DF_CPU= -DARDUINO=100 -I. -I/usr/share/arduino/hardware/arduino/cores/arduino -I/usr/share/arduino/hardware/arduino/variants/   -g -Os -w -Wall -ffunction-sections -fdata-sections -std=gnu99 /usr/share/arduino/hardware/arduino/cores/arduino/wiring.c -o build-cli/wiring.o
In file included from /usr/share/arduino/hardware/arduino/cores/arduino/wiring_private.h:33:0,
                 from /usr/share/arduino/hardware/arduino/cores/arduino/wiring.c:25:
/usr/share/arduino/hardware/arduino/cores/arduino/Arduino.h:213:26: fatal error: pins_arduino.h: No such file or directory
compilation terminated.
/usr/share/arduino/Arduino.mk:432: recipe for target 'build-cli/wiring.o' failed
make: *** [build-cli/wiring.o] Error 1

I installed everything like in the guide but it doen't work ..

sej7278, 2015-12-01:
we don't use ard-parse-boards anymore, which guide are you following, as it looks like you're using arduino 1.0.0 and arduino-mk 0.8.5, which pre-date this repo.

if you use raspbian jessie you should get arduino-mk 1.5 and arduino-core 1.0.5 from apt-get

http://archive.raspbian.org/raspbian/pool/main/a/arduino-mk/
http://archive.raspbian.org/raspbian/pool/main/a/arduino/

sej7278, 2015-12-18:
closing as no response and looks like this pre-dates this repo

closed 2015-12-18 by sej7278