sudar/Arduino-Makefile #397: [integration / linking] Forward a true IDE-free environment ?

jcahier, 2015-12-03:
Hello,
Arduino-mk is a package i have known and loved (after first step using IDE, as soon as i had to manage multi-library projects, Makefile became a real urge, what this project fullfit).

I came back to arduino for a drone project and was pleased to find out it was packaged in Arch Linux AUR, with for major dependencies standard packets of Arch. 
Except the arduino package, which is also in AUR, and some of its dependencies... Which also pulls java RE... 

My concern is : could it be possible to remove this IDE / java / other exotic packages ... dependencies, but keeping Arduino "syntax" compatibily (allow linking to Arduino API to fully support any sketche) ?
Basically, i guess this is just about installing Arduino API headers (to compile) and binaries (to link).

I know this is not your job as a source repo maintener(s), i just ASK your opinion on this topic, please.

Kind regards, and many thanks for this handsome tool !

sej7278, 2015-12-03:
in debian there is a package called arduino-core https://packages.debian.org/sid/arduino-core

which is the boards, libraries etc; but not the java ide. if you install that and arduino-mk then you're good to go, as it'll install avr-gcc, avrdude, pyserial etc. as dependencies.

looking at the arch packages the equivalent is https://aur.archlinux.org/packages/arduino-noide-avr/

jcahier, 2015-12-03:
hooo ! Merry Chrismas to you too !!
Many thanks and kind regards,
Johann

sej7278, 2015-12-04:
glad to help, Merry Christmas! :smile:

closed 2015-12-04 by sej7278