sudar/Arduino-Makefile #398: Still missing argument to ‘-mmcu=’

T3KNOCRAT, 2015-12-08:
I've read through the previous tickets on this ([#45](https://github.com/sudar/Arduino-Makefile/issues/45), [#204](https://github.com/sudar/Arduino-Makefile/issues/204), [#327](https://github.com/sudar/Arduino-Makefile/issues/327)), and I'm a little unclear as to which version(s) of this Makefile works with which versions of the Arduino IDE.

For example, I'm trying to compile and upload Blink to an Arduino Pro Mini (ATMega328, 3.3V, 8MHz), when I get thwarted with the error:
    
    avr-g++: error: missing argument to ‘-mmcu=’

I'm also having the exact problem when I try to compile/upload to my Adafruit Trinket.

I was able to solve this by downloading an [older release](https://www.arduino.cc/en/Main/OldSoftwareReleases#previous) of the Arduino IDE, specifically version 1.0.5 (after trying 1.6.5, 1.6.4, 1.5.8, and probably a few others) onto my Debian/x64 workstation.

To ease my life during these experiments, I created a symlink to the version I wanted to test, a la:

```
┌─[/usr/local/bin]──────────────────────────────────────[[pts/6]][20:59:00]─
└[]─ $ rm arduino && ln -s arduino-1.0.5 arduino
```

As expected, when I run `make show_boards` (using arduino-1.05), the results yield:

```
┌─[/z/sketches/Blink]──────────────────────────────────  [[pts/5]][21:02:44]─
└[]─ $ make show_boards
 . . .
-------------------------
atmega168     Arduino NG or older w/ ATmega168
atmega328     Arduino Duemilanove w/ ATmega328
atmega8       Arduino NG or older w/ ATmega8
bt328         Arduino BT w/ ATmega328
bt            Arduino BT w/ ATmega168
diecimila     Arduino Diecimila or Duemilanove w/ ATmega168
esplora       Arduino Esplora
ethernet      Arduino Ethernet
fio           Arduino Fio
leonardo      Arduino Leonardo
lilypad328    LilyPad Arduino w/ ATmega328
lilypad       LilyPad Arduino w/ ATmega168
LilyPadUSB    LilyPad Arduino USB
mega2560      Arduino Mega 2560 or Mega ADK
mega          Arduino Mega (ATmega1280)
micro         Arduino Micro
mini328       Arduino Mini w/ ATmega328
mini          Arduino Mini w/ ATmega168
nano328       Arduino Nano w/ ATmega328
nano          Arduino Nano w/ ATmega168
pro328        Arduino Pro or Pro Mini (3.3V, 8 MHz) w/ ATmega328
pro5v328      Arduino Pro or Pro Mini (5V, 16 MHz) w/ ATmega328
pro5v         Arduino Pro or Pro Mini (5V, 16 MHz) w/ ATmega168
pro           Arduino Pro or Pro Mini (3.3V, 8 MHz) w/ ATmega168
robotControl  Arduino Robot Control
robotMotor    Arduino Robot Motor
uno           Arduino Uno
```

But a much more condensed list appears when I have the symlink pointing to arduino-1.6.6:

```
┌─[/z/sketches/Blink]───────────────────────────────────[[pts/5]][21:02:44]─
└[]─ $ rm /usr/local/bin/arduino && ln -s /usr/local/bin/arduino-1.6.6 /usr/local/bin/arduino
┌─[/z/sketches/Blink]───────────────────────────────────[[pts/5]][21:02:48]─
└[]─ $ make show_boards
. . .
-------------------------
atmegang      Arduino NG or older
bt            Arduino BT
diecimila     Arduino Duemilanove or Diecimila
esplora       Arduino Esplora
ethernet      Arduino Ethernet
fio           Arduino Fio
gemma         Arduino Gemma
leonardo      Arduino Leonardo
lilypad       LilyPad Arduino
LilyPadUSB    LilyPad Arduino USB
megaADK       Arduino Mega ADK
mega          Arduino/Genuino Mega or Mega 2560
micro         Arduino/Genuino Micro
mini          Arduino Mini
nano          Arduino Nano
pro           Arduino Pro or Pro Mini
robotControl  Arduino Robot Control
robotMotor    Arduino Robot Motor
uno           Arduino/Genuino Uno
yun           Arduino Yún
```

I saw someone advise that `BOARD_SUB` should be set, and maybe `ALTERNATE_CORE` should be set, but neither would work when I have the `arduino symlink` pointing to any IDE that's more current than v1.0.5. The difficulty I'm having here is *where* do I find that info?   I mean, I saw that commit [#393](https://github.com/sudar/Arduino-Makefile/commit/edcaa8557b1117dcee96b3a7c4d6fa736d80ac81) added "Better parsing of boards.txt file which allows Adafruit Trinket to be used", but for some reason, the Makefile does *not* find that board. I would presume that the HEAD would have an updated `show_boards` that lists sub-boards, and the Adafruit boards (I am able to see those via the IDE).

In short, I think that I must be missing something in the documentation that should explain this. Rather than this being a bug in the Makefile, it might be a bug in the Documentation - as in there isn't an explanation about how this works, and/or how to troubleshoot this error.  I mean, other than reading through some Issues that are many versions old.

Could someone try to explain to me why `make` isn't "finding" the board I've specified in the Makefile?  Or maybe a link to a specific document that can help me through this?

For what it's worth, my set up is as follows:

Makefile

```
ARDUINO_DIR   = /usr/local/bin/arduino
BOARD_TAG     = pro328
#BOARD_TAG     = pro
#BOARD_TAG     = trinket
#BOARD_SUB     = pro328
MONITOR_PORT  = /dev/ttyUSB0
#ARCHITECTURE  = avr
include /usr/local/arduino/Arduino-Makefile/Arduino.mk
ARDUINO_SKETCHBOOK = /z/sketches
```

The error message:

```

┌─[/z/sketches/Blink]───────────────────────────────────[[pts/5]][21:18:50]─
└[]─ $ make
-------------------------
Arduino.mk Configuration:
- [AUTODETECTED]       CURRENT_OS = LINUX 
- [USER]               ARDUINO_DIR = /usr/local/bin/arduino 
- [USER]               ARDMK_DIR = /usr/local/arduino/Arduino-Makefile 
- [AUTODETECTED]       ARDUINO_VERSION = 166 
- [DEFAULT]            ARCHITECTURE = avr 
- [DEFAULT]            ARDMK_VENDOR = arduino 
- [AUTODETECTED]       ARDUINO_PREFERENCES_PATH = /home/reaper/.arduino15/preferences.txt
- [AUTODETECTED]       ARDUINO_SKETCHBOOK = /z/sketches (from arduino preferences file)
- [USER]               AVR_TOOLS_DIR = /usr
- [COMPUTED]           ARDUINO_LIB_PATH = /usr/local/bin/arduino/libraries (from ARDUINO_DIR)
- [COMPUTED]           ARDUINO_PLATFORM_LIB_PATH = /usr/local/bin/arduino/hardware/arduino/avr/libraries (from ARDUINO_DIR)
- [COMPUTED]           ARDUINO_VAR_PATH = /usr/local/bin/arduino/hardware/arduino/avr/variants (from ARDUINO_DIR)
- [COMPUTED]           BOARDS_TXT = /usr/local/bin/arduino/hardware/arduino/avr/boards.txt (from ARDUINO_DIR)
- [DEFAULT]            USER_LIB_PATH = /z/sketches/libraries (in user sketchbook)
- [DEFAULT]            PRE_BUILD_HOOK = pre-build-hook.sh
- [USER]               BOARD_TAG = pro328
- [COMPUTED]           CORE =  (from build.core)
- [COMPUTED]           VARIANT =  (from build.variant)
- [COMPUTED]           OBJDIR = build-pro328 (from BOARD_TAG)
- [DEFAULT]            ARDUINO_CORE_PATH = /usr/local/bin/arduino/hardware/arduino/avr/cores/arduino
- [ASSUMED]            MONITOR_BAUDRATE = 9600
- [DEFAULT]            OPTIMIZATION_LEVEL = s
- [DEFAULT]            MCU_FLAG_NAME = mmcu
- [DEFAULT]            CFLAGS_STD =
- [DEFAULT]            CXXFLAGS_STD =
- [COMPUTED]           DEVICE_PATH = /dev/ttyUSB0 (from MONITOR_PORT)
- [DEFAULT]            FORCE_MONITOR_PORT =
- [AUTODETECTED]       Size utility: AVR-aware for enhanced output
- [COMPUTED]           BOOTLOADER_PARENT = /usr/local/bin/arduino/hardware/arduino/avr/bootloaders (from ARDUINO_DIR)
- [COMPUTED]           ARDMK_VERSION = 1.5
- [COMPUTED]           CC_VERSION = 4.8.1 (avr-gcc)
-------------------------
mkdir -p build-pro328
/usr/bin/avr-g++ -x c++ -include Arduino.h -MMD -c -mmcu= -DF_CPU= -DARDUINO=166 -DARDUINO_ARCH_AVR -D__PROG_TYPES_COMPAT__ -I/usr/local/bin/arduino/hardware/arduino/avr/cores/arduino -I/usr/local/bin/arduino/hardware/arduino/avr/variants/    -Wall -ffunction-sections -fdata-sections -Os -fno-exceptions  Blink.ino -o build-pro328/Blink.ino.o
avr-g++: error: missing argument to ‘-mmcu=’
/usr/local/arduino/Arduino-Makefile/Arduino.mk:1225: recipe for target 'build-pro328/Blink.ino.o' failed
make: *** [build-pro328/Blink.ino.o] Error 1
```

ANY help would *very* appreciated!

sej7278, 2015-12-08:
if you're using ide 1.5 or later then you need to specify a BOARD_SUB not just BOARD_TAG for the pro mini, you have to look in boards.txt to see that as not all boards need it (only those with a menu.cpu) in ide 1.0 you don't have to.

1.6.6:
```
BOARD_TAG = pro
BOARD_SUB = 8MHzatmega328
include /usr/share/arduino/Arduino.mk
```

1.0.5:
```
BOARD_TAG = pro328
include /usr/share/arduino/Arduino.mk
```

i'd stay away from using boards manager and make sure to install new boards to your hardware directory instead, but certainly any regular avr board works for me with any build from 1.0.1-1.6.6

have a look at the [blink example](https://github.com/sudar/Arduino-Makefile/blob/master/examples/Blink/Makefile) for examples of how to specify boards and subs etc.

T3KNOCRAT, 2015-12-08:
AHHA!!  Ok, now I understand what's happening.  That explanation was *just* the one I needed.  Thanks so very much!!

closed 2015-12-08 by T3KNOCRAT