sudar/Arduino-Makefile #401: avr-g++: error: missing argument to '-mmcu='

yerzhik, 2015-12-23:
Trying to make the following project: 
https://github.com/rpicopter/ArduinoMotionSensorExample

Installed arduino on arch linux and followed steps to install the arduino makefile.
I'm having arduino inside the intel edison board.

Here is makefile:

    ARDUINO_LIBS = Wire SoftwareSerial 
    BOARD_TAG = atmega328
    # BOARD_SUB = atmega328 # tried to add this before, didn't help. not sure what to put there as well
    # BOARD_TAG = uno
    # BOARD_TAG = mega2560
    # BOARD_TAG = leonardo
    F_CPU = 8000000L
    # CXXFLAGS+=-DI2CDEV_SERIAL_DEBUG 
    # CXXFLAGS+=-DI2CDEV_SERIAL_DEBUG -DMPU6050 -DMPU_DEBUG 
    CXXFLAGS+=-DMPU9150 
    
    include ../Arduino-Makefile/Arduino.mk

Now trying to make and it gives the following error:
```
make
-------------------------
Arduino.mk Configuration:
- [AUTODETECTED]       CURRENT_OS = LINUX 
- [AUTODETECTED]       ARDUINO_DIR = /usr/share/arduino 
- [COMPUTED]           ARDMK_DIR = /home/erjik/projects/robotics/ArduinoMotionSensorExample-master/Arduino-Makefile (relative to Common.mk)
- [AUTODETECTED]       ARDUINO_VERSION = 167 
- [DEFAULT]            ARCHITECTURE = avr 
- [DEFAULT]            ARDMK_VENDOR = arduino 
- [AUTODETECTED]       ARDUINO_PREFERENCES_PATH = /home/erjik/.arduino15/preferences.txt 
- [AUTODETECTED]       ARDUINO_SKETCHBOOK = /home/erjik/Arduino (from arduino preferences file)
- [BUNDLED]            AVR_TOOLS_DIR = /usr/share/arduino/hardware/tools/avr (in Arduino distribution)
- [COMPUTED]           ARDUINO_LIB_PATH = /usr/share/arduino/libraries (from ARDUINO_DIR)
- [COMPUTED]           ARDUINO_PLATFORM_LIB_PATH = /usr/share/arduino/hardware/arduino/avr/libraries (from ARDUINO_DIR)
- [COMPUTED]           ARDUINO_VAR_PATH = /usr/share/arduino/hardware/arduino/avr/variants (from ARDUINO_DIR)
- [COMPUTED]           BOARDS_TXT = /usr/share/arduino/hardware/arduino/avr/boards.txt (from ARDUINO_DIR)
- [DEFAULT]            USER_LIB_PATH = /home/erjik/Arduino/libraries (in user sketchbook)
- [DEFAULT]            PRE_BUILD_HOOK = pre-build-hook.sh 
- [USER]               BOARD_SUB = atmega328 
- [USER]               BOARD_TAG = atmega328 
- [COMPUTED]           CORE =  (from build.core)
- [COMPUTED]           VARIANT =  (from build.variant)
- [COMPUTED]           OBJDIR = build-atmega328-atmega328 (from BOARD_TAG)
- [DEFAULT]            ARDUINO_CORE_PATH = /usr/share/arduino/hardware/arduino/avr/cores/arduino 
- [DETECTED]           MONITOR_BAUDRATE = 38400  (in sketch)
- [DEFAULT]            OPTIMIZATION_LEVEL = s 
- [DEFAULT]            MCU_FLAG_NAME = mmcu 
- [DEFAULT]            CFLAGS_STD =  
- [DEFAULT]            CXXFLAGS_STD =  
- [AUTODETECTED]       DEVICE_PATH = /dev/ttyACM0 
- [DEFAULT]            FORCE_MONITOR_PORT =  
- [AUTODETECTED]       Size utility: AVR-aware for enhanced output
-
-                      ARDUINO_LIBS =
- [PLATFORM]             SoftwareSerial
- [PLATFORM]             Wire
- [COMPUTED]           BOOTLOADER_PARENT = /usr/share/arduino/hardware/arduino/avr/bootloaders (from ARDUINO_DIR)
- [COMPUTED]           ARDMK_VERSION = 1.5 
- [COMPUTED]           CC_VERSION = 4.8.1 (avr-gcc)
-------------------------
mkdir -p build-atmega328-atmega328
/usr/share/arduino/hardware/tools/avr/bin/avr-g++ -MMD -c -mmcu= -DF_CPU=8000000L -DARDUINO=167 -DARDUINO_ARCH_AVR -D__PROG_TYPES_COMPAT__ -I/usr/share/arduino/hardware/arduino/avr/cores/arduino -I/usr/share/arduino/hardware/arduino/avr/variants/    -I/usr/share/arduino/hardware/arduino/avr/libraries/SoftwareSerial   -I/usr/share/arduino/hardware/arduino/avr/libraries/Wire -I/usr/share/arduino/hardware/arduino/avr/libraries/Wire/utility  -Wall -ffunction-sections -fdata-sections -Os -DMPU9150  -fno-exceptions  I2Cdev.cpp -o build-atmega328-atmega328/I2Cdev.cpp.o
avr-g++: error: missing argument to '-mmcu='
../Arduino-Makefile/Arduino.mk:1207: recipe for target 'build-atmega328-atmega328/I2Cdev.cpp.o' failed
make: *** [build-atmega328-atmega328/I2Cdev.cpp.o] Error 1
```

sej7278, 2015-12-23:
i've no idea about the edison, but suspect it uses an x86 chip not an atmega328

anyway, the first thing that stands out is that you've defined BOARD_TAG and BOARD_SUB as the same thing, that isn't correct.

do you have a specific boards.txt (and core) for the edison, or does it have a bare atmega328 onboard or somesuch. essentially if its not a regular board like an uno, then you'll need a boards.txt

if it emulates an uno then you'll need to set BOARD_TAG=uno and not set BOARD_SUB

yerzhik, 2015-12-23:
Thank you! 
I tried as you;ve said to set it to uno. It passed ok but not uploading to device (I set  MONITOR_PORT  = /dev/ttyACM0 )
Here is the [output of make] (http://pastebin.com/4Na5aXie)
Regarding the It doesn't say anything about what is it, I found just this:

Digital pins 0 to 13 (and the adjacent AREF and GND pins), analog inputs 0 to 5, the power header, ICSP header, and the UART port pins (0 and 1) are all in the same locations as on the Arduino Uno R3 (Arduino 1.0 pinout).

yerzhik, 2015-12-23:
I found the boards.txt file:
```
# See: http://code.google.com/p/arduino/wiki/Platforms

menu.cpu=Processor

##############################################################

izmir_ec.name=Intel® Edison
izmir_ec.upload.tool=izmirdl
izmir_ec.upload.protocol=sam-ba
izmir_ec.upload.maximum_size=10000000
izmir_ec.upload.use_1200bps_touch=false
izmir_ec.upload.wait_for_upload_port=false
izmir_ec.upload.native_usb=false

izmir_ec.build.mcu=i686
izmir_ec.build.f_cpu=-m32
izmir_ec.build.core=arduino
izmir_ec.build.variant=edison_fab_c
#izmir_ec.build.variant_system_lib=libx86_izmir_gcc_rel.a
izmir_ec.build.toolchain_path=x86_64-pokysdk-linux-eglibc/usr/bin/i586-poky-linux
izmir_ec.build.sysroot_path=i586-poky-linux-eglibc
izmir_ec.build.toolchain_prefix=i586-poky-linux-
```

What should I do next?

sej7278, 2015-12-23:
i'm not convinced this is going to work as it uses an x86 toolchain (do you even have that installed?) but it should be ```BOARD_TAG=izmir_ec```, see where that gets you

yerzhik, 2015-12-23:
here is the log of compiling and uploading to the intel edison board. logged by arduino:

[Compilation log] (http://pastebin.com/cL2NLE7K)

Can you please take a look on that. I think its using some special tools for intel edison board. So correct me if I'm wrong but does that mean I have to setup makefile to do same things like its seen from the link above?

I tried setting board tag as you suggested but its showing same error. Also I tried to transfer the code compiled with BOARD_TAG set to uno, it didn't work. (said its transferred but serial monitor is showing garbage like: 
```
**B0100000063f694
�**B0100000063f694
�**B0100000063f694
�
```

sej7278, 2015-12-23:
i've no experience of the edison, but suspect you'd need to figure out what toolchain its uses and the weird uploader script etc. you may then be able to plug that into a makefile like chipkit.mk or teensy.mk but doubt its simple enough just to plug variables into the regular arduino.mk

you may be better off just calling arduino-builder like the ide does....

yerzhik, 2015-12-23:
Thank you very much. 
I tried that way already, same output as arduino itself does it says: 

```
sketch/I2Cdev.cpp: In static member function 'static boolean Fastwire::waitInt()':
I2Cdev.cpp:718: error: 'TWCR' was not declared in this scope
         while (!(TWCR & (1 << TWINT)) && l-- > 0);
                  ^
I2Cdev.cpp:718: error: 'TWINT' was not declared in this scope
         while (!(TWCR & (1 << TWINT)) && l-- > 0);

```

etc

sej7278, 2015-12-23:
the library says its been tested on arduino 1.0.5 with our makefile and a pro mini 328p, so i'd say ditch the edison and go with a pro mini or uno clone from ebay, that whole library looks very specific to AVR chips

sej7278, 2017-04-13:
closing as old and we don't support x86

closed 2017-04-13 by sej7278