sudar/Arduino-Makefile #402: bootloader.path is never present in Arduino IDE 1.6.x

kota65535, 2015-12-28:
bootloader.path parameter is never present in boards.txt at Arduino IDE 1.6.x, while IDE 1.0.5 is.

At 1.0.5, the bootloader.path and file parameter is defined as follows:

```
mega2560.bootloader.path=stk500v2
mega2560.bootloader.file=stk500boot_v2_mega2560.hex
```

But at 1.6.x, bootloader.path is deleted and integrated to bootloader.file.

```
mega.menu.cpu.atmega2560.bootloader.file=stk500v2/stk500boot_v2_mega2560.hex
```

Because of this, parsing for BOOTLOADER_PATH fails and null string is assigned to BOOTLOADER_PATH. This results in AVRDUDE_ISP_BURN_BOOTLOADER also being null string, and causes that "make burn_bootloader" does not burn bootloader.

sej7278, 2015-12-28:
probably need to remove the check for BOOTLOADER_PATH at line 1361/1364

https://github.com/sudar/Arduino-Makefile/blob/master/Arduino.mk#L1361

sej7278, 2015-12-28:
Can you check this works for you please?

https://github.com/sej7278/Arduino-Makefile/commit/c67f505a710c594c36974d23b3223a95aabd9c8f

sej7278, 2015-12-30:
Fix merged

closed 2015-12-30 by sej7278