sudar/Arduino-Makefile #404: Travis-CI no longer works

sej7278, 2015-12-28:
Seems its looking for python setuptools v18.8.1 but v19.2 is being installed over the top of it. noticed this hasn't worked for the last 2 commits, relevant part of the log:

```
Traceback (most recent call last):

  File "/usr/local/bin/pip", line 9, in <module>

    load_entry_point('pip==7.1.2', 'console_scripts', 'pip')()

  File "/usr/local/lib/python2.7/dist-packages/pip-7.1.2-py2.7.egg/pip/__init__.py", line 217, in main

    return command.main(cmd_args)

  File "/usr/local/lib/python2.7/dist-packages/pip-7.1.2-py2.7.egg/pip/basecommand.py", line 248, in main

    pip_version_check(session)

  File "/usr/local/lib/python2.7/dist-packages/pip-7.1.2-py2.7.egg/pip/utils/outdated.py", line 102, in pip_version_check

    installed_version = get_installed_version("pip")

  File "/usr/local/lib/python2.7/dist-packages/pip-7.1.2-py2.7.egg/pip/utils/__init__.py", line 858, in get_installed_version

    working_set = pkg_resources.WorkingSet()

  File "/usr/local/lib/python2.7/dist-packages/pip-7.1.2-py2.7.egg/pip/_vendor/pkg_resources/__init__.py", line 629, in __init__

    self.add_entry(entry)

  File "/usr/local/lib/python2.7/dist-packages/pip-7.1.2-py2.7.egg/pip/_vendor/pkg_resources/__init__.py", line 685, in add_entry

    for dist in find_distributions(entry, True):

  File "/usr/local/lib/python2.7/dist-packages/pip-7.1.2-py2.7.egg/pip/_vendor/pkg_resources/__init__.py", line 2075, in find_eggs_in_zip

    if metadata.has_metadata('PKG-INFO'):

  File "/usr/local/lib/python2.7/dist-packages/pip-7.1.2-py2.7.egg/pip/_vendor/pkg_resources/__init__.py", line 1605, in has_metadata

    return self.egg_info and self._has(self._fn(self.egg_info, name))

  File "/usr/local/lib/python2.7/dist-packages/pip-7.1.2-py2.7.egg/pip/_vendor/pkg_resources/__init__.py", line 1963, in _has

    return zip_path in self.zipinfo or zip_path in self._index()

  File "/usr/local/lib/python2.7/dist-packages/pip-7.1.2-py2.7.egg/pip/_vendor/pkg_resources/__init__.py", line 1843, in zipinfo

    return self._zip_manifests.load(self.loader.archive)

  File "/usr/local/lib/python2.7/dist-packages/pip-7.1.2-py2.7.egg/pip/_vendor/pkg_resources/__init__.py", line 1783, in load

    mtime = os.stat(path).st_mtime

OSError: [Errno 2] No such file or directory: '/usr/local/lib/python2.7/dist-packages/setuptools-18.8.1-py2.7.egg'

```

sudar, 2015-12-28:
I don't remember us changing anything in our test scripts in a while. Not sure if Travis have changed anything in their end.

sudar, 2015-12-28:
Okay, looks like something have indeed changed at Travis - https://blog.travis-ci.com/2015-11-27-moving-to-a-more-elastic-future

commit 6ac297de (2016-02-09): Don't upgrade setuptools while bootstrap travis environment

For some reason the command `pip install --upgrade setuptools` fails in
the new travis build infrastructure.

You can see the full error log in #404

Fix #404
closed 2016-02-09 by sudar