sudar/Arduino-Makefile #405: Arduino IDE 1.6.7 plus custom boards

linuxdevel, 2016-01-14:
Hi
Nice work with the makefile for arduino!

If using the Arduino-Makefile with the above mentioned IDE plus adding an extra board using the "board admin" GUI, the makefile does not set correct include paths.

The problem (?) is that the IDE does not add new boards to the default boards.txt file, instead it is located as shown below for the Blend Micro board:

username@sekenani:/opt/home/username/.arduino15/packages/RedBearLab/hardware/avr/1.0.2> pwd
/opt/home/username/.arduino15/packages/RedBearLab/hardware/avr/1.0.2
username@sekenani:/opt/home/username/.arduino15/packages/RedBearLab/hardware/avr/1.0.2> ls 
boards.txt  bootloaders  platform.txt  variants

If I now set the BOARDS_TXT to point to the new location, the makefile failes to set correct include paths.
If I dont set the BOARDS_TXT, the BOARD_TAG, e.g: blendmicro8 is not found since the board configuration is not in the arduinofolder/hardware/arduino/avr folder.

Arne

sej7278, 2016-01-14:
yes boards mangler puts the boards in the wrong place. if you put them in ~/sketchbook/hardware then it works fine, for example the adafruit boards package:

Makefile:
```make
ARDUINO_DIR = /opt/arduino-1.6.7
ALTERNATE_CORE = adafruit
BOARD_TAG = trinket3
include /usr/share/arduino/Arduino.mk
```

~/sketchbook directory structure:
```
hardware/
├── adafruit
│   └── avr
│       ├── boards.txt
│       ├── bootloaders
│       │   ├── caterina
│       │   └── optiboot
│       ├── platform.txt
│       └── variants
│           ├── bluefruitmicro
│           ├── flora
│           └── tiny8
```

linuxdevel, 2016-01-14:
Does board mangler put its config in any of the config files in ./arduino15 ? If so maybe it has location for boards that are downloaded - which can be used in the Makefile? :)

Arne

sej7278, 2016-01-14:
as i recall it puts various files in ~/.arduino15/staging, potentially you could plug those paths into BOARDS_TXT, ALTERNATE_CORE_PATH etc.