sudar/Arduino-Makefile #413: Uploading to Micro failed: AttributeError: 'Serial' object has no attribute 'setBaudrate'

benjaminaigner, 2016-02-21:
The Makefile works well compiling the sourcecode, but the upload fails with following error:

`
/home/xxx/.Arduino-Makefile/bin/ard-reset-arduino --caterina  /dev/ttyACM0
Traceback (most recent call last):
  File "/home/xxx/.Arduino-Makefile/bin/ard-reset-arduino", line 20, in <module>
    ser.setBaudrate(1200)
AttributeError: 'Serial' object has no attribute 'setBaudrate'
/home/xxx/.Arduino-Makefile/Arduino.mk:1452: recipe for target 'reset' failed
make[1]: *** [reset] Error 1
make[1]: Leaving directory '/home/beni/sync/Projects/unsorted/gesture'
/home/xxx/.Arduino-Makefile/Arduino.mk:1428: recipe for target 'upload' failed
make: *** [upload] Error 2
`

The upload with the Arduino IDE is working.

The Makefile snippet:

`
ARDUINO_DIR   = /usr/share/arduino
ARDMK_DIR     = /home/xxx/.Arduino-Makefile
AVR_TOOLS_DIR = /usr
ARDUINO_LIBS = 
BOARD_TAG = micro
MONITOR_PORT = /dev/ttyACM0
ARDUINO_LIBS = NewPing
include /home/xxx/.Arduino-Makefile/Arduino.mk
`

System setup:

Debian Testing x64
Arduino 2:1.0.5+dfsg2-4 (through apt)

Is there any relation to the missing baudrate on a USB CDC device?

Best regards

sej7278, 2016-02-21:
did you try doing a new git pull as this issue was down to a typo in a recent python3 commit that's been fixed

https://github.com/sudar/Arduino-Makefile/commit/03f5852f2dd34476ba833656593af8b4658ffbd3

you may be better off using arduino-mk from apt unless you want the very latest commits.

benjaminaigner, 2016-02-21:
I updated the my Makefile to use the apt version, but the error still exists:

`ARDMK_DIR = /usr/share/arduino/`
`include /usr/share/arduino/Arduino.mk`

According to `python`, I have the version `Python 2.7.11 (default, Jan 11 2016, 21:04:40) `, so it seems the default python binary is pointing to python2.7.

sej7278, 2016-02-21:
seems sid has moved to pyserial 3 and always calls /usr/bin/ard-reset-arduino so we need to get the pyserial3 fix into debian

in the meantime you can overwrite /usr/bin/ard-reset-arduino with the git version

sudar, 2016-02-21:
@sej7278 

Can you please explain the issue with python3 fix and our debian package?

benjaminaigner, 2016-02-21:
Replacing /usr/bin/ard-reset-arduino with the current git version e2f63c0 fixes this error

closed 2016-02-21 by benjaminaigner

sej7278, 2016-02-21:
@sudar debian have moved from python-serial 2 to python-serial 3 which spells "baudrate" differently, so without the python 2 vs 3 check in the recent commit, the makefile defaults to the python2 method, which is no longer installed so fails to upload over serial.

oddly enough for debian, they replaced pyserial2 with pyserial3, they used to be able to be installed concurrently, which is why its not been spotted before i guess.

luckily enough the recent 2 vs 3 commit is backwards compatible, so saves some headache with packaging for debian stable vs testing - we can use the same package, so big shout out to @gauteh

i've made a new build [here](https://www.the-jedi.co.uk/downloads/arduino/arduino-mk_1.5-3_all.deb) which i'll forward onto scott for upload

sudar, 2016-02-22:
> debian have moved from python-serial 2 to python-serial 3 which spells "baudrate" differently, so without the python 2 vs 3 check in the recent commit, the makefile defaults to the python2 method, which is no longer installed so fails to upload over serial.

Thanks for the explanation.

I initially thought that the recent change to support Python 3 caused this issue, but it looks like the recent change actually fixed it :)

Also thanks for making the debian build. Did you built it from master? If yes, then I will tag master as well in git.

sej7278, 2016-02-22:
yes i built it from the latest commit in master (submenu one) as its a bit of an emergency to get it into debian i didn't wait for a whole release

sudar, 2016-02-22:
Okay, I will go ahead and tag the release.

What is the version number that you used for the debian build?

sej7278, 2016-02-22:
1.5-3 is the debian build (and the unofficial fedora rpm i made)