sudar/Arduino-Makefile #417: Fuse set problem

thalesmaoa, 2016-03-02:
Hi, I have all my make configured, but when I try to set fuses, they don't work.

> ISP_LOCK_FUSE_PRE  = 0xfc
ISP_LOCK_FUSE_POST = 0xfc
ISP_HIGH_FUSE      = 0xd9
ISP_LOW_FUSE       = 0xff
ISP_EXT_FUSE       = 0xfc

But I can only get them working with:

> AVRDUDE_OPTS 	    = -U hfuse:w:0xD9:m -U lfuse:w:0xFF:m -U efuse:w:0xFC:m -v -v

Besides, AVRDUDE_OPTS run before the erase process, but the fuses are supposed to write after.

Any idea?

sudar, 2016-03-02:
Can you post your entire makefile and the complete output?

sej7278, 2016-03-02:
yes can't really tell without the Makefile and output.

i've never tried using lock bits, but your AVRDUDE_OPTS has skipped them too, have you tried removing them from your Makefile to see if the h/l/e fuses get set then?

also avrdude ignores unset bits, so 0xFF should be 0x07, i've come across that problem myself.

ref. http://www.avrfreaks.net/comment/673717#comment-673717

thalesmaoa, 2016-03-03:
Thanks

Makefile
http://pastebin.com/CShtKUKY

With that file I have the following output:

http://pastebin.com/2J4LXEXf
No fuses written.
If I change:
`AVRDUDE_OPTS 	    = -U hfuse:w:0xD9:m -U lfuse:w:0xFF:m -U efuse:w:0xFC:m -v -v`

I get the following:

http://pastebin.com/yrFwGJxC

Fuses are written but out of order

sej7278, 2016-03-03:
ah i see the problem - ```make ispload``` doesn't set the fuses, you do that with ```make set_fuses```.

```make burn_bootloader``` uploads the sketch, bootloader and sets the fuses, but we separated out the fuse and ispload recipes.

in your second test case, this is a known [avrdude bug](http://savannah.nongnu.org/bugs/?40794), you can see it reads the 3 fuses in the correct order, but then the last line is out of order - just ignore the last line or upgrade to avrdude 6.1 or later.

```
avrdude: safemode: lfuse reads as FF 
avrdude: safemode: hfuse reads as D9  
avrdude: safemode: efuse reads as FC  
avrdude: safemode: Fuses OK (H:FC, E:D9, L:FF)    <-- only this line is wrong
```

thalesmaoa, 2016-03-03:
Very good to know, thank you.

About the order, I was referring that lock bits are reset during chip
erase. So I must set them after burn my code.

Using in avr_options they all set before, and them I lost my lock fuse.

Now I will try make set_fuse after ispload.

Thank you

thalesmaoa, 2016-03-03:
Perfect! Works very well

I was wondering, is there a way to create custom targets?

`make project1`
`make project2`

In general, I work with the same program files, but in the lines codes, I just need to change a few variables:

Project 01
`int x = 1`

Project 02
`int x = 2`

If I create a target in make file, I can create multiple definition files:

`var_proj1.cpp`
`var_proj2.cpp`

And them just copy and rename it:
`make project1`
Inside Makefile:
`mv var_proj1.cpp var_proj.cpp`

Where var_proj.cpp is the real include in my main.cpp.
`#include "var_proj.cpp"`

sej7278, 2016-03-03:
i think you should probably look at preprocessor defines (-D flag) so you can just modify what the code does based in a variable, rather than using separate files and moving them:

https://gcc.gnu.org/onlinedocs/gcc/Preprocessor-Options.html#Preprocessor-Options

thalesmaoa, 2016-03-04:
When I use make set_fuses my processor is erased.

thalesmaoa, 2016-03-04:
I found a problem:

`ifneq ($(strip $(AVRDUDE_ISP_FUSES_PRE)),)`

Always runs true and my processor is always erased

sej7278, 2016-03-04:
see: https://github.com/sudar/Arduino-Makefile/blob/master/Arduino.mk#L1308

if you set ```AVRDUDE_ARD_OPTS``` and override the defaults, make sure to include the -D flag

thalesmaoa, 2016-03-04:
-D flag is not a solution.

Please, take a look at:
https://github.com/sudar/Arduino-Makefile/blob/master/Arduino.mk#L1471

ISP_LOCK_FUSE_PRE means that the lock fuse is perform before the burning, which means that -e flag is ok.
ISP_LOCK_FUSE_POST means that the lock fuse is perform after the burning.

In my case, I just only set the post flag, but it happens that both are run.
First it run the pre and after the post.
It is doing write, first -e and after not.

The problem is that:

`ifneq ($(strip $(AVRDUDE_ISP_FUSES_PRE)),)`

is always true when I try make set_fuses

sej7278, 2016-03-04:
you can only use one lock command - so either PRE or POST not both. it doesn't really make sense to use the -e flag with POST and both of your PRE and POST values are the same, so just use PRE not POST i guess

thalesmaoa, 2016-03-04:
Indeed, I do not use PRE, only POST, but as I mentioned, I always runs PRE. The conditional is always true no matter what.
It seems a bug.

sej7278, 2016-03-04:
could you try changing:

```
ifneq ($(strip $(AVRDUDE_ISP_FUSES_PRE)),)
```

to:

```
ifndef AVRDUDE_ISP_FUSES_PRE
```

sej7278, 2016-03-04:
yup, AVRDUDE_ISP_FUSES_PRE is always going to be defined and non-empty if you set any of the fuse variables:

https://github.com/sudar/Arduino-Makefile/blob/master/Arduino.mk#L1340

unless you can figure out a fix, the workaround is to use PRE not POST

thalesmaoa, 2016-03-04:
`ifneq ($(strip $(AVRDUDE_ISP_FUSES_PRE)),)`

replace by

`ifneq ($(ISP_LOCK_FUSE_PRE),)`

If not set, do nothing.

sej7278, 2016-03-04:
don't confuse lock fuse with isp fuse, almost nobody uses lock fuses and that change would force everyone to use the lock fuse!

thalesmaoa, 2016-03-04:
I will generate a pull request soon.

thalesmaoa, 2016-03-05:
Sorry, didn't have time to try, but I would propose something like:

set_fuses:
ifdef ISP_LOCK_FUSE_PRE
	ifdef ISP_LOCK_FUSE_POST
		$(error Only set one lock fuse, pre or post)
	else
		AVRDUDE_ISP_OPTS += -e
	endif
endif

ifneq ($(strip $(AVRDUDE_ISP_FUSES_PRE)),)
		$(AVRDUDE) $(AVRDUDE_COM_OPTS) $(AVRDUDE_ISP_OPTS) $(AVRDUDE_ISP_FUSES_PRE)
endif
ifneq ($(strip $(AVRDUDE_ISP_FUSES_POST)),)
		$(AVRDUDE) $(AVRDUDE_COM_OPTS) $(AVRDUDE_ISP_OPTS) $(AVRDUDE_ISP_FUSES_POST)
endif

closed 2017-08-15 by sej7278