sudar/Arduino-Makefile #418: Windows make trouble

thalesmaoa, 2016-03-02:
I was trying to build a multi-platform with the project and had a lot of troubles with windows (cygwin).

The first one is that my com port can not be found. I already have python and pyserial. Tried upper and lower case as suggested in other issues, but it seems not to work.

Also, a friend want to help with my project, so I wanted to share my makefile with him to compile in Windows. In cygwin, the make works ok (it always compile the whole project again).

But I tried the make from arduino folder without sucess. I just want to use the makefile in eclipse, but I can't find a suitable solution.

Is there a way to make in cmd? Or embed in eclipse?

sudar, 2016-03-02:
> The first one is that my com port can not be found. I already have python and pyserial. Tried upper and lower case as suggested in other issues, but it seems not to work.

I think this is related to #294 

Try out the solutions provided there to see if that fixes your problem.

thalesmaoa, 2016-03-02:
I have checked the issue, and made more testes.

If I ran:
`MONITOR_PORT=com3 make reset`
or 
`MONITOR_PORT=3 make reset`
or
`MONITOR_PORT=/dev/ttyWhatEver make reset`

It always give me:
`Resetting Arduino...
Arduino-Makefile-master/bin/ard-reset-arduino  /dev/ttyS2
make: *** [reset] Error 1`

If I ran:
` make ispload`
`Arduino-Makefile-master/Arduino.mk:1468: *** ISP port com3 not found!.  Pare.`

I tried the suggested variations: 'com3', '3'

If I try with /dev/ttyS0 or ttyS2

, 'avrdude.exe: ser_open(): can't open device "/dev/ttyS0": O sistema nao pode encontrar o caminho especificado.'

Python and pyserial are ok.

thalesmaoa, 2016-03-03:
Just re-installed everything and the error persist.

`$ python --version
Python 2.7.10`

`$ perl --version
This is perl 5, version 22, subversion 1 (v5.22.1) built for cygwin-thread-multi`

I just figured that make upload works:
`MONITOR_PORT = 3`
or 
`MONITOR_PORT = com3`

If they are uppercase it doesn't.

If I set NO_CORE, I must also set 
`AVRDUDE_ARD_PROGRAMMER = wiring`
`AVRDUDE_ARD_BAUDRATE = 115200`

But I can only get ispload working with linux. I will keep digging.

thalesmaoa, 2016-03-03:
I found the problem:

`get_isp_port = $(if $(wildcard $(ISP_PORT)),$(firstword $(wildcard $(ISP_PORT))),$(if $(findstring Xusb,X$(ISP_PORT)),$(ISP_PORT),$(error ISP port $(ISP_PORT) not found!)))`

Give me and erro because wildcard can`t fint pattern in com3.

I just declare after the call:
`get_isp_port = com3`

It worked.

`$ uname -a`
`CYGWIN_NT-6.1 2.4.1(0.293/5/3) 2016-01-24 11:26 x86_64 Cygwin`