sudar/Arduino-Makefile #419: avrdude.conf in yet another location (Linux)

marcusrugger, 2016-03-05:
Today I decided I started using Arduino-Makefile, but discovered a problem with the location avrdude.conf.

I've read the comments in the Arduino.mk concerning the location of avrdude.conf and see that on Linux avrdude.conf can usually be found in /usr/bin or /usr/etc.  However, on my system I didn't find it in either location.  Instead, I found it here:

/usr/share/arduino/hardware/tools/avrdude.conf

I changed the variable AVRDUDE_CONF on my system to point to the correct location and I'm running fine.

However, I thought I'd bring it up since I didn't see any information expecting to find it in this location.  As far as I'm aware, I didn't install anything in an unusual way on my system.  I download the Arduino IDE and unzip it.  The IDE works fine.  I've been using the IDE for a couple of years, so perhaps it was in the way I originally installed the AVR tools, I don't know because I don't remember.

Anyway, thought I'd bring it up.

Current setup:
Ubuntu 14.04
Arduino 1.6.7

sej7278, 2016-03-05:
the current setup is based on the debian/ubuntu/arch -packaged version of avrdude, not the arduino zip from the website, /etc is where you'd expect to find a conf file, /usr is a bit odd - you sure its not just a symlink to /etc ?

marcusrugger, 2016-03-05:
You're right.  It is a symlink to /etc, and avrdude.conf is in there.

But, if I don't set AVRDUDE_CONF and just let the makefile do its thing, it's looking for it in /usr/etc, which doesn't exist.

marcusrugger, 2016-03-05:
It appears to be because of the following lines in Arduino.mk:

```make
    ifndef AVRDUDE_CONF
        ifeq ($(shell expr $(ARDUINO_VERSION) '>' 157), 1)
            AVRDUDE_CONF = $(AVR_TOOLS_DIR)/etc/avrdude.conf
        else
            AVRDUDE_CONF = $(AVR_TOOLS_DIR)/../avrdude.conf
        endif
    endif
```

My version is 1.6.7 and the tools dir is set to /usr, so the makefile is setting the conf file to /usr/etc/avrdude.conf.

sej7278, 2016-03-05:
when i use 1.6.7 i set ```ARDUINO_DIR``` to where i unzipped it e.g. ``` ARDUINO_DIR = $(HOME)/arduino-1.6.7``` and then its finds everything just fine e.g. ```AVR_TOOLS_DIR = $(HOME)/arduino-1.6.7/hardware/tools/avr``` see https://github.com/sudar/Arduino-Makefile/blob/master/examples/Blink/Makefile#L18

what's in your Makefile, as if its setting the tools dir to /usr then it seems its trying to pull in the 1.0.5 ubuntu package instead of the 1.6.7 zip

marcusrugger, 2016-03-05:
OK, you're right, again.  I'm not quite sure how I got down the path I went down.

Thanks for your time.

closed 2016-03-05 by marcusrugger

sej7278, 2016-03-05:
not a problem. maybe we need clearer documentation. have fun!